1000001101000/Debian_on_Buffalo #1: Network Shut Down

jcmoore1977, 2018-11-25:
I have tried this, using firmware 1.84, one hard drive in my 4 bay LS441DE. When it boots to installer and I ssh to it. The network card on the ls turns off. Only way to get it back on is to plug power plug on ls. Then start the process over only to have it do the same thing. Network card on the ls will not even stay on long enough to run installer

1000001101000, 2018-11-26:
Good to hear from you!

I never ran into that on the LS441 but it's been a while at this point. Assuming it's not something wrong with your device (which seems unlikely if stock firmware was working), I have some ideas on how to troubleshoot.

How far have you gotten into the installation process before loosing connection (what screen have you gotten to)?

Do you have a usb network adapter you can use for troubleshooting? If not you can get one from amazon for < $5.

The installer enables all attached network devices including usb ones, this would allow you to connect via the second network connection and grab the dmesg output after the network drops so we can try to understand what's happening. You could also try completing the installation that way and see whether the issue persists or is specific to the installer. 

I'll try it out on mine as well and see if I can reproduce the issue, though it'll probably be the weekend after next before I get a chance.

Thanks for the feedback, I'm sure we can get this straightened out.

jcmoore1977, 2018-11-26:
Yes I can get one of those, with stock firmware the connection never shuts off. As far as the installation I have made it all the way to grub failing to install. But sometimes it stops before this. It is a hit or miss. I also start with a blank drive ext2 at 100mb. I put the INITRD and uImage on there. I format drive using the installer. It starts down loading files then ssh will stop. I could put router on open all ports for my ls. I just don’t know if that’s the issue, because of stock firmware working perfectly.

1000001101000, 2018-11-26:
That's frustrating, I hate intermittent issues. But if you're getting that far we should be be able to learn a lot once we get another NIC working. 

Let me know when you get that far, it should fairly easy to start narrowing down the issue once we get an alternate way to connect to the device.

1000001101000, 2018-11-26:
So, I tried running the installer on a TS1400D and encountered what seems to be the same issue you reported. That device had been running the latest kernel + device tree for about a month without any issues.

This makes me wonder if there could be something wrong with this version of the installer. Could you give it a try with an older version of the installer image, you should be able to get them from these links:

https://github.com/1000001101000/Debian_on_Buffalo/blob/002ca2bb4a1de8f8292b9d52c404ed10d8239264/Stretch/installer_images/uImage.buffalo.ls441d

https://github.com/1000001101000/Debian_on_Buffalo/blob/9fd313e0ae79d4052d808b7954e66df5eab56105/Stretch/installer_images/initrd.buffalo

I'll do the same, let me know if that helps or not. If so this might be a fairly easy fix.

jcmoore1977, 2018-11-26:
I will try this as soon as I can, and report back.

1000001101000, 2018-11-28:
> So, I tried running the installer on a TS1400D and encountered what seems to be the same issue you reported. That device had been running the latest kernel + device tree for about a month without [...]

jcmoore1977, 2018-11-28:
I downloaded the files you post above. Unfortunately I will not be able to test this out until weekend. I am busy with work atm. I will post the progress this weekend for sure.

1000001101000, 2018-11-29:
Sounds good, let me know what happens.

I went ahead and ran through the install a couple more times on the ts1400d (closest thing to a ls441de I have online currently), I haven't run into the issue again. 

I went ahead and rebuilt all the dtbs and downloaded the latest installer image and kernel and rebuilt all the installer files, the resulting files were all identical to what I have posted. That rules out a few possibilities but doesn't tell us much.

I should be able to try the install out on an actual ls441de sometime next week.

jcmoore1977, 2018-12-01:
Installer tells me this. I am proceeding to see how this will work.
" No kernel modules were found. This probably is due to a mismatch      │
   │ between the kernel used by this version of the installer and the      │
   │ kernel version available in the archive.                              │
   │                                                                       │
   │ If you're installing from a mirror, you can work around this problem  │
   │ by choosing to install a different version of Debian. The install     │
   │ will probably fail to work if you continue without kernel modules. "

jcmoore1977, 2018-12-01:
Okay that did not work.. I will try the uImage on here and the initrd I downloaded above.

1000001101000, 2018-12-01:
Sorry, I must have given you bad links. Sounds like you’ve got the 4.9.0.7 kernel but an initrd with 4.9.0.8 modules (or vice versa). 

If you boot up again you can tell which the kernel is by running uname -a. You can see which module are installed by looking at /lib/modules.

I was trying to link to a matching kernel/initrd from before the current one but messed up

jcmoore1977, 2018-12-01:
yes give me a few mins. I will report back

1000001101000, 2018-12-01:
Looks like I linked you the right kernel but the wrong initrd somehow (I’m assuming you’ll see 4.9.0.8 modules).

I think if you find the oldest commit for initrd.buffalo it should be the right one for that kernel.

1000001101000, 2018-12-01:
If not it may be that it may be that debian isn’t hosting the 4.9.0.7 modules anymore.

Sorry, I should have tried this before telling you to try it. I was just trying to find something we can try until I can get set up for testing.

jcmoore1977, 2018-12-01:
I think i may have downloaded the correct initrd. I am going to copy it over and try.

jcmoore1977, 2018-12-01:
No luck. I may need to down load all files for this comment e69b79

jcmoore1977, 2018-12-01:
Yeah that doesn't work.

1000001101000, 2018-12-01:
I should be able to set up my ls441de and do some testing around Wednesday.

In the meantime you could try the latest files for the ts1400d. They should work on the ls441de (the leds/buttons just wont work). That could help us narrow down what the problem is a bit.

Otherwise once I can test with the same setup you have we should be able to make better progress

jcmoore1977, 2018-12-01:
I can try. Will report back.

jcmoore1977, 2018-12-01:
Set up partitions, running installer now. When I get to the grub error. I will continue with notes, but use the dtb for 441de instead.

1000001101000, 2018-12-01:
Cool.

If you run:

echo “Buffalo Linkstation LS441D” > /etc/flash-kernel/machine

It will pick the right dtb when flask-kernel runs

jcmoore1977, 2018-12-01:
update-initramfs: Generating /boot/initrd.img-4.9.0-8-armmp
Unsupported platform.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
error.......

jcmoore1977, 2018-12-01:
If I take that comment out of machine it tells me it can not find ts1400d.dtb looks like i need to install that one for it to work.

jcmoore1977, 2018-12-01:
I will rename file first.

1000001101000, 2018-12-01:
Unsupported platform means the model didnt match anything in the .db file

The line in “machine” needs to match the entry in the .db file exactly (inculding any spaces which can be annoying). If there is no “machine” file it goes off of /proc/device-tree/model.

You can always just rename the dtb to match of that’s easier.

jcmoore1977, 2018-12-01:
Got machine to work had to take out quotes.
update-initramfs: Generating /boot/initrd.img-4.9.0-8-armmp
Using DTB: armada-370-linkstation-ls441d.dtb
Installing /etc/flash-kernel/dtbs/armada-370-linkstation-ls441d.dtb into /boot/dtbs/4.9.0-8-armmp/armada-370-linkstation-ls441d.dtb
Installing new armada-370-linkstation-ls441d.dtb.
Installing /etc/flash-kernel/dtbs/armada-370-linkstation-ls441d.dtb into /boot/dtbs/4.9.0-8-armmp/armada-370-linkstation-ls441d.dtb
Taking backup of armada-370-linkstation-ls441d.dtb.
Installing new armada-370-linkstation-ls441d.dtb.
flash-kernel: installing version 4.9.0-8-armmp
flash-kernel: appending /etc/flash-kernel/dtbs/armada-370-linkstation-ls441d.dtb to kernel
Generating kernel u-boot image... done.
Installing new uImage.buffalo.
Generating initramfs u-boot image... done.
Installing new initrd.buffalo.

1000001101000, 2018-12-01:
That looks promising

jcmoore1977, 2018-12-01:
wrapping up installer.. crossing fingers

jcmoore1977, 2018-12-01:
reboot successful. Continue with post install.

1000001101000, 2018-12-01:
Nice!

Hopefully it will be more stable than it was in the installer. If you do run into crashes send me a syslog and we’ll see what’s going on.

I’ve got some changes from the other devices I need to port to the ls441, i’ll make sure to do some extra stability testing once I get the new dts/dtb ready.

jcmoore1977, 2018-12-01:
Okay just making my ip static for now, then will post back the syslog after completing all other post installs.

jcmoore1977, 2018-12-01:
Now it's not detecting my static ip.... mmmmm

1000001101000, 2018-12-01:
Might try regenerating your inird (update-initramfs -u). Otherwise if you post your interfaces file i’ll take a look

jcmoore1977, 2018-12-01:
got it. I forgot to add something to interfaces files..

jcmoore1977, 2018-12-01:
I will say the reboot command does not work. It just turn off the device

jcmoore1977, 2018-12-01:
Network just shut down. I will have to reboot. I hope the syslog grab the info

jcmoore1977, 2018-12-01:
do you know what this error may be, I am wondering if its the cause of network issues.

debian systemd-udevd[209]: Could not generate persistent MAC address for eth0: No such file or directory

jcmoore1977, 2018-12-01:
Okay after network shut down system powered off.
Let see what the log has to say
[syslog.zip](https://github.com/1000001101000/Debian_on_Buffalo/files/2636294/syslog.zip)

1000001101000, 2018-12-01:
Looks like i’ve gotten that error at some point for a device that has otherwise worked fine.

root@LS421DE:~# grep MAC /var/log/*log*
/var/log/daemon.log.1:Nov  3 12:16:47 LS421DE systemd-udevd[228]: Could not generate persistent MAC address for eth0: No such file or directory
root@LS421DE:~#

1000001101000, 2018-12-01:
Sounds more and more like something specific to the 441 dtb, if you keep having instability try switching back to the 1400d and see if that works better for a few days. If so it should be easy enough to identify the difference.

I’ll dig through that syslog once I get in front of a computer.

jcmoore1977, 2018-12-01:
Cool I'll leave you to it.. Gonna play around now, see what else I can do.

jcmoore1977, 2018-12-01:
One more though, I change it to the ls1400d, network still shuts down.I just don't get it.

1000001101000, 2018-12-01:
might mean it's something specific to the ls441 (hopefully not just yours). 

hopefully the samething will happen on mine and will be something I can fix

jcmoore1977, 2018-12-01:
I am wondering if there is a bug in the network drive named driver=mvneta driverversion=1.0

I am going to boot my original firmware to see the driver name, maybe I can find one or build one for this.

1000001101000, 2018-12-01:
I’ve been meaning to ask whether we’re sure its the network failing rather than a kernel panic or something (hard to tell with headless devices).

I see some errors in that syslog I want to follow ip on ( some also show up on the 1400d but none are on the 421de). I didn’t see anything that looked specific to the network failing, but i’m just getting started.

You could try running a script that blinks an led in a loop and see if it keeps blinking after the next network failure. Presumably if the whole system crashed it would stop blinking.

jcmoore1977, 2018-12-01:
Yeah I can try, I did also see you set your MAC address in /etc/network/interfaces

I did not do this. Once I return home I will set mine and see if that randomly setting one will fix the issue. As of now it does stop at random times. It could be a crash.

jcmoore1977, 2018-12-02:
The light keeps blinking, for about 30 secs, then system powers off. I was just using the network for 30 mins.

jcmoore1977, 2018-12-02:
I may have something....

This error keeps popping up......
systemd-logind[351]: Power key pressed.

I am not even near the machine, I am going to edit file /etc/systemd/logind.conf

and change HandlePowerKey to ignore and see how long it will stay on..

jcmoore1977, 2018-12-02:
That didn't work. Back to drawing board. It shut down less than 2 mins

jcmoore1977, 2018-12-02:
Okay, I have formatted my drive again. I am so fed up right now. Just let me know when you get your 441d working. And I will follow the steps. I will continue this adventure next week sometime. 48 hours and still network cutting out.

I did notice a difference in your dts files from one to another.....
Some say 

mdio {
				/* Marvell 88E1518 */
				phy0: ethernet-phy@0 {
					reg = <0>;
					marvell,reg-init = <0x3 0x10 0xf000 0x081e>,
							   <0x3 0x11 0xfff0 0x0001>;
					compatible = "marvell,88e1518", "ethernet-phy-ieee802.3-c22";

and other say...

mdio {
				pinctrl-0 = <&mdio_pins>;
				pinctrl-names = "default";

				phy0: ethernet-phy@0 {
					reg = <0>;
				};
			};

This might as well be Greek to me. I don't understand device tree that much, even after searching all day and reading.

1000001101000, 2018-12-03:
I'm hoping with all the things you've tried and errors you've uncovered It should be pretty easy to replicate the issue and hopefully correct it. From the syslog and other errors you've sent me I've got a couple of things to address right away, hopefully one of them is the culprit.

I'm goi g to revisit that network binding again in particular, it worked just fine for me whenI tested on the ls441 a year ago and has been working on the ts1400d and ts1400r without issue but it seems worth checking into.

Hopefully I'll have some new dtbs for you to test in a few days

jcmoore1977, 2018-12-03:
I am wondering if I use the 421 uImage will my 441 boot. If so see if network crashes, then add the 441.dtb to install at that part if it add all my hardware. This would be another trouble shooting step.

1000001101000, 2018-12-03:
Way back when I was first working with just the 421 and 441 a year ago I was able to boot the 441 with a 421 dtb but the network didn't work at all. I was able to use a USB NIC to connect it to the network for troubleshooting and was eventually able to get everything else working including the built in network. 

Things have evolved quite a bit since then so I'm not entirely sure what would happen if you try it now. 

At this point I think the best first step is to try on another device and see if we get the same result as yours, hopefully I should get that far tonight or tomorrow.

1000001101000, 2018-12-04:
I put my ls441de online for about an hour without issue. I just updated to the latest kernel and made sure I've got the current dtb from the repository. So far it's been stable. Looking at my syslog I am seeing the same errors I saw in yours. I'm going to leave it online a while longer then try running through the installer.

which log did you see that "power key pressed" message in? I haven't seen that yet but want to be on the look out for it.

jcmoore1977, 2018-12-04:
Under var/apt or something like that, if yours has been online, I am wondering if my router is causing the issue, I will not be able to install until I get the usb nic adapter, but wondering if I network my pc and link station to a hub, not internet router if it will stay online. It’s just weird how I don’t have issues with stock firmware. Ever

jcmoore1977, 2018-12-04:
Also what size drive you using, and how did you format it. I use a 4TB with only one part, at 100mb for the uImage and initrd file

1000001101000, 2018-12-04:
I hour, mostly idle:
es from guspad (192.168.1.81): icmp_seq=118 ttl=64 time=52.4 ms
^C
--- guspad ping statistics ---
118 packets transmitted, 118 received, 0% packet loss, time 117192ms
rtt min/avg/max/mdev = 1.866/85.268/2034.549/206.266 ms, pipe 3
jeremy@earth:~$ ssh root@192.168.1.184
Linux ls441de 4.9.0-8-armmp #1 SMP Debian 4.9.130-2 (2018-10-27) armv7l
Last login: Mon Dec  3 19:40:45 2018 from 192.168.1.146
root@ls441de:~# clear

root@ls441de:~# cat /proc/device-tree/model &&  echo && md5sum /boot/dtb && uname -a && uptime
Buffalo Linkstation LS441D
5704e82139b715ef164d38e01e3bc712  /boot/dtb
Linux ls441de 4.9.0-8-armmp #1 SMP Debian 4.9.130-2 (2018-10-27) armv7l GNU/Linux
 19:57:25 up  1:07,  2 users,  load average: 0.07, 0.02, 0.00

1000001101000, 2018-12-04:
I’m using a cheap 16gb ssd for testing:

root@ls441de:~# gdisk -l /dev/sda
GPT fdisk (gdisk) version 1.0.1

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.
Disk /dev/sda: 30932992 sectors, 14.8 GiB
Logical sector size: 512 bytes
Disk identifier (GUID): 67204D91-7740-4C9F-9371-6EBC91FDFB83
Partition table holds up to 128 entries
First usable sector is 34, last usable sector is 30932958
Partitions will be aligned on 2048-sector boundaries
Total free space is 16926653 sectors (8.1 GiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048         2002943   977.0 MiB   0700  primary
   2         2002944        12003327   4.8 GiB     0700  primary
   3        12003328        12005375   1024.0 KiB  0700  primary
   4        12005376        12007423   1024.0 KiB  0700  primary
   5        12007424        14008319   977.0 M

1000001101000, 2018-12-04:
I haven’t checked everything yet, but I have a hunch about yours.

Let me look at a couple of things then send you a dtb to try

1000001101000, 2018-12-04:
Give this one a try please
[dtb_test1.zip](https://github.com/1000001101000/Debian_on_Buffalo/files/2642121/dtb_test1.zip)

[21 more replies]

closed 2019-01-18 by 1000001101000