1000001101000/Debian_on_Buffalo #3: LS420D image doesn't work

rogers0, 2018-12-23:
I tried the images you built, under Stretch folder, but I cannot find the device on my network after booted some time.

Since I saw there's another report stating not working, could you kindly list the models you tested, and another list for un-tested devices?
Thank you!

1000001101000, 2018-12-23:
good to hear from you. I learned a lot from your scripts for setting up the orion/kirkwood devices. I've been thinking of reaching out for advice on moving foward.

As far as testing goes I've tested all the devices, and actively re-test as I make changes. There were some issues with the ls441de that I beleive I've resolved now, though I haven't heard from the reporter in a couple weeks.

I'll load up the current image on my ls420d and try it out today to confirm. 

Currently when using the kernel provided by debian the device is assigned a random MAC address at boot, could the reason you're not finding it be because of that (either getting assinged an unexpected ip address or some issue with MAC whitelisting)?

1000001101000, 2018-12-23:
I booted it up with the installer image and it came up just fine, accept for the whole random MAC thing.

eventually I want to put together a preseed file and some scripts to automate some of the non-standard steps in the process to make it a bit more seemless.

right now I'm focused on updating the dts's to encorporate some of the things I learned working on the ls441 issue and better standardize some of the naming/etc within them. Once that's all done and tested I'll probably start workig on that... unless Debian moves 4.19+ to testing in which case I'll probably work on the Buster dts's first.

1000001101000, 2018-12-25:
My understanding is that it's felated to how this soc passes the MAC address to the kernel. There is a patch that adds that support and a device tree binding to support it. You can see the patch and the device tree entry here:

https://github.com/tohenk/linkstation-mod/tree/master/lskernel/patches/4.9

I haven't inculded it since my goal has been to make things work with the kernel provided by Debian. If you know of a way to make it work I'd be glad to hear it, I included instructions on how to retrieve and hard-code the MAC after setup but it is still rather annoying.

1000001101000, 2018-12-25:
root@LS420D:~# ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.133  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::126f:3fff:fecc:223b  prefixlen 64  scopeid 0x20<link>
        ether 10:6f:3f:cc:22:3b  txqueuelen 532  (Ethernet)
        RX packets 44068  bytes 5996932 (5.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4695  bytes 458294 (447.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 26  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@LS420D:~#

1000001101000, 2019-01-10:
I've updated the installer to automate all of the steps required to get the mac address from the spi flash at run time and then hardcode it into /etc/network/interfaces at the end of the install process.

i'd really prefer to not have to do it that way but don't see an alternative at the moment. At some point I plan to dig into the differences between how this works for kirkwood/orion vs armada-370/xp to understand the difference. It seems that patch I linked to (or something similar) would be the way to deal with it but i don't personally plan to try to get it into the mainline kernel, though I do wonder if anyone else has tried.

closed 2019-02-18 by 1000001101000