1000001101000/Debian_on_Buffalo #5: Better Instructions

iamnorbiato, 2019-02-05:
Hi, sorry about this kind of topic, but, I think I'm not the only one with installations informations problems.
I acquired this device LS441D 6 months ago supposing to be able to use rsync, nfs server, smb server (I have at home linux, mac and windows devices), but I got so disappointed and want to use "a linux based" kernel.
I'm not a guru like you guys, I know the basics. I'm trying to get this working fine, but the step-by-step in the main page, for me at least is too complicated:
Steps 1 to 4, for a not Linux developer guru is a little bit complicated, I appreciate if the steps could me be more.... easier and detailed - have no idea how to obtain the DTB or transfer the installer...
Once again... I'm very sorry to post this message here.
I was so excited when plug the device on, put  the 4 4TB disks, but after created the raid, transfer the files then... the daily basis stuffs such as sync files, copy files and so on, eveything using windows explorer, copy from raid to usb disk must be made.... over the lan.. oh it is crazy...

Thanks.

Marco

1000001101000, 2019-02-05:
hey, good to hear from you!

it probably is time for me to update the readme page as things have changed a little bit since I first posted. It sounded from your description like you may not have seen the fulltext instructions. checkout these files in the main directory and see if they are what you're looking for:
1-Obtain install files.txt
2-Copy_Installer_to_Device.txt 
3-Run_Installer.txt            
4-Post_Install_Setup.txt

let me know if that helps, I may have misunderstood. If you still need help we'll look closer and see what is needed.

Thanks for reaching out!

iamnorbiato, 2019-02-05:
Hey - this is so cool, you replied !

Some thought
#1 and #2 - I found the vmlinuz file
 #3 - Append the DTB - should I do that using a linux box ? to do the cat command ? then make the image ? for that I think I need to have a linux with dev tools installed and make the image - what linux distro should work better ? is it possible to already put the image created for us to download ?

I'm not saying that I don't want to do it, would be awesome for me to learn more about that (I'm more a database guy - db2 in fact) but, for us that just want a nas that supposedly should be... "turn on and done" do all these tricks sometimes sounds a little bit scary...

All the other steps such as "load the modules, spi flash".. well this is the piece that make me.... "quit" and leave the nas as is... but... I want more... was using an old laptop as "nas like"...

Anyway, should be better if we (regular users, not linux developers) could at least with Linux/Unix skills following a step by step instructions do all the steps and have it working as a linux nas allowing us to configure:
Wake on lan, ssh, rsync, nfs, smb and done.....

P.S. - right now, I'm taking a backup of my data, cuz I'm very confident that before next monday this piece of junk (so far it is) will be running as a linux nas !

Thanks.

1000001101000, 2019-02-06:
Ah, ok. Make sure you have latest version of the instructions most of those steps are automatic now.

1000001101000, 2019-02-06:
I should probably look at separating the instructions for using the tools I provide and the instructions for how to do things manually, I think that's where the confusion is coming from. simply put the process is:

1. Download initrd.buffalo and uImage.buffalo.<model> for your device (I suggest using Stretch)
2. rename uImage.buffalo.<model> to uImage.buffalo
3. Copy both files to your device using one of the methods described in 2-Copy_Installer_to_Device.txt
4. reboot and run through the installer as described in 3-Run_Installer.txt (it's just the normal debian installer over ssh) the only requirement is that /boot is mounted on the first partition of the drive (or the first software raid). the easiuest way is to re-use the partitions/arrays you already have from the buffalo firmware.
5. since you have the ls441 you'll want to setup the ls400_restart service as described in 4-Post_Install_Setup.txt

If you can handle a DB2 install you'll be fine with this. I once setup SQL Server Reporting Services to connect to a DB2 datasource..... this is much easier.

iamnorbiato, 2019-02-06:
Hi there.. good news... I read again, and the BOOMER here did not realized that all the files where there.. for download...

I'm very happy and excited - I'm connected into the NAS (telnet) and... this is so cool, as soon as I have it done, will let you know..

Rock and Roll !!!!!! Now I know that I'll be able to setup .. nfs, smb, torrent, I'll be able to do a rsycn between nas and macos !

iamnorbiato, 2019-02-07:
I had an error, or I did something wrong.
I ran into steps #1 #2 and #3 after all packages copied, it asked to boot but... root passwd nor created user worked, looks like I forgot to mount someting...

What should I do after install ? (I did not see any grub message)

1000001101000, 2019-02-07:
I don't understand. can you post a screenshot of the error/message you see?

1000001101000, 2019-02-07:
the most likely place to make a mistake (the main thing I haven't automated) is setting up the partitions in the installer. If you try again post a screen shot of that part of the installer and I can validate your setup

iamnorbiato, 2019-02-07:
Starting a new fresh install, as soon as I reach the partition piece, will post the screenshots

iamnorbiato, 2019-02-07:
[image]

[image]

More soon

iamnorbiato, 2019-02-07:
[image]

[image]

Will say now... finish and write partition... should I go for it ?

1000001101000, 2019-02-07:
That shows you installing everything to disk 2, if you only install /boot on 1 drive it has to be the first one or it wont boot.

I’d recoment using the raid devices instead with the first 1gb as /boot, the 5gb as / and the other 1gb as swap

iamnorbiato, 2019-02-07:
[image]

Am I in the right path now ?

1000001101000, 2019-02-07:
Yes, that looks right

iamnorbiato, 2019-02-07:
Stay tuned... it is performing the rest of the install

iamnorbiato, 2019-02-07:
[image]

Time to boot now....

1000001101000, 2019-02-07:
Good luck!

iamnorbiato, 2019-02-07:
I can ping, but ssh refused... all 4 green disks led on (no blink), first white on (no blink) and the bottom is red (no blink) - looks like it is... briked ?

1000001101000, 2019-02-07:
Ping is good, ssh takes a minute or two to come up

iamnorbiato, 2019-02-07:
no good... more then 10 minutes.. all lights has no blink.. and a red one is on...

1000001101000, 2019-02-07:
Which one?

My guess is that the first time you did the install you overrote the paritioms on the first disk so that they are no longer part of the raid arrays.

Try swapping disk 1 and disk 4 and try booting again

iamnorbiato, 2019-02-07:
[image]

I dont care for the raid now, I have a backup of everything..

iamnorbiato, 2019-02-07:
picture is 90 degrees

iamnorbiato, 2019-02-07:
disks 1 and 4 switched

iamnorbiato, 2019-02-07:
same thing... I think I need I need to back to the factory firmware...

1000001101000, 2019-02-07:
Yeah, i’d say let the stock firmware rebuild the raid arrays for you then try again. Starting with the raid arrays built saves a lot of trouble.

iamnorbiato, 2019-02-07:
I can "flame" this raid, the data is saved into other 3 external disks... so, will do that "default procedure" to build the nas.. and move on, start over

iamnorbiato, 2019-02-07:
I made it !

Let me share what I did:
1 - I took a backup of all data
2 - in the partition part, I deleted all RAID partitions and configurations
3 - removed all scsii partitions.... leaved all 4 disk with no partitions
4 - in the disk #1 I created 3 partitions - 1GB for /boot 1GB for swap 5GB for /

Done, I'm in now installing - sudo, smb, nfs, mdadm, trying to install webmin or any other "server admin" web interface to make my life easier !

The good news is... I'm in

1000001101000, 2019-02-07:
Nice!

trying it that way was going to be my next recommendation, if only so you could get familiar with the process and be sure that you're able to get it working.

personally I'd recommend going back at some point and getting it working with the raid1 arrays for /boot, / and swap so that the OS can survive a disk failure..... but that does add complexity.

I had some luck setting up openmediavault on one of these devices a year or so ago. I'm more of a "do it yourself" type and didn't keep it and don't have any long term experience with it.

1000001101000, 2019-02-07:
crap.... I just looked at your partition screenshot again and I see what the problem was. 

When you tried raid you formatted /boot as ext4, it has to to be ext3(or ext2) for the bootloader to access it .

sorry I didn't notice it the first time.

iamnorbiato, 2019-02-07:
One no good news is... having some issues to create the RAID.. saying that the /dev/sda busy, of course it is, the / filesystem is /dev/sda1....

Now trying to figure out what is wrong... the worst scenario.. will put the default firmware again, create the raid5 then.. do your procedure again (I'm starting to sord of... dominate the process :-) )

iamnorbiato, 2019-02-07:
ohhhh.. another thing, well, my case first, I think most of the "linkstation" users are the same:
1 - we are not storage wizards (well, I need to understand due to the database skills stuffs)
2 - We used to be like "buy it, use it", so, sometimes it is a challenge to update firmware "manually"
3 - attempt and error - we can learn a lot of things such as:
    I think, re-creating the RAID from the factory firmware will point me to: the swap boot and / partitions when switching to debian THEY MUST EXISTS IN ALL DISKS (RAID) my mistake was - I was using only in one RAID od SCSII, but they must live spread accross all disks.
Anyway, I'll do it all over again, will post some screenshots step by step, will add some hints and latter on will post the webmin screen.
Lets build a better install documentation and spread this out, and make Buffalo create a better OS

1000001101000, 2019-02-07:
Yeah, these make for a good learning environment to practice managing software raid. A lot of the same concepts apply working with database servers even though the tools are very different.

iamnorbiato, 2019-02-07:
1200 minutes to finish the sync - created a new array from scretch.
This time, after copy the install images to the NAS and start the installation, will choose the 1st option in the partitioning and let the install pick the best part of the disks, will only check the sizes.

1000001101000, 2019-02-07:
don't do that, the debian installer doesn't know the requirements to make these devices work.

1000001101000, 2019-02-07:
infact, the /boot array has to be metadata version 0.90 or 1.0 for the device to boot from raid. The debian installer will only create arrays with metadata version 1.2.

iamnorbiato, 2019-02-07:
well, after array sync I'll start all over and will post here when I reach the partition part...

iamnorbiato, 2019-02-08:
Bit guy... starting another install attempt, this time, will leave the disks as is (created by default firmware) and will let the installer pick the space in the partitions... fingers crossed...

1000001101000, 2019-02-08:
Again, the installer will not make correct choices for you

Go with these:
#1 raid (1gb) ext3 /boot
#2 raid (5gb) ext4 /

Feel free to have me look at a screenshot again when you get there
#3 raid (1gb) swap

iamnorbiato, 2019-02-08:
well, I already say Ok.. and it is going... all "default" and I had this:
raid1 - 1GB /boot
raid1 - 5GB /
raid1 - 1gb swap

rai5 - 6TB

then the "scsii" but this is the "physical" disks - if this attempt did not work, I can paste here the screenshot and then you can help me...

No big deal, no data will be lost

1000001101000, 2019-02-08:
The defaults are better than I would have thought.

Still, if /boot is not ext3 the device will not boot

iamnorbiato, 2019-02-08:
ext3 both / and /boot

iamnorbiato, 2019-02-08:
time of the truth

iamnorbiato, 2019-02-08:
yep... did not boot as expected... will need to configure again the 3 partitions (/boot / and swap)

iamnorbiato, 2019-02-08:
[image]

[image]

What should I pick ?

1000001101000, 2019-02-08:
Those raid choices look good this time

iamnorbiato, 2019-02-08:
except the rai5 correct ? this is what I choose:

RAID1 device #0 - 1.0 GB Linux Software RAID Array
>     #1      1.0 GB    F  ext3                 /boot
RAID1 device #1 - 5.1 GB Linux Software RAID Array   
>     #1      5.1 GB    F  ext3                 /    
RAID5 device #10 - 6.0 TB Linux Software RAID Array  
>     #1      6.0 TB       xfs                       
RAID1 device #2 - 1.0 GB Linux Software RAID Array   
>     #1      1.0 GB    F  swap                 swap 
SCSI1 (0,0,0) (sda) - 2.0 TB ATA ST2000VM003-1ET1        
>     #1      1.0 GB       ext3        primary           
>           458.2 kB       FREE SPACE

iamnorbiato, 2019-02-08:
This is what I picked:

**RAID1 device #0 - 1.0 GB Linux Software RAID Array
     #1      1.0 GB    F  ext3                 /boot**
**RAID1 device #1 - 5.1 GB Linux Software RAID Array   
     #1      5.1 GB    F  ext3                 /**    
RAID5 device #10 - 6.0 TB Linux Software RAID Array  
     #1      6.0 TB       xfs                       
**RAID1 device #2 - 1.0 GB Linux Software RAID Array   
     #1      1.0 GB    F  swap                 swap** 
SCSI1 (0,0,0) (sda) - 2.0 TB ATA ST2000VM003-1ET1        
     #1      1.0 GB       ext3        primary           
           458.2 kB       FREE SPACE

1000001101000, 2019-02-08:
Right on

iamnorbiato, 2019-02-08:
good !!!

run forrest run !!!

iamnorbiato, 2019-02-08:
Rebooting

iamnorbiato, 2019-02-08:
I'm in !!!!!!!!!!

iamnorbiato, 2019-02-08:
It did not mount the RAID

norbiato@HomeNAS:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            232M     0  232M   0% /dev
tmpfs            50M  972K   49M   2% /run
/dev/md1        4.7G  682M  3.8G  16% /
tmpfs           248M     0  248M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           248M     0  248M   0% /sys/fs/cgroup
/dev/md0        946M   45M  853M   5% /boot
norbiato@HomeNAS:~$

Will do the post migration stuffs now !

1000001101000, 2019-02-08:
Good work!

iamnorbiato, 2019-02-08:
[image]

I have a Linux/Debian storage "in da house" !!!!!!!!

Rock and roll !!!!!!!!!!!!

iamnorbiato, 2019-02-08:
I'll write out a few comments for you add in the install procedure - some "hints and tips" regarding partitions....   hope that could help others !

iamnorbiato, 2019-02-08:
latter will post a webmin scrennshot !

Time for a cold beer now !

iamnorbiato, 2019-02-09:
Hi there... yesterday the RAID5 (create from stock firmware) was too slow, so I decided to start from scretch.

I did all the setup parts, kept only the raid1 for /boot / and swap

Now, I'm trying to create from debian the new raid5 with the remaining space, do you mind to help me out ?

1000001101000, 2019-02-09:
This seems like a good guide:
https://www.digitalocean.com/community/tutorials/how-to-create-raid-arrays-with-mdadm-on-ubuntu-16-04

Once you get the array working you want make sure to add it to mdadm.conf the  generate a new initrd (it tells you to do both)

iamnorbiato, 2019-02-09:
Will try...
 I "loose" again the last install due to... removing partitions... but, will be a good practice...

[6 more replies]

closed 2019-02-14 by 1000001101000