1000001101000/Debian_on_Buffalo #6: Users Account

amroth3003, 2019-02-22:
Hi, use the configuration of the raid of linkstattion ls220D, I start with the normal debian installation, but once it finishes, I do not create my user account and do not set the password in root or in my account, which could be

1000001101000, 2019-02-22:
I don't understand.

Could you post a screen shot of your disk/partition layout from that part of the installer and a screenshot of the last page you are seeing or the error message you are getting?

1000001101000, 2019-02-22:
Here's a screenshot from someone who successfully set up their ls220d:
[image]

whcrg, 2019-02-23:
I am pretty sure the installer asked for a root password (needed once to go and setup sudo for the normal account)

1000001101000, 2019-02-23:
You set a root password as like the 3rd prompt before even partitioning. You make a good point though, if you skip that I think it goes a different path and sets up sudo. I’ve sadly never tried that and am unfamilar with the difference though it shouldn’t be that confusing.

1000001101000, 2019-02-23:
The poster left a comment on a commit a while back that clearly indicated they hadn’t set /boot properly because it kept rebooting back into the installer. I’m not sure if that’s still the case.

amroth3003, 2019-02-23:
specifically that happens in the ls220d the installation is reinitiated despite making the complete installation, setting the root passwords and local user

1000001101000, 2019-02-23:
The reason that keeps happening is that for some reason the installer boot files aren’t being replaced by the new ones created by the installer. The most likely explaination being that you are not specifying the separate /boot when you choose the partitions.

Take a look at the screenshot of @whcrg ’s partitions as an example of a working setup.

amroth3003, 2019-02-23:
use the own partitions of the ls220d that includes raid0

1000001101000, 2019-02-23:
Specifically /boot must be on a seperate partition which is the first partition of the first disk and must be formated ext3 (this is because of how the stock bootloader works). It can be part of a raid1 array as long as it’s still the first partition and as long as the array uses metadata version 0.90 or 1.0 (sadly the installer will only create version 1.2 so I recommend using arrays set up by buffalo’s firmware)

1000001101000, 2019-02-23:
Yes, you basically want to use the same partition layout buffalo uses.

amroth3003, 2019-02-23:
I do not eliminate the buffalo matrix, that I try to explain to you, I use the partitions of the buffalo itself

1000001101000, 2019-02-23:
For some reason /boot is not getting reformatted/replaced.
If it was it wouldn’t be able to boot back into the installer as you describe, the installer would be deleted/replaced by the new boot files.

Please try again and post a screen shot of your partition layout when tou get that far, that should let us see what’s going on.

amroth3003, 2019-02-23:
ok

amroth3003, 2019-02-23:
now ls220d does not open port 22936

1000001101000, 2019-02-23:
in your previous message you said it was still booting the debian installer, if that was the case only port 22 would be open. 22936 will only open if it boots th buffalo OS which you replaced with the debian installer, then an attempted debian install.

what are you attempting to do now?

amroth3003, 2019-02-23:
Ls220d bricked and restore firmware buffalo, but now not open port 22936 for acp commander

amroth3003, 2019-02-23:
Instead of opening port 22936 opens port 22939

1000001101000, 2019-02-23:
Off the top of my head I don’t remember which ports it uses or if they’re the same all the time.

The acp commander cli version has a “-f” which can search the network for devices which might help. Otherwise the gui version will just find it for you when it opens.

I’m still not entirely clear what point your at, are you saying you’ve successfully re-installed the buffalo firmware and are now trying to copy over the installer again?

amroth3003, 2019-02-23:
Yes, restoring buffalo firmware and now try enter ssh and install firmware debian, but the port is change 22936 to 22939, my computer used windows s.o not linux

1000001101000, 2019-02-23:
Sounds good, you should be able to get it working via the methods I mentioned above. You may need to log into the web interface and make sure an admin password is set/ run the setup wizard before it will talk to acp commander.

amroth3003, 2019-02-25:
Hi Step for install
**Set Password Roor**
[image]

**Set Password For my User**
[image]

**Partitions**
[image]

1000001101000, 2019-02-25:
So far so good.
You’ll want raid#0 set to use as /boot (keep ext3)
Raid#1 as “/“ (format ext4)
Raid#2 as “/mnt/array” (format ext4 unless you prefer something else)
Raid#3 as swap

amroth3003, 2019-02-25:
Brick again!!

1000001101000, 2019-02-25:
Were those raid devices created by buffalo’s firmware or did you create them?

1000001101000, 2019-02-25:
judging by the layout they aren’t from buffalo’s firmware. To boot of a raid1 “/boot” the array has to be built using metadata=0.90 or 1.0. Unfortunately the debian installer will only create them with metadata=1.2. 

If you keep the first raid device from the buffalo install it will be setup properly, otherwise you’ll need to do it manually from the command line similar to:

mdadm —create md0 -l 1 -n2 —metadata=1.0 /dev/sd[ab]1

amroth3003, 2019-02-25:
Partitions original buffalo ls220d

[image]

now error

[image]

1000001101000, 2019-02-25:
looks like you have the raid#2 device marked as swap but also have sda5 and sdb5.

I believe you should change sda5 and sdb5 to "do not use" and oly have raid#2 as swap

amroth3003, 2019-02-25:
ok pased this part, check finish installation

1000001101000, 2019-02-25:
while you're waiting for that to finish... could you connect another session, start a shell, and post the output of "mdadm --detail /dev/md0" ?

amroth3003, 2019-02-25:
[image]

1000001101000, 2019-02-25:
looks good!

amroth3003, 2019-02-26:
Ready, install sucess
Thanks for you help and time

closed 2019-02-26 by amroth3003

amroth3003, 2019-04-18:
Hello, i need resize partition / , free space is small

[image]

1000001101000, 2019-04-18:
1. Resizing raid arrays in that manner is impractical, I'd back up the files and start over with bigger partitions rather than attempt it.
2. 5GB is usually more than enough space for a root partition, you can always move large files to the array if needed. 
3. If you'd like to try clearing some space before proceeding I usually have success with:
apt-get clean
apt-get autoclean
investigate large files in /tmp, /var/tmp, /var/log