1000001101000/Debian_on_Buffalo #7: Stretch on LS21oDC35 boots without network

fristersoft, 2019-03-10:
Hello,
Installer for Stretch seemed to work flawlessly on LS210D but after rebooting at end of install, I can't find any open ports at the address I used during the installation, nor do I find any new DHCP slots taken in the 192.168.1.1/24 subnet. Network lights are lit on the box and the router. Should I be looking on a different subnet?
I have a nice solid white status light but no way of shutting down other than pulling the power plug.
I took the hard drive out of the box and mounted it directly on another machine, I'm not sure where I should be looking to see what went wrong. The log files all seem to end at the point where I rebooted. 

Installation of Jessie works and boots up just fine, but I'd rather run the latest stable.

Thanks.

1000001101000, 2019-03-10:
Good to hear from you!

I have some thoughts though from your description it's not clear to me what happened.

1. The installer looks up the device's MAC address from the flash memory and hardcodes it into /etc/networking/interfaces. If something went wrong with that process (though it hasn't happened before) it's conceivable that your network config would be broken. Take a look at /etc/networking/interfaces and see if it looks correct. it could also have gotten a randomized MAC address but in that case you'd still expect to see it online just with a new IP.

2. the solid white light seems to indicate that it's booting something though what it's booting and how far it's getting can be tough to determine depending on how far it's getting. If another part of the finalization process failed (again this has never happened before) it could be failing to mount the rootfs at startup.

3. almost everyone who has trouble is because they messed up formatting /boot as ext3 and ensuring it's on the first partition. Your description makes it sound like that's not the problem but I thought I'd mention it.

4. The process is basically identical with the 3 distros, if Jessie worked I'd expect them all to work.

To start I'd suggest checking that interfaces file and try the stretch install again for good measure. 

I'm going to run through the stretch install on my ls210d and see if I run into trouble. They updated the stable kernel a few days ago, I wouldn't expect that to break anything but it's possible.

1000001101000, 2019-03-10:
and yes, definitely dun't run Jessie unless you have a specific reason to. I might take those files down when Buster becomes the Stable Distro in the next few months.

fristersoft, 2019-03-10:
Hi, good to hear from you also. You replied just as I was correcting my initial comments. 
Partition 1 is ext3, mounted as /boot.
I also tried with a USB-Ethernet dongle. That didn't even show up as a dead address in my port scanner.

fristersoft, 2019-03-10:
This is the interfaces file:
    source /etc/network/interfaces.d/*
    
    # The loopback network interface
    auto lo
    iface lo inet loopback
    
    # The primary network interface
    allow-hotplug eth0
    iface eth0 inet dhcp
    hwaddress ether 88:57:EE:57:1C:35
    hwaddress ether 88:57:EE:57:1C:35
    # This is an autoconfigured IPv6 interface
    iface eth0 inet6 auto

1000001101000, 2019-03-10:
the duplicate hwaddress line is unexpected, I don't know if that would hurt anything or not.

I've never tried ipv6, I don't know how that will affect things either. DId you do that intentionally or did it do that automatically?

1000001101000, 2019-03-10:
also: my install is running now, I should know what happens in like 15 minutes

fristersoft, 2019-03-10:
the interfaces file contents are all automatic.

fristersoft, 2019-03-10:
back in a few hours

1000001101000, 2019-03-10:
I didn't notice your log excerpt when I first responded (serves me for relying on the automatic emails)
looks like it's definitely that extra line in the interfaces file, I suspect removing it would resolve the issue.

My install didn't have any trouble and I ended up with this in my interfaces file:

# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
allow-hotplug eth0
iface eth0 inet dhcp
hwaddress ether 88:57:EE:57:46:CF

Did you have that usb NIC attached when you ran the install?
also, did you run through a standard or advanced install?

fristersoft, 2019-03-10:
I shall edit the interfaces file and put the box back together. What about the
> /bin/sh: 2: ether: not found

line?

I ran the advanced installer 'cause I didn't see how the standard install would keep my partition table untouched from my  previous Jessie install. I only wanted it to reformat sda1 (1GB /boot) and sda2 (15GB as /). sda3 is swap, sda4 is the rest in xfs, mounted as /data.
Did you just run standard install?
I had run the install both without and with the USB NIC installed. I got the same result of no open ports found. The second time, I was hoping to see two different IP addresses so I could have two different ssh sessions open, one running the install, and the other to monitor logs. But, that idea didn't pan out. I only ever saw one IP address.

fristersoft, 2019-03-10:
Seems odd, but it now works! All I did was manually edit interfaces file to remove duplicate hw ID. I shall now try edit the file to assign fixed IP, then restart the network. If that works, I she;; then try a system shutdown followed by manual (switch to on) startup, just to make sure I can move the box safely.

1000001101000, 2019-03-11:
sounds like it's an issue with the script I use to add that mac address line. The script I created to hard code the MAC address is a bit crude and appears to have added that line twice since there were two "iface eth0" lines. Aparently the second line gets interpreted as an ifup script and the whole thing fails when it fails to execute.

I'll working on making that script more robust and test it against that scenario for future revisions

1000001101000, 2019-03-11:
I've always just run standard install. If you choose manual partitioning it will leave alone any partitions you don't explicitly tell it to mess with.

fristersoft, 2019-03-11:
Thanks for the **very** prompt attention. Issue closed.

closed 2019-03-11 by fristersoft

1000001101000, 2019-03-18:
Just for reference: I made some changes to the installer which should prevent those duplicate lines from being created in the future.

fristersoft, 2019-03-19:
Good to know. Thanks.