1000001101000/Debian_on_Buffalo #9: How to switch back original firmware?

jxr111, 2019-03-19:
I installed Debian 9 on my Ls421de , it's run fine. Then I want to test with original firmware,so I replace /boot  initrd.buffalo and uImage.buffalo with orginal file. After reboot ,the Ls421de is dead.
I want konw the method that switch from Debian and original firmware ,thks!

1000001101000, 2019-03-19:
To go back to the stock firmware:

1. Get the original .buffalo files if you don't already have them:
https://buffalonas.miraheze.org/wiki/Extract_Boot_Files_from_Stock_Firmware

2. Put them in your /boot partition and power on the device:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP

3. Use the debug mode of lsupdater.exe to reformat/repartition and reinstall:
https://buffalonas.miraheze.org/wiki/Enable_Debug_mode

for the ls421de there is an even easier way. just insert blank drives and wait for the flashing light, then press the button and wait:
http://d.buffalo.jp/LS421DE/?dst=AP (quick start guide)

1000001101000, 2019-03-19:
if you plan to go back and forth it saves a lot of trouble to keep one hard drive on each and swtich the drives as needed

closed 2019-03-22 by jxr111