1000001101000/Debian_on_Buffalo #10: LS441D - Failed to boot when manually create RAID10 array for Disk #1 with metaversion 0.9

kinyat, 2019-05-29:
Hi, 

I have been bricked my NAS a few times but still could get it working.

I am a bit fussy and want to get my disk partitions to look clean.

What I did is I remove all stock arrays and partition and use the installer partition disks to re-create the arrays and partition in a nice way and manually create Disk #1 Array with `mdadm` with version 0.9 as per mentioned in https://github.com/1000001101000/Debian_on_Buffalo/issues/6#issuecomment-467149142

The way how I create the array is the same as https://github.com/1000001101000/Debian_on_Buffalo/issues/5#issuecomment-461928602

Just that mine is RAID10 arrays with 4 disks

I couldn't trigger the error message about the GRUB when the installation finished.

My NAS bricks after the reboot.

I think my next step is to repartition the arrays using the stock firmware and remove everything except the Disk #1 Array which would be where the `/boot` mount to.

I think in that case, I will have a RAID 1 array Disk #1 for `/boot` and the rest for other RAID10 arrays (i.e. `/` and swap)

Just want to post my idea here so I can get some recommendations to have nice looking arrays and partitions.

Cheers

1000001101000, 2019-05-29:
Good to hear from you!

it sounds like you're on the right track.

I have some thoughts based on your questions:
1. Make sure to clone/pull the latest version of the repo, I resolved the grub error message a while back and have made some other improvements too.
2. uboot in this configuration only looks at one drive/partition to load boot files, as such raid1 is the only raid mode which you can boot from (it ignores the mdadm superblock and treats it like a normal partition)
3. I usually use raid1 for the rootfs and swap to ensure that each drive can be booted individually if needed. raid10 should work but trades flexibility for more space/performace.

if you remember, take a screenshot next time you have your partitions configured in the installer. if you run into trouble post it and I'll try and help further.

kinyat, 2019-05-29:
@1000001101000 Thanks for the prompt response :)

I am going to create 2 RAID 1 Arrays for `/root` and `swap` partitions and 1 RAID10 Array for `/`.

I will post my setting later today when I start to work on my NAS again :D

closed 2019-05-29 by kinyat