1000001101000/Debian_on_Buffalo #11: LS441D - Failed to see fan in sensor

kinyat, 2019-05-29:
When I run `sensor`, it is missing the fan.

```
root@ls441db64:~# sensors
armada_thermal-virtual-0
Adapter: Virtual device
temp1:        +70.4°C
```

As a result, when I installed `fancontrol`, it throws the following error:

```
root@ls441db64:~# pwmconfig
# pwmconfig revision 6166 (2013-05-01)
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

/usr/sbin/pwmconfig: There are no pwm-capable sensor modules installed
```

1000001101000, 2019-05-29:
Which version did you install?

This probably caused by the gpio-fan module not being loaded. Debian Jessie’s kernel didn’t come with that module which might explain that.

kinyat, 2019-05-29:
@1000001101000 Thanks for the prompt reply again :D

I was using Jessie. I will reinstall my NAS with Stretch :)

Thanks

closed 2019-05-29 by kinyat

1000001101000, 2019-05-29:
Excellent.

I think i’ll put some sort of note up about picking distros. I think I still recommend Stretch at this point though that will change in a few weeks/months when Buster gets closer to being the new “Stable”

isido, 2020-03-19:
Commenting on this. I recently installed Buster on LinkStation 220D and I also didn't see fan in the sensor output. I tried the `# modprobe gpio-fans`, but I think the module was already loaded - at least the dmesg didn't give any indications of loading the module (lsmod showed the module loaded). I will try to investigate this further.