1000001101000/Debian_on_Buffalo #12: LS441D - Issue `shutdown -r now` command will power off the NAS but then it power on again by itself.

kinyat, 2019-05-29:
I installed the latest installer image of Stretch.

After I issue `shutdown -r now`, it will turn off the NAS but straight away it is turned on again by itself.

It leaves me to choice but to unplug and plug again to power it off.

Would be appreciated if I can get some help :D 

Cheers

1000001101000, 2019-05-29:
“shutdown -r now” will cause a reboot “shutdown now” will cause it to poweroff.

shutdown --help
shutdown [OPTIONS...] [TIME] [WALL...]

Shut down the system.

     --help      Show this help
  -H --halt      Halt the machine
  -P --poweroff  Power-off the machine
  -r --reboot    Reboot the machine
  -h             Equivalent to --poweroff, overridden by --halt
  -k             Don't halt/power-off/reboot, just send warnings
     --no-wall   Don't send wall message before halt/power-off/reboot
  -c             Cancel a pending shutdown

kinyat, 2019-05-30:
@1000001101000 Thanks! I feel so dumb by asking this question lol

closed 2019-05-30 by kinyat

1000001101000, 2019-05-30:
I assume my notes on how shutdown works for each device were part of the confusion.

It’s a surprisingly complex topic with these devices