1000001101000/Debian_on_Buffalo #13: Not Issue - Sord of... Request - Wake on Lan

iamnorbiato, 2019-07-16:
HI there, is me again !
Device is working awesome !

However, is it possible to enable wake on lan ?
The storage during the night has no activities, and was wondering as a home user, to "relief" fan noise or even... cool down processor, disks and so on...

Thaks.

1000001101000, 2019-07-16:
Good to hear from you again!

If I remember correctly your device is an LS441D, correct?

WOL support is near the top of my list of things to look into for devices that support it. Unfortunately the only devices that seem to support WOL are the TS3000 series.

Interestingly, I've done some testing on the LS441D and have found that interrupt pin from the ethernet controller is connected to the power button on this device. This allows you simulate the power button being pressed when you send it a WOL packet. The problem is that there doesn't seem to be a way to keep the ethernet control active while the device is powered off, so you can only do it when the device is already on. It should be possible to use WOL to wake it from sleep/suspend, though I've never tested whether these devices support any kind of sleep mode. 

I'll play around with trying to use sleep/suspend/hibernate and wake it up via WOL when I get some time in the coming days.

iamnorbiato, 2019-07-17:
Well, the idea is not shutdown, it is put the puppy to sleep (suspend/hibernate) then with a magic packet wake it up, thus, this might save unnecessary disk run, fan, temperature and so on...
I was afraid to recompile the network module and screw up with the device that is working so good !

1000001101000, 2019-07-17:
I think we're talking about the same thing. I'll do some experiments over the next few days and let you know what I find out.

1000001101000, 2019-07-22:
I'm continuing to research this, it seems like it should be possible but some of the details are provide difficult. I'll let you know when I make some progress

1000001101000, 2019-08-09:
I've got this frustratingly close to working but haven't been able to get it to work yet. I'm going to take a break from this issue and come back to it later after I focus on some other projects for a while. One of those other projects will be testing WOL on the ts3000/ts5000 series which might give me some insight on how to get this to work someday.

what I've been able to do so far:
confirm WOL is supported on device's ethernet controller
confirm that sending WOL packet to device triggers GPIO15 (same as power button)
confirm that device keeps ethernet powered on suspend when WOL is enabled

what isn't working:
waking up device from suspend via WOL packet or power button

what I've tried:
adding wakeup-source; to the power-button entry in the DTB. 
this resulted in the device failing to suspend stating an IRQ problem with the pinctl device.
I researched that error and came across a patch in armbian which corrects this issue for other marvel armada SoCs:
https://github.com/armbian/build/blob/master/patch/kernel/mvebu-default/92-mvebu-gpio-add_wake_on_gpio_support.patch
I recompiled the kernel with this patch and the error no longer occured when trying to suspend but the device still doesn't wake up from the power-button/WOL packet.

I then tried a few different suspend settings but nothing made a difference.

I'll take this up again someday when I have some more experience with successfully suspending/waking under linux.

kaymueller, 2020-05-28:
Hey there,
I set up my device (LS220) some days ago (it went really smooth - thank you very much for all your work!) and tried to set up WOL (without knowing about this issue) today and encountered nearly the same problems as you did.

First I was trying to get WOL after poweroff to work:
As you know the LS220 has a switch on the back of it to turn it on and off at stock firmware. So first of all it's not possible to poweroff/shutdown correctly without switching the switch to "off". It will just reboot then. I may use a little workaround and just take the switch out in favour of a simple button (or just solder the button parallel to it) as it seems to just need a very short trigger to start. In fact you can switch the switch on and off in a millisecond and it will boot just fine. BUT: the network gets disabled after poweroff and I don't have a clue if it's possible to keep it alive somehow.

So next I was trying to get it to work after suspend:
The device suspends just fine in either the positions of the switch and stays suspended, but it's not possible to get it back to life. Neither with the switch at the back or WOL. BUT: the network is still enabled and the green light is on and even flickering, ping is still working.

You said that "sending WOL packet to device triggers GPIO15 (same as power button)" --> _how have you achieved to track this?_ I'm curious if it's the same on my device.

Maybe we'll find something by giving hints to each other ;-)

best regards

1000001101000, 2020-05-28:
Good to hear from you!

I think having someone else to focus on this should greatly improve the chance of some progress being made on it. 

I believe the GPIO15 -> power switch connection is specific to the LS400 devices. Particularly on the LS441DE (which has a button rather than switch) you can clearly see that toggling the LED2/WOL pin on the network chip changes the value of the GPIO pin the same way as pushing the power button. You can even see in the buffalo kernel source where this is used to signal shutdown vs poweroff on these devices. I was able to confirm this using a script I wrote that scans a range on gpio pins and looks or changes (this is how I find what pin corresponds to what button when making device-trees).

I believe I looked for something similar on the LS200 when I first discovered that but didn't find anything. I believe I even checked the stock firmware and found there isn't a separate shutdown vs restart but honestly don't remember for sure. 

As noted earlier in the thread I altered the device tree of one of these devices to try to make a gpio button a "wakeup-source" so that it could be used to wake a suspended device. I believe this prevented the device from suspending after throwing an error. I believe I tried something similar with the external RTC on an LS400 device but also failed... I don't remember any details though. In fact I remember being unsure whether I was successfully even suspending the device I was testing with.

I think a good path forward for you might be to experiment first with waking the device from suspend using a usb keyboard or something. We can also try modifying the device tree to make one of the buttons (maybe the reset button?) a wakeup-source and see if that works or not. This device doesn't have an external RTC but the SoC has one. i don't recall if I looked into that as a wakeup-source at all but it would be another thing to look at. 

If wakeup/suspend ends up being a dead end you could always look into adjusting the logs/etc to go to a tmpfs so you can spin down the hard drives when idle. From a power standpoint I think that would accomplish nearly the same savings as suspending the system. I've never looked into how to do that on a system that boots from hard drives but imagine it would be useful to folks.

Let me know how I can help you move forward in your research!

kaymueller, 2020-07-10:
Hello there,
sorry for not responding for so long, but some private stuff got in the way. 

I like the approach you mentioned in the last part. I dont think anyone really cares about the power saving they get from the suspend, it's more about the noises this thing makes the whole time. So if I can manage to let the drives spin down and maybe slow down the fan a bit it might be just enough.

I will try to spin down the disks with hdparm und check with fatrace what processes are waking up the device. It's been some time since I setup a tmpfs but it shouldn't be that hard if I remember correctly.

I'll let you know when I made some progress!

1000001101000, 2020-07-10:
Good to hear from you!

I would love it if someone could come up with a good procedure for doing that which we could add to the wiki.

kaymueller, 2020-07-10:
Well, it was quite easy to initialize a silent NAS!

I hope that I don't confuse anything, because it was quite some fiddeling around.

**1. set up the fans**
- sudo apt install fancontrol
- sudo pwmconfig
- Hit enter to continue
- Would you like to generate a detailed correlation (y)? y
- then it slows down the fans step by step
- it seems like the fans has only 3 states of running (1850, 2600 and 3500 RPM)
- The fancontrol script can automatically respond to temperature changes of your system by changing fanspeeds. Do you want to set up its configuration file now (y)? y
- What should be the path to your fancontrol config file (/etc/fancontrol)? ENTER

```
Select fan output to configure, or other action:
1) hwmon1/pwm1
2) Change INTERVAL
3) Just quit
4) Save and quit
5) Show configuration
select (1-n): 
```

- change Interval to 10 or 15 seconds
- hwmon1/pwm1:

```
Enter the low temperature (degree C)
below which the fan should spin at minimum speed (20): 60

Enter the high temperature (degree C)
over which the fan should spin at maximum speed (60): 80

Enter the minimum PWM value (0-255)
at which the fan STOPS spinning (press t to test) (100): t

Enter the minimum PWM value (4-255)
at which the fan STARTS spinning (press t to test) (150): t

Enter the PWM value (0-4) to use when the temperature
is below the low temperature limit (0): 0

Enter the PWM value (4-255) to use when the temperature
is over the high temperature limit (255): 255
```
- save and quit
- sudo systemctl enable fancontrol && sudo systemctl start fancontrol

_You have to adjust the temperature settings to how hot it is at your place - mine seems to run at like 60-62°C and the lowest fanspeed the whole time when it's around 25°C in my room_

**2. Configure the HDDs to spin down**
_I have two Toshiba drives in my NAS. these are the ones that got delivered with the NAS. Some drives don't seem to work with the tools I used, but most modern drives should! I include some testing, so you can see if everything works for you._
- sudo apt install hdparm
- sudo hdparm -i /dev/sda && sudo hdparm -i /dev/sdb
- look for "AdvancedPM=yes", then everything should work as expected
- sudo ls /dev/disk/by-id > look for your physical harddrives (should start with ata-)
- sudo nano /etc/hdparm.conf
- simply add this to the end of the file:
```
/dev/disk/by-id/ata-TOSHIBA_DT01ACA300_XXXXXXXXX {
        apm = 255
        acoustic_management = 128
        spindown_time = 180
}
/dev/disk/by-id/ata-TOSHIBA_DT01ACA300_YYYYYYYYY {
        apm = 255
        acoustic_management = 128
        spindown_time = 180
}
```
_apm = 255 disables the automatic power management, which enables us to set it manual with spindown_time. You shouldn't set it too low, because a lot of stops and starts isn't that good for the drive. The maximum value is 255. Just multiply the value with 5, to get the spindown time in seconds. So in my case it's 15 minutes._

- sudo udevadm trigger

The last command should trigger the new values to be enabled. But I would recommend to simply

- sudo reboot

**The problems**
- It seems like that changes in the fanspeed are triggering the drives to spin up, so playing around with the values might help. But I'm not sure if it's just correlating or if it's the real reason it spins up.
- cronjobs - I identified them using a little script I found at: https://stackoverflow.com/a/137173 
```
mi  h  d  m  w  user  command
17  *  *  *  *  root  /etc/cron.hourly/fake-hwclock
25  6  *  *  *  root  /etc/cron.daily/apt-compat
25  6  *  *  *  root  /etc/cron.daily/apt-show-versions
25  6  *  *  *  root  /etc/cron.daily/bsdmainutils
25  6  *  *  *  root  /etc/cron.daily/dpkg
25  6  *  *  *  root  /etc/cron.daily/logrotate
25  6  *  *  *  root  /etc/cron.daily/man-db
25  6  *  *  *  root  /etc/cron.daily/mdadm
25  6  *  *  *  root  /etc/cron.daily/ntp
25  6  *  *  *  root  /etc/cron.daily/passwd
47  6  *  *  7  root  /etc/cron.weekly/man-db
57  0  *  *  0  root  if [ -x /usr/share/mdadm/checkarray ] && [ $(date +%d) -le 7 ]; then /usr/share/mdadm/checkarray --cron --all --idle --quiet; fi
```
- the only problem I see here: the fake-hwclock, which is running 17 minutes after every hour. I'm not sure if it would be okay to set it at 6:25 every day like all the other. Perhabs I might change them all to 12:00 because I'm not at home when it happens. Also I'm not sure what the last one does, but it shouldn't care, because it only runs 0:57 every sunday
- Other then that I'm pretty happy, that I didnt need to fight with finding thousands of processes.
- I set up a tmpfs in etc/fstab beforehand, but as I didn't need it, I just disabled it again
- fatrace doesn't work on this kernel, because it doesn't support fanotify. Another method according to: https://linuxwiki.de/hdparm would be wrinting every event to /var/log/kern.log temporarly:
```
activate:
echo 1 > /proc/sys/vm/block_dump

deactivate:
echo 0 > /proc/sys/vm/block_dump
```
- but I didn't try it out, so I don't know

**_let me know if my steps are easy to understand_**

1000001101000, 2020-07-10:
interesting, I didn't realize that fake-hwclock was that noisy. 

I would hope the fan spinning up was a correlation but it might not be. When I've used fan control I typically end up with a config where it keeps adjusting up and down every minute and have to set the thresholds higher and higher till I find a suitable one. 

One thing to watch out for (if you haven't already) is swap usage. I think you want to adjust the swap settings to reduce disk usage for swap.

I think I've seen folks mount /var/log as a tmpfs which might also be an option.

kaymueller, 2020-07-11:
Good points there. I dont think swappiness had any negative effect for me, but I turned it down from 60 to 10. Also I changed all the cronjobs to be daily at around 12:00.

**move the fake-hwclock to be daily**
sudo mv /etc/cron.hourly/fake-hwclock /etc/cron.daily/

**change the start-time of the cronjobs**
sudo nano /etc/crontab
sudo systemctl restart cron.service

```
59  11  *  *  *  root  /etc/cron.daily/apt-compat
59  11  *  *  *  root  /etc/cron.daily/apt-show-versions
59  11  *  *  *  root  /etc/cron.daily/bsdmainutils
59  11  *  *  *  root  /etc/cron.daily/dpkg
59  11  *  *  *  root  /etc/cron.daily/fake-hwclock
59  11  *  *  *  root  /etc/cron.daily/logrotate
59  11  *  *  *  root  /etc/cron.daily/man-db
59  11  *  *  *  root  /etc/cron.daily/mdadm
59  11  *  *  *  root  /etc/cron.daily/ntp
59  11  *  *  *  root  /etc/cron.daily/passwd
```
**change swappiness**
cat /proc/sys/vm/swappiness 
> 60

sudo nano /etc/sysctl.conf
add: vm.swappiness = 10
sudo sysctl -p
>vm.swappiness = 10

free -lh
```
              total        used        free      shared  buff/cache   available
Mem:          238Mi        80Mi        66Mi       1,0Mi        92Mi       150Mi
Low:          238Mi       172Mi        66Mi
High:            0B          0B          0B
Swap:         975Mi          0B       975Mi
```

_I dont think swappiness would have done anything by now, but maybe later when the system is running for a long time or whatever. I'll let it run for now and see in some days how often it woke up and if it's related to my use or if it's random spinups. I also made a share using NFS and I put it in the fstab of my computer (without automounting). So as long as I don't click on that folder to mount it, it shouldn't wake up the drives -> fingers crossed_

kaymueller, 2020-08-08:
Short Update:
Have it running for a while now and it works pretty well. The speed with Debian is just amazing. When I copy larger files it gets up to 100 MB/minute.
The tinkering with the fans and drives is enough so I can let it run over night without waking me up, even tough it's 30°C/86°F in my room.

So it's not the answer to the question, but it's a somewhat good alternative.

I still have some very few spinups I can't connect to anything user related, but I also couldn't track it down really and at this point I think it's acceptable. There might be more spinups if you add more packages to it that might wake it up. I've done a somewhat minimalistic setup without any webinterface or such.