1000001101000/Debian_on_Buffalo #14: ls210 partitioning scheme

ville-h, 2019-07-19:
Whilst trying to install on ls210 it struck me I don't know which partitions I can change from the default:
SCSI1 (0,0,0) (sda) - 2.0 TB ATA WDC WD20EZRZ-22Z                                                                                    (1)      1.0 GB       ext3        primary
        458.2 kB       FREE SPACE
(2)      5.1 GB       ext3        primary
(3)    393.7 kB    K  biosgrub    primary
(4)     512.0 B                   primary
(5)      1.0 GB    F  swap        primary  swap
          65.0 kB       FREE SPACE
(6)      2.0 TB       xfs         primary
           8.4 GB       FREE SPACE
I believe (1) is the partition where the initrd.buffalo and uImage.buffalo reside. (2) seems to be where the original operating system from buffalo is installed, (4) is swap, (6) is presumably the "data" partition. Question is which ones of these do I need to leave as-is, which can I wipe clean of data and which can I remove completely and repartition as I see fit? What sort of partitioning scheme is recommended here?

1000001101000, 2019-07-19:
Good to hear from you!

I believe you are correct in how you've identified each of the existing partitions. You can change the partitions however you like, the only requirement is that the first partition on the first disk must be:

either a regular partition or a software raid1 array with metadata version 0.90 
formatted either ext2 or ext3
a reasonably small size. (I usually use 1GB, I haven't determined the exact upper limit)
contain the *.buffalo files required for boot. This is handled automatically if you specify it as /boot when you run the installer.

These requirements are all dictated by the version of uboot the device is running and how it is configured. The boot command being used is:
initrd=initrd.buffalo
kernel=uImage.buffalo
ext2load ide 0:1 0x1200000 /$kernel; ext2load ide 0:1 0x2600000 /$initrd

The full set of uboot parameters can be found at: https://buffalonas.miraheze.org/wiki/Linkstation_LS210D#U-Boot_env

ville-h, 2019-07-19:
I went with leaving the original (1) ext3 as-is and marking it for use as /boot, removed everything else and then created some additional partitions:
sda         8:0    0   1.8T  0 disk 
├─sda1      8:1    0 976.6M  0 part /boot
├─sda2      8:2    0   9.3G  0 part /
├─sda3      8:3    0   954M  0 part [SWAP]
└─sda4      8:4    0   1.8T  0 part /home
Seems to have worked out. Obviously boots alright and I can login and run some basic commands. Just in case someone else wonders. No claims how suitable the particular scheme is beyond "seems to work".

closed 2019-07-19 by ville-h

1000001101000, 2019-07-19:
looks pretty good to me. The only change I would make is to make the large partition /mnt/disk or something like that. that way you can keep your bulk storage separate from your use config files. It makes things easier if you start wanting to clone the disk to setup other devices or want to make separate system backup etc.