1000001101000/Debian_on_Buffalo #15: Not an issue - USB ethernet compatibility

blaudp79, 2019-07-27:
Hi,

I'm planning to buys an LS210D, due to the fact that there is a working linux firmware for it.
I'd like to use two ethernet ports, so I'm plannig to use it with an USB ethernet adapter.
This is the linux driver for the adapter that I'm planning to use with it:
https://github.com/rickhofstede/linksys_usb3gigv1_linux

What are my chances to get it working on an LS210D?

Thanks.

1000001101000, 2019-07-27:
Good to hear from you!

What do you plan to do with the second ethernet port?

one thing to keep in mind is that since the ls210d only has a USB 2.0 port you won't be able to get gigabit speed out of the adapter, it will be limited by the 480Mbps USB 2.0 max speed. 

It looks like the mainline kernel driver has had a lot of work done on it over the last few years and is now included in the default Debian kernel. I think if you use Debian Buster it should work just fine.

The link you posted looks like it hasn't been updated in several years, I wouldn't recommend using it with a modern kernel.

blaudp79, 2019-07-27:
Good Morning! 
Thanks for the quick reply. USB 2.0 will be more than enough. I'm going to use it as an FTP server for surveillance cameras. I don't have enough switch ports to connect the NAS to the network, so one camera will be connected via the USB Ethernet device directly to the NAS.

1000001101000, 2019-07-27:
sounds like a fun project.

1000001101000, 2019-08-15:
Did you get a chance to try it yet?

blaudp79, 2019-08-16:
Hi, Not yet, but I'm planning to some time next week. I'll post the results
once I've tried it.

blaudp79, 2019-08-23:
I've tried installing Debian on the device today, but I'm stuck at trying to copy the necessary files to the device. I tried it via acp_commander, but it does not "see" my device. I'm assuming that the newest firmware my device came with will not work with acp_commander anymore. I'm not using firewall, so that can't be the reason for no connection. This is the version of my firmware: 
LS210D (LS210DD58) Version 1.70-0.01

>Using random connID value = 829B93717991
Using target:	LS210DD58.lan/192.168.66.247
Starting authentication procedure...
Sending Discover packet...	
A SocketTimeoutException usually indicates bad firewall settings.
Check especially for *UDP* port 22936 and make sure that the connection to your LS is working.

1000001101000, 2019-08-23:
I would try generic things like rebooting the device to see if it starts showing up. You can also use acp_commander -f to have it search for it (more like send a broadcast and see what responds). 

I've had really good luck with acp_commander for pretty much every model and firmware version i've tried (with the weird exception of the Terastation-III). In that case I took a look at packet captures from tsupdater and could see it's using the same protocol but one of the fields was a different length. Trying to fix that is on my todo list but extremely low down it. 

I don't recall having trouble with the ls200 series but I honestly don't remember what I have and haven't tried with them. 

Typically I just recommend pulling the drive and transferring the files that way but I understand why you would want to avoid that with the ls210, it's a real pain to take apart.

let me know if rebooting/etc doesn't help and I'll try loading the firmware on mine and test acp commander.

blaudp79, 2019-08-23:
Neither rebooting or resetting has helped. I tried downgrading to 1.67, but the factory FW update utility does not allow downgrading. NAS Navigator2 sees my unit on the network, so I think the new v1.7 can not be used with acp_commander anymore. I will try to disassemble it.

blaudp79, 2019-08-23:
Hold on... I got some progress by specifying the MAC address. Now I see some responses but still faults out. Let me try a couple more options.

blaudp79, 2019-08-24:
OK... I did succeed with acp_commander at the end. It needed the -m parameter with the mac address. It threw some errors that it cannot change the IP, so I removed the -IP parameter after which there was no more error messages. However the I did not get telnet access, so I re-run both commands with -IP & -m parameters and even though I got error messages, telnet did become live.
I've copied the files & I rebooted the device. It booted up with live SSH access via port 23, so I tried to log in but I did not know un/pw, so I did some digging online and I've found out that it is installer/install. I've completed the installer process, but now I'm unable to login with either root or user access. The passwords I provided during the installation process don't work. Any recommendation?

1000001101000, 2019-08-24:
Sweet. I don’t know why but -ip seems to be required, and the errors are normal. I’m just impressed it works at all really.

I would guess that your username and password are still installer/install. If so that also means you didn’t set up your partitions properly.

I discussed the requirements briefly in 3-Run_Installer.txt. the installer process will generate new initrd.buffalo and uImage.buffalo files but for the device to boot from the, they need to end up in the same location as the old ones in the first partition.

this time have the installer format the first partition as ext3 and mount it as /boot, it should work as expected after that.

let me know if you run into trouble

blaudp79, 2019-08-24:
The first time I went with the fist option in the guided partitioning menu. This time when I chose the first option again, I got an error that it cant be done because probably the partition is too small or something. So I opted for the "entire disk" second option. I noticed that by default the boot partition is configured as ext2, so I changed that to ext3 and now I'm re-running the whole installer...
Partition #1 is ext3 and mounted as /boot
Partition #2 is ext4
Partition #5 is swap
I hope it goes through fine.

1000001101000, 2019-08-24:
That sounds right.

I wasn’t able to figure out how to remove the guided partitioning options when I set up the installer, i should probably look into it again sometime since manual partitioning is the only why to make this work.

blaudp79, 2019-08-24:
Here is the dmesg output for the USB ethernet adapter:

> [  892.832022] usb 1-1: new high-speed USB device number 2 using orion-ehci
[  892.993548] usb 1-1: New USB device found, idVendor=2357, idProduct=0601, bcd                       Device=30.00
[  893.001824] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[  893.009068] usb 1-1: Product: USB 10/100/1000 LAN
[  893.013828] usb 1-1: Manufacturer: TP-LINK
[  893.017973] usb 1-1: SerialNumber: 000001000000
[  893.152267] usbcore: registered new interface driver r8152
[  893.212155] usbcore: registered new interface driver cdc_ether
[  893.288069] usb 1-1: reset high-speed USB device number 2 using orion-ehci
[  893.492064] r8152 1-1:1.0 eth1: v1.09.9
[  893.558736] r8152 1-1:1.0 enx503eaadbd205: renamed from eth1

1000001101000, 2019-08-24:
Looks promising

blaudp79, 2019-08-24:
It did not work seamlessly right out of the box. I downloaded the R8153 (R8152) driver from realtek.com, moved it over to the drive via vsFTPd and installed it via these steps:

> tar xf r8152.53.56-2.12.0.tar
cd r8152.53.56-2.12.0/
apt-get update && sudo apt-get upgrade
apt-get install libelf-dev build-essential linux-headers-`uname -r`
make -j2
make install
depmod -a
nano /etc/modprobe.d/rtl_usb.conf (add this line: alias usb:v2357p0601d*dc*dsc*dp*ic*isc*ip*in* r8152)
depmod -a

I added these lines to the end of /etc/network/interfaces

> allow-hotplug enx503eaadbd205
 iface enx503eaadbd205 inet dhcp
 hwaddress ether 50:3e:aa:db:d2:05

I also had to configure environment variables properly because "make" couldn't run some of the required commands

blaudp79, 2019-08-27:
> Which Debian version did you install?

Buster