1000001101000/Debian_on_Buffalo #16: No Path To Host

Centurion83, 2019-08-02:
wget http://192.168.1.146:8000/initrd.buffalo
--2019-07-11 08:51:02--  http://192.168.1.146:8000/initrd.buffalo  

Gives no path to host error, I am replacing the IP with my own.

Centurion83, 2019-08-02:
vader@chaos:~$ wget http://192.168.1.146:8000/initrd.buffalo
--2019-08-01 17:46:03--  http://192.168.1.146:8000/initrd.buffalo
Connecting to 192.168.1.146:8000... failed: No route to host.
vader@chaos:~$ --2019-07-11 08:51:02--  http://192.168.1.146:8000/initrd.buffalo

1000001101000, 2019-08-02:
Good to hear from you!

I assume you are getting those commands from "99-Copy_example.txt" that I uploaded recently. The specific command you're showing needs to be run from the device itself after you enable telnet access and setup a simple webserver on your PC to temporarily host the files you downloaded from github. The commands to do that can also be found in "99-Copy_example.txt".

If you still have trouble post the output of all the commands when you run them and I can try to help.

Centurion83, 2019-08-02:
Thank you for your response ill give that a shot. I typically compile ROMs
this is a first with servers

1000001101000, 2019-08-22:
Any luck?

Centurion83, 2019-08-22:
Hello yes I did figure it out it was a dumb and simple mistake on my end
not yours haha. I had reinstalled Ubuntu and had not set paths for several
of my build environment tools or actually anything. My errors were throwing
because the front end was busy trying to pull an update. I have it
installed now and all is good. Great instructional and fantastic Dev work
on your end . Cheers

1000001101000, 2019-08-22:
Excellent!

Let me know if you run into any issues.

closed 2019-08-22 by 1000001101000