1000001101000/Debian_on_Buffalo #17: HDD leds definition on LS421D

vale-max, 2019-08-22:
In LS421D device tree the HDD leds are classified as hdd1 and hdd2 and the white function led has no default trigger defined.
Actually the two red HDD leds are marked as "Error" led and are hidden behind the front door.

To be consistent with the documentation the two HDD leds should be marked as hdd1err and hdd2err and the white function led have "disk-activity" trigger defined in the device tree

1000001101000, 2019-08-22:
At the time I wrote that I liked having that disk activity trigger in part because it made it easier to tell when the kernel took over and when the boot process finished. With the addition of devices that have dedicated leds for each drive it made less sense to me to set the trigger for the function led. likewise it would make it more difficult for someone to work with the other two function led colors. 

I'll take another look at the documentation and the device trees and make sure that the behavior matches the documentation better.

I'll look at the naming as well but I am hesitant to make any changes to those names in case someone has hardcoded the names in a script somewhere. 

That reminds me, I need to add some code to initialize the power led on the TS1400r. For the ts3000x devices I initialize it along with the LCD and for all the other non-LCD devices it's initialized in the DTB. Somehow I overlooked the TS1400r in all that.

vale-max, 2019-08-23:
I had suggested the led rename as I have seen other dts have the hddXerr names, and in the LS421D these leds are unmistakably error leds, but I agree with you that changing the names can break some existing script.

1000001101000, 2019-09-25:
I'm going to go ahead and close this one for now, I may make the changes you've suggested when Bullseye testing ramps up.

closed 2019-09-25 by 1000001101000