1000001101000/Debian_on_Buffalo #18: Effective usage of RTC clock in LS421D

vale-max, 2019-08-22:
The post-installation section suggests to build module RS5C372 to use the device RTC as it is not enabled natively in debian kernel.
However at boot time the module is enabled after the kernel tries to pick up the HW clock time, so system time is not updated until fake-hwclock or ntp come into play:
[    6.448247] hctosys: unable to open rtc device (rtc0)
[...]
[    7.926074] rtc_rs5c372: loading out-of-tree module taints kernel.
[    8.001860] rtc-rs5c372 0-0032: rs5c372a found, 24hr

ref: 
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=855203

I circumvented this problem by adding the following udev rule as suggested in [this forum post](https://www.raspberrypi.org/forums/viewtopic.php?t=209700)

KERNEL=="rtc0", RUN+="/sbin/hwclock --rtc=$root/$name --hctosys"

It may be worth mentioning it in the post-install steps doc.

1000001101000, 2019-08-22:
Good to hear from you!

I had not noticed that. Truthfully, I went several months without even noticing that the RTC wasn't supported by the Debian kernel because NTP was generally working well enough for me not to notice. Other than confirming that I could read/set the time in the RTC when the module was loaded. I haven't spent much time working with it. 

I will add that to the document when I have a moment. 

Thanks!

1000001101000, 2019-09-28:
I've added this to the wiki which I'm in the process of using to replace the text-file based instructions.

Thanks again!

closed 2019-09-28 by 1000001101000