1000001101000/Debian_on_Buffalo #21: Built-in ethernet on LD410d0401 randomly shuts down

RQWorldblender, 2019-09-07:
[kern.log](https://github.com/1000001101000/Debian_on_Buffalo/files/3586935/kern.log)
By having the ethernet randomly shut down on me, it causes significant downtime that can only be solved either by rebooting my machine, or disconnecting its power source and connecting it back again. My only workarounds are to use a USB ethernet adapter or a different computer with different ethernet chipset entirely.
I have installed Debian 10 Buster, with kernel version 4.19.0-5-armmp.

I had this happen at least twice to me already, and most workarounds I specified will either take too long, or give worse performance:
`
Sep  4 17:16:37 ls410d0401 kernel: [253058.158786] mvneta d0074000.ethernet eth0: Link is Down`
`
Sep  6 02:37:33 ls410d0401 kernel: [106025.969282] mvneta d0074000.ethernet eth0: Link is Down
`

1000001101000, 2019-09-07:
Good to hear from you!

I've never run into this personally (between ~15 devices using this same SoC/ethernet combo) but I've now heard from two different folks seeing the same thing on their ls441de. Here is the thread:
https://github.com/1000001101000/Debian_on_Buffalo/issues/8

They are looking at a few possible solutions but haven't had much luck yet. Unfortunately, I haven't been much help since I haven't been able to reproduce the issue at all. 

When I was looking at buffalo's source code last week I did notice that at one point auto-negotiation was disabled (it may have been re-enabled later). Could you give the following a try?

ethtool -s eth0 autoneg off speed 1000 duplex full

Also, did this happen when you ran the installer or only afterward?

RQWorldblender, 2019-09-07:
I have experienced these shutdowns hours after turning on my device, which does take a while to reproduce. There is nothing in the kernel log that indicates why the shutdowns happen, unfortunately.

1000001101000, 2019-09-07:
it would only persist until the next reboot, technically probably just until the next ifup.

RQWorldblender, 2019-09-07:
Well, I do have to be able to have my device running for days without me being at home since I am at school, so if these shutdowns happen again and I forget to run the command you showed me, I'm in big trouble, unless I use a USB ethernet adapter. Here, would it matter is it was 100MBit or 1GBit?

1000001101000, 2019-09-07:
if turning off auto negotiation were to solve the problem we would be able to make the change permanent. 

USB 2.0 has a top speed of ~480mbps, as such a Gigabit adapter wouldn't be able to run at full speed but it can still perform significantly faster that 100M

RQWorldblender, 2019-09-07:
Okay then; which one of the four situations should I do that will guarantee the lowest chances of me losing my internet connection to my device and be a cost-saving measure at the same time?
* Disable auto-negotiation on every bootup with a script, if I continue to use dhcpd
* Use NetworkManager to manage the built-in ethernet and change its settings to disable auto-negotiation
* Buy a USB 3 GBit ethernet adapter (not USB 2, as I can a USB 3 device for use with other computers), and attach it to a USB hub attached to the device
* Use a USB 2 100MBit ethernet adapter that I already have, and attach it to a USB hub attached to the device

I only have until the end of tomorrow to make a decision, as I don't always come back home, and a family member may not always be available to help me reset my device.

1000001101000, 2019-09-07:
It depends on what you plan to do with the device.

I don’t know if disabling auto negotiation will actually help so I wouldn’t depend on that.

If you do decide to get a usb3 GBE adapter you’ll want to see if it has good linux support and works on usb2, and possibly how much power it needs if the hub isn’t powered.

If 100M is fast enough for what you’re doing and you already have one that is likely a good solution.

RQWorldblender, 2019-09-07:
The primary current usage for my device is being an ownCloud server, but later on, I may install WordPress other webapps. They will work with 100MBit speeds, but performance will be better if I use GBit speeds.
I can add power to most of my hubs as I have power cables and adapters that work with them, so power isn't much of a problem.

1000001101000, 2019-09-07:
Sounds like a plan. Hopefully the folks on the other thread will find a solution for the built-in adapter so you won’t need it in the future.

RQWorldblender, 2019-09-07:
Actually, I do have a USB 3 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet adapter (TRENDnet brand) that does work on USB 2, but I'm saving that one as I found out that it does work with Nintendo Switches, even though this compatibility is unadvertised. Should I use this one with my LSD410d0401, or get another USB 3 GBit ethernet adapter anyways?

1000001101000, 2019-09-07:
It’s up to you. It might be a good idea to test that one and see if it works well, then you could get another if you decide 100M isn’t good enough

RQWorldblender, 2019-09-07:
There exist USB 2.0 GBit Ethernet adapters, and while they can't reach the full 1 GBit speed, I think this is the best compromise between performance and cost for me.

1000001101000, 2019-10-01:
on the other thread they have been able to get a stable connection by dropping to 100M:
`ethtool -s eth0 autoneg off speed 100 duplex full`

It might be worth trying when you're by the device again.

mailminhthang, 2020-04-24:
I face to same issue, ethernet suddenly shutdown, I don't think cause of speed or duplex. Let see my detail below. Do you have any way to resolve ?
Thanks

root@LS410D:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP MII FIBRE ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: g
        Wake-on: d
        Link detected: yes

1000001101000, 2020-04-24:
The following has made it stable for folks who have tried (though it limits to 100M):
`ethtool -s eth0 autoneg off speed 100 duplex full`

I've never been able to replicate this with my dozen or so devices, and the vast majority of other people have not seen this issue. 

I'm tempted to conclude the issue is switch/cable related but the original reporter tried several switches and cables which never seemed to change anything. Also, whatever is going on doesn't seem to affect the stock firmware (or at least it has a way of recovering).

mailminhthang, 2020-04-25:
I already try "ethtool -s eth0 autoneg off speed 100 duplex full" before, issue still remains.
I didn't face to this issue with Nas Buffalo firmware, I used this firmware long time before
Tks

1000001101000, 2020-04-25:
can you post the contents of your /etc/network/interfaces?

I'm looking at some documentation to find what options are available to make these connections more fault tolerant.

mailminhthang, 2020-04-26:
It's below (there is no file in  /etc/network/interfaces.d/) :

# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

source /etc/network/interfaces.d/*

# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
# allow-hotplug eth0
# iface eth0 inet dhcp
auto eth0
iface eth0 inet static
  address 192.168.1.141
  netmask 255.255.255.0
  gateway 192.168.1.1
  dns-nameservers 8.8.8.8 8.8.4.4

mailminhthang, 2020-05-10:
Hi Mr.1000001101000 
Do you have any new idea for this issue ?
Tks

1000001101000, 2020-05-11:
Not really. 

There seem to be two related problems:
1. The ethernet link on your device (and a few others) is dropping seemingly randomly.
2. When this happens it is not reconnecting for whatever reason.

Problem 1: seems like a cable/switch/hardware issue. The fact that it is so rare makes it seem like it has to be specific to individual network setups in someway. The first user to report it went to great lengths to try different cables/switches though.

Problem 2: the network does not reconnect after it drops. This may be related to the way it comes down.

It seems like the stock firmware has a different way of dealing with these problems but i haven't found what it is. Likewise, I haven't found a standard way to auto-reconnect (I assumed there would be a setting or package to do it, but i have not been able to find it).

I haven't been able to replicate the problem so i don't have a good way to test anything. About all I can recommend is swapping ethernet cables to see if it helps. You could also write a script that runs a few times a minutes and attempts to run ifup if it detects the network is down. 

If you wanted you could send your device to me, I could see if it does the same thing on my network or not. I could probably even compare it to my ls410, though I'm not entirely sure what to look for.

mailminhthang, 2020-05-13:
I see image files in your Buster folder are uploaded some days before. Is there any update ?

1000001101000, 2020-05-13:
I've got a workflow setup that generates new images whenever Debian publishes an update. In this case they released Debian 10.4 with kernel 4.19.0-9.

mailminhthang, 2020-05-13:
It is difficult for me to send my LS410d to you in this Covid-19 prior. I don't think cause of hardware or network because it runs smoothly with Nas Buffalo ROM before.
Can I ghost my hard disk with compress mode about 600MB and send ghost image to you ?

1000001101000, 2020-05-13:
Yeah, I'm not eager to have people sending me things both because of viruses and general hassle. 

Whatever is going the buffalo firmware has a way to handle it. It is most likely something different about their old 2.6 kernel. armada-370 support wasn't part of the mainline kernel back then and there are some differences in the drives etc. 

At the same time I've had about a dozen of these devices and never see this issue and the majority of other users also don't see it. It can't be as simple as a driver issue because that would affect all these devices. 

I don't see much use in trading disk images. We could both install using the latest installer images on our ls410 without installing additional packages which should give us basically identical rootfs and then see if they behave differently.  If your only becomes unstable after installing something that could also give us a clue.

mailminhthang, 2020-05-17:
So do you have a LS410d device in our side ?

1000001101000, 2020-05-17:
Yes, I do have one.

mailminhthang, 2020-05-17:
Could you please share me the image ghost file of your hard disk in LS410d, that runs stably, so I try to restore image into my hard disk
My hard disk is ssd 60GB, I prefer to use Terabyte to ghost and restore hard disk
Thanks a lot

1000001101000, 2020-05-17:
I can take an image of mine, do you have an sftp/dropbox/etc for me to upload it to?

mailminhthang, 2020-05-17:
I use a SSD, format absolute (non quick format) to fill all zero, then install Debian Buffalo, next I backup by terabyte using high level compress mode. So image file is just about near 400MB. So small
Please try like that and update file to your google drive and please share link to me
Tks a lot

1000001101000, 2020-05-17:
I do something similar for testing. I'll put something together.

1000001101000, 2020-05-17:
ugh, that was time consuming.

here's a link:
https://drive.google.com/open?id=1IkJtjmweKNmjc5pvEi7VZBj0D6KTbi8_

you can put it on a drive with:
`xzcat ls410d_buster.img.xz > /dev/sdx`

You'll need to fix the partition table (unless you put it on a 16gb drive like mine). I usually do this with gdisk, all you need to do is open the disk with gdisk and write out the partition table with "w"

1000001101000, 2020-05-17:
let me know when I can delete that.

mailminhthang, 2020-05-18:
Ok, let me try it, please keep it a day.
Thanks very much

mailminhthang, 2020-05-18:
Your google link can not be accessed

> 404. That’s an error.
> 
> The requested URL was not found on this server. That’s all we know.

1000001101000, 2020-05-18:
https://drive.google.com/file/d/1IkJtjmweKNmjc5pvEi7VZBj0D6KTbi8_/view?usp=drivesdk

mailminhthang, 2020-05-18:
I cloned to my ssd and it's running, please give me "root" password 
Thanks

1000001101000, 2020-05-18:
sorry. it's user:password. It's setup for sudo.

mailminhthang, 2020-05-18:
Ok, I can ssh now. Let me take time to monitor it
Thanks

mailminhthang, 2020-05-20:
I try your image in my LS410d, issue is the same as my image, Nas randomly shutdowns ethernet and halts after that (I don't see any action in log files in /var/log/ during this time).
I back to stock firmware of Buffalo, Nas and ethernet run smoothly day by day, so the reason is not by my hardware, something in Debian is not correct in my Nas.
Anyway thanks very much.

1000001101000, 2020-05-20:
yeah, seems like something specific to a subset of these devices (since most don't have the problem) or to certain networks/switches/routers (since no one has yet reported a network with one working and one not). 

whatever the issue is buffalo's firmware deals with it. 

I'm hoping someone with the problem will trade me for one of my working devices. That would allow us to confirm whether it's the device itself or something outside the device (router config etc). 

Assuming the device would also have problems on my network I could then work on some kind of fix.

mailminhthang, 2020-05-21:
ok, thanks so much

1000001101000, 2020-08-29:
@RQWorldblender @mailminhthang 
We've recently been provided a fix that has been tested for the ls410d and ls421de
https://github.com/1000001101000/Debian_on_Buffalo/issues/47#issuecomment-674566928

Let me know if you're interested in giving it a try.

mailminhthang, 2020-08-29:
Ok, I will try in any my free time.
I prefer the way to generate the image file and copy to hdd as we did with ls-chl-v2 before, it saves time.
Tks

1000001101000, 2020-08-29:
That's my favorite method now too.

mailminhthang, 2020-09-12:
Hi, do you have any new update please ?

1000001101000, 2020-09-12:
Sorry, I don't. I've been working on some other issues. This has led me to learn a little more about building initramfs that could be useful in setting this up, but I haven't done the actual work yet.

I'll try to get something ready for you to test in the next week or two.

mailminhthang, 2020-09-12:
Ok, thank you very much

1000001101000, 2020-11-26:
@mailminhthang I think we resolved this in the other thread, is it alright if I close this one?

closed 2021-06-29 by 1000001101000