1000001101000/Debian_on_Buffalo #22: RAID Setup

styelz, 2019-09-15:
Hi, I've done a few test installs using MDRAID and BTRFS (RAID10 with 4x4TB HDD). All work fine until I pull one of the drives to test breaking the RAID for recovery. At this stage I am unable to login to the NAS. Does anyone have any ideas about whats going wrong here?

I had a separate /boot and / partition as ext3 on all 4 disks, then created the raid on the remaining space on all 4 disks. I Tried pulling out a different disk each time with no difference in results.

1000001101000, 2019-09-15:
Can you post the output of lsblk when the device is working?

1000001101000, 2019-09-19:
I have some ideas about what you might be running into but would need the details of how your disks are set up to look into it. 

Let me know when you get a chance to pull some info for me. I'd like the output of:
lsblk
cat /etc/fstab
mdadm --detail --scan

styelz, 2019-09-21:
Sorry for the late reply, busy all week and had to rebuild the NAS.
I've setup 3x2TB drives MD-RAID5, latest Buster install for the LS441D.

I've tested pulling a drive out, and have the same issue.
When I boot up with a drive missing the network never comes up and the white LED on the NAS never stops flashing. Here's the output you asked for.

># lsblk
>NAME      MAJ:MIN RM  SIZE RO TYPE  MOUNTPOINT
sda         8:0    1  1.8T  0 disk  
├─sda1      8:1    1  243M  0 part  
└─sda2      8:2    1  1.8T  0 part  
  └─md0     9:0    0  3.7T  0 raid5 /
sdb         8:16   1  1.8T  0 disk  
├─sdb1      8:17   1  243M  0 part  /boot
└─sdb2      8:18   1  1.8T  0 part  
  └─md0     9:0    0  3.7T  0 raid5 /
sdc         8:32   1  1.8T  0 disk  
├─sdc1      8:33   1  243M  0 part  
└─sdc2      8:34   1  1.8T  0 part  
  └─md0     9:0    0  3.7T  0 raid5 /
mtdblock0  31:0    0  960K  1 disk  
mtdblock1  31:1    0   64K  1 disk  
mtdblock2  31:2    0   32M  0 disk  
mtdblock3  31:3    0  382M  0 disk  
mtdblock4  31:4    0   32M  0 disk  
mtdblock5  31:5    0   66M  0 disk 

># cat /etc/fstab | grep -v "^#"
>UUID=266bd2b1-4642-46b4-bc43-7faaf0a4018b /               ext4    errors=remount-ro 0       1
UUID=35cdd16a-78dc-4de0-a484-827df059f9ed /boot           ext2    defaults        0       2
/.swapfile	none	swap	defaults	0	0

># mdadm --detail --scan
>ARRAY /dev/md/0 metadata=1.2 name=debian:0 UUID=b8c2cf4c:a5189df1:7924b034:85be05ed

># blkid 
>/dev/sda1: UUID="35cdd16a-78dc-4de0-a484-827df059f9ed" TYPE="ext2" PARTUUID="64d92700-01"
/dev/sda2: UUID="b8c2cf4c-a518-9df1-7924-b03485be05ed" UUID_SUB="8b5daefa-25a8-b5fa-dd7b-f5f435910c9f" LABEL="debian:0" TYPE="linux_raid_member" PARTUUID="64d92700-02"
/dev/sdb1: UUID="35cdd16a-78dc-4de0-a484-827df059f9ed" TYPE="ext2" PARTUUID="f022999b-01"
/dev/sdb2: UUID="b8c2cf4c-a518-9df1-7924-b03485be05ed" UUID_SUB="d69cab36-70e9-8253-75dc-b197675ee612" LABEL="debian:0" TYPE="linux_raid_member" PARTUUID="f022999b-02"
/dev/sdc1: UUID="35cdd16a-78dc-4de0-a484-827df059f9ed" TYPE="ext2" PARTUUID="b91456c0-01"
/dev/sdc2: UUID="b8c2cf4c-a518-9df1-7924-b03485be05ed" UUID_SUB="1ef9d643-236d-b52f-5dd6-005bb092f677" LABEL="debian:0" TYPE="linux_raid_member" PARTUUID="b91456c0-02"
/dev/md0: UUID="266bd2b1-4642-46b4-bc43-7faaf0a4018b" TYPE="ext4"
/dev/mtdblock2: UUID="1475953182" TYPE="ubi"
/dev/mtdblock3: UUID="57361769" TYPE="ubi"
/dev/mtdblock4: UUID="1734171436" TYPE="ubi"

1000001101000, 2019-09-22:
Sorry for the delay, i’ve been traveling this weekend.

I have a few thoughts/suggestions. I believe added together they may account for the behavior you are seeing.

The first thing to note is that the system won’t finish booting if any of the filesystems in fstab aren’t present (unless you mark them optional somehow). It does offer you a recovery shell but since you don’t have access to the console.

This likely means that it’ll fail to finish booting if the disk with /boot is missing. I’d suggest changing /boot to be a raid1 array, this ensures /boot is always present and up to date for each disk. Remember that the array must ise metatdata version 0.90 for uboot to see it.

I’d also suggest moving to raid1 for your rootfs. I’ve read conflicting things about how degraded raid5 arrays are handled at boot, but it’s possible that it’s not automatically starting your array with a disk missing (presumably to protect you from a second disk failure)  which leads to boot failure. It might be possible to configure it to force starting degraded arrays automatically but I think you’re better off moving to raid1.

It’s not related to the failures but i’d suggest moving to a swap partition rather than a swap file. Although in theory raid5 would give you some performance improvement, I believe in this case the overhead introduced by the filesystem and raid parity will hurt performance somewhat. I’d recommend moving that to raid1 as well.

I’d also move /boot to ext3 as journaling will help protect you if something happens to the device during a kernel/initrd update.

styelz, 2019-09-23:
Hey, thanks for the info it was very useful. I used metadata=0.90 when manually creating the md partitions which allowed /boot to work. 

I created /boot (ext3), / (ext4) and swap as md raid1 partitions with hot spares on each and a raid5 on the remaining space for storage. 

I had an issue at first when removing a drive. This was caused by me manually creating the md partitions during install and making them md0, md1, md2 and md3. When I booted for the first time it changed md3 to md127 which caused the rebuild to fail during the test. I fixed that by adding the appropriate entries in /etc/mdadm/mdadm.conf using "mdadm --examine --scan". 

Heres the layout: 

>\# cat /etc/fstab
>UUID=7489dbdb-f75b-4b90-b890-990323554335 /               ext4    errors=remount-ro 0       1
>UUID=04ba48b3-4677-416c-96a7-6868092836dd /boot           ext3    defaults        0       2
>UUID=0e417102-5ee7-4fab-9908-13c34df61fb6 /storage        xfs     defaults        0       2
>UUID=06a7e0b5-b2d1-4c56-9601-2ce8fe8f7b05 none            swap    sw              0       0

>\# cat /etc/mdadm/mdadm.conf
>HOMEHOST <system>
>MAILADDR root
>ARRAY /dev/md0 UUID=57c45699:2c4b964f:9d4deba6:47ca997f
>   spares=1
>ARRAY /dev/md1 UUID=4aacb2af:3abd9508:9d4deba6:47ca997f
>   spares=1
>ARRAY /dev/md2 UUID=206950a4:9ac53357:9d4deba6:47ca997f
>   spares=1
>ARRAY /dev/md127 UUID=cd1f8fd8:f1a6cdf8:9d4deba6:47ca997f

>\# cat /proc/mdstat 
>Personalities : [raid1] [raid6] [raid5] [raid4] [linear] [multipath] [raid0] [raid10] 
>md0 : active raid1 sdc1[1] sdb1[0]
>      248768 blocks [2/2] [UU]
>      
>md1 : active raid1 sdc2[1] sdb2[0]
>      19530688 blocks [2/2] [UU]
>      
>md127 : active raid5 sda4[0] sdc4[2] sdb4[1]
>      3866467328 blocks level 5, 512k chunk, algorithm 2 [3/3] [UUU]
>      bitmap: 0/15 pages [0KB], 65536KB chunk
>
>md2 : active raid1 sda3[0] sdc3[2](S) sdb3[1]
>      499648 blocks [2/2] [UU]

1000001101000, 2019-09-23:
that was the other thing I was going to mention. Anytime you change fstab or mdadm.conf make sure to update your initramfs (`update-initramfs -u`). Otherwise it could use old values when assembling devices at boot.

1000001101000, 2019-09-25:
Let me know if you're still having trouble, otherwise I'll close out the issue in a couple days.

styelz, 2019-09-26:
Working fine now. Thanks!

closed 2019-09-26 by styelz