1000001101000/Debian_on_Buffalo #23: LS220D login issues after installation

squarecode, 2019-10-07:
First, thank you for all of your work to make the Buffalo NAS a lot more useful!
I'm facing a strange issue at the moment...
After completion of the installation on an LS220D the device restarts and after a few minutes the ssh entrance is back open. However I'm just not able to log in with the users that I've set up during the installation. I thought maybe I've mistyped something and went through the installation once again. But no... The device is still not allowing to let me in. Neither with the created user nor with the root user (I used the username 'test' and password 'test' to avoid some strange things happening on my second try). Also I've checked whether the installation process might have crashed or restarted by trying the installation login and my old original buffalo admin login. Currently I'm out of ideas... Maybe someone has a clou what I'm doing wrong. Thanks for your help!

1000001101000, 2019-10-07:
Good to hear from you!

I think you'll find the username/password are still "installer" and "install".

This happens when the first partition is not set up as /boot when choosing partitions in the installer. When /boot is set up properly the final step of the install process replaces the *.buffalo files with new ones that boot into your new system instead of back into the installer.

squarecode, 2019-10-07:
To further clarify my situation: I have already tried to log in with the installer credentials, still the same result... I did the installation in this order: Wipe the 2 hdds from all partitions and setup a msdos partition table. After that I installed the drives in the unit, switched it on and used the factory reset to let the device init the drives (default RAID 1 as I've later seen in the partition menu of the debian installer). After that I copied the installer files to one of the disks and then copied them after setting permissions to the boot directory. Rebooting the device brought up the installer. I followed the installation process up to the partitionizer. There I switched it to manual mode and set the first partition (RAID0) to /boot following by setting root and swap partition in the same manner to the matching RAID partitions 1 and 2. I also enabled to wipe the partitions during installation as the installer suggested this. The installation took on and after the sceduled reboot the device takes a few minutes (however faster than the buffalo image) and then is back up on the same ip as before and ssh is open. Now trying to log in with either the root account or the non root user set up during the installation both fail. Also the installer install combination does not work. And also the admin login used during the buffalo setup does not log in either. I hope that does clarify it a little bit...

1000001101000, 2019-10-07:
hmm. I get people booting back into the installer fairly often (hopefully less with the new documentation). 

I have never seen someone's credentials fail and have it not be the installer starting again. One thing to note is that by default ssh root logins are disabled so that wouldn't work. 

One thing I have noticed is that for some reason on other architectures I cannot get the network console to start unless I hard-code the keyboard map though for this installer (armhf) it starts without me doing so. i could imagine it somehow translating the password entered during install through a different keymap than the one interpreted later by ssh. Are you using a keymap other than "US"?

I don't see how that would actually make a difference but it's the only thing I can think of that might affect a password.

1000001101000, 2019-10-07:
When you try again, could you take a screenshot of your partition screen? 
also let me know what locale/language settings you are using.
also, are you using Buster or Stretch?

squarecode, 2019-10-08:
Thank you for your fast response! I tried it with Buster. For the locales I used german as system language and the US UTF8 keymap, as the german one was not available. I'll try all default locales and do some screenshots before and after my changes to the partitionizer next.

squarecode, 2019-10-08:
Interestingly default locales did the trick... Not sure why though. If I use 'test' as the user and also 'test' as the password there is no difference between the german and the us layout. Thats why I used this during my tests. I will now do it again with default locales but matching users and so on...

Here are my screenshots
Before:
[image]
After
[image]

1000001101000, 2019-10-08:
I'm glad to hear that it's working. 

I've set up similar installers for amd64, armel and armhf, they all behave slightly differently and require different menus to be hardcoded before starting the ssh server. I think this is the only one where I didn't have to hardcode the locale to make it work. If it's causing problems I could hardcode this one to us/en too.

I know people from all over have done installs, some have even sent me screenshots showing they weren't using the english locale. So far this is the only report of this issue.

My hope has been to find a way to re-enable that option for all of them since I'd prefer have better language support if at all possible. Hopefully it won't also start triggering password problems for others. 

Let me know if you run into any other issues.

closed 2019-10-08 by 1000001101000