1000001101000/Debian_on_Buffalo #24: Installing on LS-WVL

alvinhochun, 2019-10-11:
I'm hoping to try this on a LS-WVL. The mainline kernel has the device tree (https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/kirkwood-linkstation-lswvl.dts). Is it only a matter of building the dtb file and following the "hard way" in https://github.com/1000001101000/Debian_on_Buffalo/blob/master/1-Obtain%20install%20files with the provided armel image? I've already got ssh root access on the LS.

1000001101000, 2019-10-11:
Good to hear from you. It’s even easier than that, Debian provides installer images for the kirkwood-based linkstations including the LS-WVL. See:

http://ftp.nl.debian.org/debian/dists/buster/main/installer-armel/current/images/kirkwood/network-console/buffalo/

alvinhochun, 2019-10-11:
Oh cool. Looks like my quick web search had failed me. I don't know why I found this repo but nothing mentioning the official images. I'll give it a try and see how it goes. Thanks.

I think you should mention that in the readme file too.

1000001101000, 2019-10-11:
For whatever reason those installers don't come up very well in searches, I spent an embarrassing amount of time trying to follow old blog posts/etc for LS-QVL before I ran across information about the installer images. 

I've tried to improve that somewhat by putting links here and on the wiki:
https://buffalonas.miraheze.org/wiki/Main_Page

I think you're right, at some point  I think I'll add a more prominent mention in the readme. Something like "check here if your device isn't listed"

alvinhochun, 2019-10-11:
It worked after all, though it wasn't uneventful. Somehow I ended up with a corrupted `uImage.buffalo` file (I wget'ed it directly from within the LS and didn't verify the checksums) and had to plug the drive into a PC to fix it. Thanks again for the help!

closed 2019-10-11 by alvinhochun

1000001101000, 2019-10-11:
fantastic!

I don't think I've ever corrupted a download that way but the wget in the stock firmware is a pain just due to its age. It would be much easier if folks could just download the installer straight form github to their device but github is HTTPS only and the on-board wget is too old to support modern SSL/TLS.

Let me know if you run into anything of interest in the future, odds are anything affecting the LS-WVL will affect the other devices and vice versa.

alvinhochun, 2019-10-11:
Yeah, I found out that the power switch doesn't work so I hacked together a script to handle that. Wondering if there's a better way to do it...

1000001101000, 2019-10-11:
I have some notes on buttons on the wiki tab. I haven't done much with them personally though.

What happens if you try to shutdown/reboot the device from the command line? I've spent more time than I care to admit trying to figure out how to make these devices shutdown/restart as expected. I've figured out most of them but I can't make the TS-XEL reboot and I believe the TS1400 won't shut down. I'm curious if I should be looking into the LS-WVL etc as well.

alvinhochun, 2019-10-12:
The LS-WVL has a physical slide switch with three positions - OFF/ON/AUTO. The driver `restart-poweroff` performs poweroff by doing a restart, so what I think is that uboot checks the power switch on start, and if it is at the OFF position it goes into deep sleep and wait for a GPIO interrupt to wake it up. This setup should work well for me for now so I don't mind if I cannot directly power off the LS in software.

I'm guessing the AUTO position probably triggers some WOL handling routine in uboot. Perhaps this position allows both software reboot and poweroff to WOL standby. I'm not sure if I want to dig deep into this, but since I just found out that Buffalo had released the uboot source code I might even check those out... I don't yet have a need for this though.

I will be putting my files on https://github.com/alvinhochun/kirkwood-linkstation. (This is getting really off-topic though, perhaps these should be in a separate issue. I always feel uncomfortable with using an issue tracker for unfocused discussions.)

1000001101000, 2019-10-12:
Sounds like you've got a handle on it. I haven't looked at the kirkwood devices specifically yet but I have dug through the buffalo source a bit and have experimented with all the devices I'm supporting in this project. I'm aware of the following mechanisms for shutdown/restart on these devices:

Simple power switch:
It looks like the ls200 series uboot just checks the position of the on-off switch to decide whether to reboot or shutdown.

WOL/LED interrupt:
The ls400 series decide whether to reboot/shutdown based on the state of the WOL line from the ethernet chip (on the ls441 it's tied to the same gpio as the power button). I've included a script in my installer that runs at shutdown for these devices to set it appropriately depending on whether a shutdown or restart was called.

power management via microcontroller:
all terastations except the ts1200d and some of the orion5x based linkstations have a microcontroller on board that can poweroff the device (along with controlling the lcd display and other functions). For most you poweroff the device by sending it a special command at the end of the shutdown sequence. I've included a script for that too though on the ts1400 it doesn't seem to work

magic values passed to RTC:
The buffalo source code has lots of references to setting "magic" values to one of the alarm registers of the RTC on certain devices to signal power states including shutdown/restart. I think I tried this with the ts1400 at one point but didn't succeed. I don't know which devices actually use this method. 

Currently the ts-xel just turns off when I try to reboot it, once I finish re-working some stuff in the installer(s) I'm planning to take a closer look at it. Since it's a kirkwood device it might end up being the same mechanism as the ls-wvl. Let me know what you figure out!

1000001101000, 2019-10-12:
I share your hesitation on using issues as IM clients though it's the main way folks have of contacting me currently. I should probably work on that too.