1000001101000/Debian_on_Buffalo #25: LS220D - booting an installation image after Debian installation

domidx, 2019-10-14:
Hi

I have Buffalo LS220D and tried to install Debian using your instructions.

Install went well but after reboot when I ssh to the box it does not accept my root password. I used only one letter password for simplicity. During installation I also created another user account (one letter password as well) and tried to log in using this user but also unsuccessful. 
Then I tried log in using installation user - and it works but of course I was displayed with the installation menu :-( 
So therefore I think that my Buffalo boots still using your installation image rather than the installed Debian.

Please can you help and advise what I missed and if there is a way to resolve?

Thanks a lot.

1000001101000, 2019-10-14:
Good to hear from you!

This happens when the boot partition is not specified as /boot in the partitioning menu. One of the final steps of the installation generates new *.buffalo files which replace the installer files but it only works if the boot partition has been specified properly.

domidx, 2019-10-14:
Hmmm, I should read manual properly next time because you have it there..

So sorry for question that is answered in instructions..

Thanks a lot for your reply, let's reinstall :-)

1000001101000, 2019-10-14:
I did the same thing in some of my early install attempts. It's the most confusing part of the process, I'm still looking for ways to make it more clear.

I've even started experimenting with adjusting the installer to make it harder to do wrong but it's hard to do that without taking away legitimate installer options too.

1000001101000, 2019-10-17:
Did it work?

1000001101000, 2019-10-18:
fantastic!

let me know if you run into any more issues.

closed 2019-10-18 by 1000001101000