1000001101000/Debian_on_Buffalo #26: update debian stretch to Buster

amroth3003, 2019-11-14:
hi, copied files DTB in your source Debian_on_Buffalo-master\Buster\device_trees to /etc/flash-kernel/dtbs, but the version not change

[image]

1000001101000, 2019-11-15:
***warning, putting the buster(4.19) device tree in place for your stretch(4.9) device could render it unable to boot. If you do anything that triggers flash-kernel such as update the kernel it could generate a new uImage.buffalo with a broken kernel due to the version mismatch. only put it in place when you are ready to upgrade to the new 4.19+ kernel.

The note about upgrades is just a reminder that if you choose to use the "dist-upgrade" feature to move to a new version you'll need to install the device tree for the new version before you install the new kernel. This would also apply if you chose to install a "backports" kernel. It's basically just a reminder that the device trees for the 4.9 and 4.19 kernels are not interchangeable.

The documentation for this type of upgrade can be found here:
https://www.debian.org/releases/stable/armhf/release-notes/ch-upgrading.en.html

A short guide for performing such an upgrade can be found here:
https://linuxconfig.org/how-to-upgrade-debian-9-stretch-to-debian-10-buster

Let me know if you have any questions.

amroth3003, 2019-11-15:
Solved, after copy files and execute apt-get dist-upgrade, execute this commands for normal upgrade debian
1.- cp /etc/apt/sources.list /etc/apt/sources.list_backup
2.- sed -i 's/stretch/buster/g' /etc/apt/sources.list
3.- apt-get update
4.- apt list --upgradable (Check packages for upgrade to system)
5.- apt-get upgrade
6.- apt-get dist-upgrade
7.- lsb_release -a

Result
[image]

closed 2019-11-15 by amroth3003

1000001101000, 2019-11-15:
Excellent, I’ll probably add some more noted about that process to the wiki at some point.

Does it show that you’re running kernel 4.19 now? (uname -a)

amroth3003, 2019-11-15:
yes 

Linux amroth 4.19.0-6-armmp #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11) armv7l GNU/Linux

kinyat, 2021-03-16:
Just a note for people upgrading to buster. There is a bug currently in the openjdk 11.0.9 which prevent ca-certificate-java in be installed. (see: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=960153)

I have everything upgraded except for ca-certificate-java so if you want to have a clean upgrade experience you might want to wait for openjdk to release 11.0.11 to stable in debian.

1000001101000, 2021-03-16:
hmm, interesting. If I read the bug correctly something is failing because it is assuming it's always running on a multi-core system...or at least an SMP enabled setup. 

It reminds me of a couple of bugs I've worked on in the last year. One was kind of similar in that IRQs were being divided between 2 cores on a SoC... but that failed for the single core variant used by that model Terastation. The other was some really old C++ code that did some neat things to ensure in could run on either X86 or PowerPC (Big vs Small endian) but would segfault on modern systems because it assumed it would always run on 32-bit architecture. 

It sounds like they should have the fixed package out soon. Thanks for the heads up!