1000001101000/Debian_on_Buffalo #27: Always boots to installer

darrentaft, 2019-12-10:
Apologies - I'm a bit of a noob in this area.  I followed the instructions step by step, but then I was left to fend for myself at the Debian install.  When it asked me to partition the disk, I allowed it to use some free space at the end of the disk (I didn't want to lose all the content on my NAS) where it created 3 new partitions.
After install, it's rebooted back into the installer.  I can only SSH in with installer/install.

Did I choose the correct option for partitions?
How do I now get my LS210 to boot to the newly-installed OS?

Cheers

1000001101000, 2019-12-10:
Good to hear from you!

I remember creating notes about that but don’t actually see them on the wiki, i’ll need to fix that.

For the process to work the first partition needs to be specified as ext3 and /boot in the partition menu. Without doing that the installer never overwrites the installer boot files and the device just keeps booting from them.

darrentaft, 2019-12-10:
Any chance you'd be able to give any guidance on how to fix that now?  I'm happy to go through setup again if needed - what would I need to do differently at the partition stage?
Or can I use the installer console to overwrite the installer boot partition with the new boot partition that setup created?

1000001101000, 2019-12-10:
While it’s possible to fix it in place from the installer console I’d recommend just running through the installer again.

Once you set up your partitions, post a screenshot of your partition screen before saving them. I can take a look and make sure it looks okay before you proceed

darrentaft, 2019-12-10:
Great, thanks.  See this image:

[image]

It created the last 3 partitions as part of setup last time.

It offers me the following "use as" options for each partition.  Which partitions do I need to select, and for each of them, which of these options should I select for it:

[image]

Also, all partitions currently show "Bootable flag: off".

Cheers,

1000001101000, 2019-12-10:
The bootable flag isn’t used in these systems.

Partition one should be used as ext3, it will then enable the mount point option. Select /boot.

I’d recommend using partition 2 as ext4 and “/“

I’d leave 5 as swap.

I would delete those new partitions 7+

darrentaft, 2019-12-10:
Perfect, thanks! That's me up and running.

closed 2019-12-10 by darrentaft