1000001101000/Debian_on_Buffalo #28: Installer Stuck - Setting debconf\language to en

Jonathan-Garber, 2019-12-24:
I might be missing something here but I ssh into installer@IP with password install. The session connects but I get stuck at

'''debconf: setting debconf/language to en'''

I can halt this and get to a busybox prompt but there isn't really anywhere to go from there.
Buffalo TS-XE4.0 -- using initrd and buffalo-ff... images from the repo. So I believe I have everything right here. Any suggetions/ideas?

1000001101000, 2019-12-24:
Good to hear from you!

I haven't seen that before. I did just generate new images using a different method (I'm trying to automate more of it). Either debian's latest installer image or some change to my build process might have introduced some sort of issue. 

I'll give it a try on my tsxl and see if I can figure out what's going on. 

In the meantime, I've attached the last version I built manually if you'd like to try again.

[tsxl_install.zip](https://github.com/1000001101000/Debian_on_Buffalo/files/3998734/tsxl_install.zip)

Jonathan-Garber, 2019-12-24:
I just tried it on a different unit this one is a model TS-XL. I have the same error. Maybe I am messing up somewhere? I used your tsxl_install.zip

I made a backup of initrd.buffalo and uImage.buffalo
I put your initrd.buffalo and uImage.buffalo in /boot
rebooted the device
SSHed into it with installer and I am back to stuck on that same line as before.

I have to be missing something. This two separate units that have the same result.

1000001101000, 2019-12-24:
Sorry about that, for some reason I interpreted TS-XE4.0 to mean "TS-XL" instead of "TS-XEL". which probably made them less than helpful. I'm more surprised they didn't work on the TS-XL then.

I'll run some tests as soon as I can.

Jonathan-Garber, 2019-12-24:
Hang tight, I think I found the problem. I was opening SSH from windows command prompt as opposed to a real ssh client. When I opened SSH via Putty everything loaded up as expected. Gotta love Microsoft. I should have known not to be lazy and just use Putty from the start.

1000001101000, 2019-12-24:
hmm, I hadn't thought of that. That would explain how it could fail so early but still manage to give you a working ssh session. 

Let me know how the rest of your installs go, I've worked with a few folks with ts-xl devices in the past but you're the first with a ts-xel to contact me.

Jonathan-Garber, 2019-12-24:
Everything seems to be going ok with the TS-XL and the images you provided. I will work more on the TS-XEL after the holiday and see how it goes.

Jonathan-Garber, 2019-12-26:
I'm still having no luck here. I am still working on the TS-XL. Maybe I am misunderstanding something.
I copied initrd.buffalo and uImage.buffalo over to the /boot then rebooted the unit.

It comes up with the debian installer and forces me into a partition manager. I have tried telling it to use the existing array. I tried cleaning everything and making a new raid5 array etc. Everything I have tried, the unit just fails to boot. No Array Info error. Any pointers?

1000001101000, 2019-12-27:
I think i messed up the installer instructions when I moved them to the wiki. I started working on a more comprehensive guide to partitioning now that I sorted out the issues the installer from creating arrays that the device could boot. Unfortunately I keep getting distracted by other things. I think I'll put up some text to make it more clear until I get around to making the more comprehensive guide.

The key to making these devices bootable is that /boot has to be a separate filesystem which has to be located on the first partition of the drive(s) and has to be formatted as ext3. It can be part of a raid 1 array as long as it uses metadata version 0.90 or 1.0 and meets the previous requirements. 

Here is a screenshot from my eventual partitioning guide showing the layout I prefer.
https://github.com/1000001101000/Debian_on_Buffalo/blob/master/.images/17-all_volumes.png

Jonathan-Garber, 2019-12-27:
What about a Raid5 array?

1000001101000, 2019-12-27:
/boot needs to be RAID 1 (it's the only thing that uboot understands). Typically Buffalo and other NAS vendors use RAID 1 for /boot / and swap which makes it so any one disk can run the whole OS if needed. Typically these are each relatively small ~5GB for rootfs, 1GB each for /boot and swap. You can then create a large partition out of the rest of each drive and then use it to create a raid5 array.

The only hard requirement is for /boot everything else can be configured however you want.

Jonathan-Garber, 2019-12-28:
so if I am understanding correctly I have to keep the main OS under RAID1 so  any drive can boot the device.

RAID 1 for
1.0 GB - boot
1.0 GB - swap
5.0 GB - root

Then
RAID 5 - DATA  at /mnt/array

1000001101000, 2019-12-28:
exactly

Jonathan-Garber, 2019-12-28:
Yup, that worked. TS-XL 4-Bay unit running Debian 10.2. I have a handful of these just laying around, not sure what to even really do now that I have Debian running on them. Not sure I have a real use for it. I just did it for fun and knowledge lol.

1000001101000, 2019-12-28:
Yeah, that’s pretty much the same reason I created the installer.

One of the folks I worked with on this model installed OMV and was pretty happy with it. They work reasonably well provided you don’t need much processing power.

1000001101000, 2020-01-05:
I added a note to the wiki, I'm sure I'll get around to a more comprehensive guide someday.

were both installs successful? is there anything else you need before I close the issue?

Jonathan-Garber, 2020-01-06:
I haven't gone back to working on the TS-XEL but on the TX-XL no issues so far. I will close this issue.

closed 2020-01-06 by Jonathan-Garber