1000001101000/Debian_on_Buffalo #29: Incorrect Password?

dlettiere, 2019-12-31:
I'm trying to get the latest installer running on my LS441D but the password listing in the wiki ('password') is not being accepted.

1000001101000, 2019-12-31:
Hi!

the username is installer and the password is install.

dlettiere, 2019-12-31:
That is what I am trying, have tried all the common ones, 'password', 'admin', 'install', 'installer', various capitalization of those as well as my own root password that I used to connect to ssh shell before attempting the install.

[image]

1000001101000, 2019-12-31:
I went ahead and tried it on my device to be sure, the credentials for the installer are installer:install

my best guess is that the device isn't actually running the installer yet and the ssh server that is responding is actually the one from the stock firmware, possibly because the device hasn't been rebooted yet or the stock firmware files are still in place for some reason.

can you give me more information about what you have done so far?

1000001101000, 2020-01-05:
any luck?

Jonathan-Garber, 2020-01-06:
I would also recommend using putty as I have encountered many errors working with ssh in windows command. Putty was the only way I got things to work on my TX-XL and TS-XEL units. So just a heads up in case you run into further issues.

dlettiere, 2020-01-08:
Sorry for the delay, holidays were busy!! I finally got it working, did a full factory reset of the device and it is currently running the installer!

Thanks for all your hard work on this!

closed 2020-01-08 by dlettiere

1000001101000, 2020-01-08:
Excellent!