1000001101000/Debian_on_Buffalo #30: Is it possible to power on/off schedully?

higeon, 2020-01-06:
Is it possible to make the debian on my linkstation power on/off schedully as the stock firmware?

1000001101000, 2020-01-06:
Good to hear from you!

I tried getting an LS441de to suspend a while back which seemed to work but it wouldn't let me use the power button/WOL interrupt to wake it up for what seemed to be an issue with the interrupt controller driver. I figured I would try again with a model that actually supports wake-on-lan someday but haven't gotten around to it.

I assume the scheduled on/off uses the RTC which might have better luck. Infact seeing if I could detect/use the rtc interrupt has been on my list of things to look at for a while now.

What model is your device?

higeon, 2020-01-14:
Sorry for my poor english and 
thank you for your reply message!

I am in Japan and have a Japanese version of LS420D0202 (No nano flash and usb 3.0)
I also found the qcontrol package for Qnap devices with debian. It seems do scheduled power on/off.
 　→　http://www.cyrius.com/debian/kirkwood/qnap/ts-219/tips/

Thank you!

1000001101000, 2020-01-14:
I spent some time experimenting with a TS1200D (same hardware as LS420D) last week. I haven't gotten it to work so far but have learned the following:

I tried using the wakealarm method described in that qnap document
-first I needed to build/install the driver for the rtc-rs5c372 (instructions for that are on the wiki)
-I then had to add the wakeup-source property to the dts to enable wakealarm for the device (I'll add that to the published tree if we get this working)
-attempting to set the alarm time via /sys/class/rtc/rtc0/wakealarm threw an error. 
-I set one manually using i2cset and tried suspending the device but it did not wake up.

I think it may still be possible to make it work with more effort, the next steps would be:

-confirm the SoC can be woken up from suspend via a different method to confirm it works, perhaps testing with a usb keyboard and all the different available suspend types
-look at the documentation for the rs5c372 and see if the alarm has to be enabled rather than just set the time if done manually
-look at buffalo's gpl code and confirm how they are accomplishing it.