1000001101000/Debian_on_Buffalo #31: (LS220D) Can't boot up if root and swap is encoded in setup

Kulturnilpferd, 2020-01-08:
Hey there,
first of all thank you soooo much for these awesome software it works really great, now I can spindown my drives can close unused ports and can use it only as SMB server :D
Because you can easily pull out the drives I just want a bit extra safety, so I tested the whole day to set up an encoded drive but it didn't work:
In the Partition Setup I have set up the /boot as not encoded drive as first drive.
then i have cerated a Raid 0 Array
After these I created a 1000MB encoded disk
and a 4TB encoded disk
after these i enabled encoding and set the 1000MB to Swap and the 4TB for the root directory.
So the /boot is not on the raid and is not encoded. 
So it should boot but there is some trouble with the encoding due the linkstation just blinking white (endless). If there was an error with the /boot the Linkstation should blink red 7 times so I guess the boot partition is fine but it can't get go further due the encoding.
Sadly this thing doesn't have a displayport to debug the startup...

1000001101000, 2020-01-08:
Good to hear from you!

I've never done that myself but i read a post about it a long time ago. It sounds to me like you're successfully booting from the kernel/initrd but it's stuck waiting for a password or something to mount the rootfs.

I ran into a lot of issues like this when first getting these devices to work with the kernel provided by Debian and have a pretty good idea what you need to get this working.

This post provides some details about what is needed:
https://unix.stackexchange.com/questions/5017/ssh-to-decrypt-encrypted-lvm-during-headless-server-boot

In a normal setup the device boots into the initrd which has some kernel modules and scripts it uses to initialize the sata ports, locate the filesystem containing the rootfs and mount it and then start the system. If you want to use an encrypted rootfs you need to provide the key/password at boot so that the system can mount it. 

In the example above they connect to it over ssh at startup and type in the password. In addition to setting up dropbear ssh you'll need to ensure the kernel modules are loaded for the network adapter. 
By default the initrd's are generated with as few modules as possible to keep the size small (required by some older devices but not the ls200). I believe you can accomplish that part by changing the entry in /etc/initramfs-tools/conf.d/modules to "MODULES=most"

If you can get it to work it would be a good thing to add to the wiki though it will be hard without a serial console. If you get stuck I can give it a try on a device with a serial console (fortunately buffalo included a read-only one on the TS3400).

Kulturnilpferd, 2020-01-08:
Awesome thank you for your quick reply <3
A few minutes ago I have managed after a few hours to get it to work:
I changed some of my partitions so that i have a seperate Raid with all my data 4GB (encoded) the swap partition 1GB (encoded) and a 10GB root partition (not encoded). With these setup it boots like a charm.
I think at these moment this is the best and fastest solution to get a secure NAS environment.
I will now log onto it and install samba and so on, if I have any trouble or problems I'll let you know =)

Kulturnilpferd, 2020-01-08:
Sadly the led is now not flashing anymore its just a white light without flashing but I can't log on. 
Ping is also negative, have scanned the whole network - nothing found. 
I will wait a few minutes if it doesn't get pinged or ssh'd I will install the plain system without encoding and format the raid partition after the debian installation routine. 
With these I should have a better view of things happen...

1000001101000, 2020-01-08:
That seems like a good compromise. Most of the complexity goes away if you don't encrypt your rootfs, though depending on why your encrypting having unencrypted data in log files etc could be a concern.

I would think that if you were able to boot once in the final configuration you would be golden. That said, if the device gets far enough to switch the led solid it means it got as far as loading the gpio-led module which typically means it got pretty far. My guess is that you made a change to fstab or mdadm.conf that is causing a volume to fail to mount at startup. If it can't mount one of your volumes it will display a warning and give you the options to reboot or enter a recovery shell.  With no terminal attached you can't see it so it ends up waiting forever. I've done the same thing many times.

I suppose it could be waiting for you to enter your decryption password too, I'm not familiar with that process yet. 

I believe you can get past that by configuring it in fstab with the "optional" option or something similar which tells the system it's okay to keep booting if it fails to mount that volume.

Kulturnilpferd, 2020-01-09:
Hey there,
with a bit of testing and fiddeling around I can tell that encoding a LS220D is a bunch of crap because the CPU can't handle the encoding properly. With Stock Firmware you get max. 25MB/s with an encoded drive I get max 8MB/s CPU always at 100% when something is written or read. 
So today I will install debian but not with any encoding just the plain debian thing with a samba server. Hopefully it's a bit faster then :D

1000001101000, 2020-01-09:
That make sense, anything CPU/RAM intensive will be a bottleneck on these devices. 

I believe the SoC has hardware accelerated crypto which you can see initializing in the dmesg output. I wonder if it is taking advantage of that or not.

Kulturnilpferd, 2020-01-09:
My knowledge about drivers and so on in linux are not so great, I doesn't even know how you could check if the SoC's encoding is used.
I am very pleased with the non encoded solution, now i have 20-25MB/s write speed without encoding.
Just one thing I've got to mention is that there is a lot of outgoing traffic at iftop on a plain installed system. I tink this is a lot of debian things like updater or something, even a tor connection want to be established, very strange thing in my eyes...
I want to store sensitive data from my studies to it so I set the NAS to a static ip without using a gateway. Then I closed all ports except SSH and SMB.
With these it fits my personal security desires beautifully ^^
In future I will buy a bigger Buffalo NAS with better CPU then the encoding should work also.
I'll let you know when I have another one and it's working =)

1000001101000, 2020-01-09:
yeah, I know next to nothing about about disk encryption and hardware acceleration currently. might be an interesting topic for another day.

Kulturnilpferd, 2020-01-10:
I have found an article after I gogled a bit: 
[https://www.anandtech.com/show/8264/seagate-armada-370-nas-4bay-review/7](url)
The Marvell ARMADA 370 will do it's job very slowly. So with these information there are only two options at all slow and secure or fast and no encoding.
In future I will also test if my two ASIC Coin Miners can connect via an USB-Hub to that little thing. Would be awesome, because that little NAS doesn't draw much power.

1000001101000, 2020-05-21:
Cleaning up old issues, let me know if you need anything else.

closed 2020-05-21 by 1000001101000