1000001101000/Debian_on_Buffalo #32: Reinstallation and other questions

samueleforconi, 2020-01-18:
Hi,
first of all thank you for this fantastic porting that you have done to use a Debian environment on Buffalo NAS.
I have installed the system on a Buffalo LS220D device, installation completed successfully and everything seems to work.
I have trouble using OpenMediaVault system, the RAID array isn't managed corrrectly from OMV and when I try to edit the fstab the system always crashes (ssh access is no more available)  and I'm forced to re-edit the fstab manually connecting the hard drives to my PC and re-editing... 
My question is:
1. Have you any experience in running OMV on these devices?
2. how should I edit the fstab to not cause system crash?
3. if I want to re-install Debian system from scratch, should I just copy the initial initrd.buffalo and uImage.buffalo or should I follow another procedure?

Thanks again for your software.
BR,
Samuele.

1000001101000, 2020-01-18:
Good to hear from you!

i have run OMV briefly on a few devices including an LS210. It seems to work reasonably well though I've only used it briefly to help others test it.

The fstab issue is pretty common and has a few variations. By default if a filesystem can't be mounted at boot the system pauses rather than continue which is a pain on a system with no console to fix it. There are a few ways to ensure this doesn't happen:

- use uuids rather than device names in fstab where possible 
- make sure any raid changes are added to mdadm.conf 
`mdadm --detail --scan >> /etc/mdadm/mdadm.conf`
- generate a new initramfs (initrd.buffalo) any time you change mdadm.conf or fstab 
`update-initramfs -u`
- make sure to test any new fstab entries before rebooting.
- if added an entry for something that won't always be present at boot I beleive there is an "optional" parameter that can be set in fstab (things like nfs/cifs do this automatically, I think)

samueleforconi, 2020-01-20:
Thanks for your suggestions!
If I'll succed in configuring OMV I'll write a mini how-to for the LS220...
Great work, again!

closed 2020-01-20 by samueleforconi

1000001101000, 2020-01-20:
I do have some notes from when I set up OMV last. At that point they still only supported Debian Stretch. Let me know if they've added Buster support.

I've also heard of folks using Webmin for similar functionality. I've never used it personally.... Or rather I used it at school for web development stuff a decade ago but have never set it up or used it on a NAS.

samueleforconi, 2020-01-22:
Hi,
just for information, I managed to setup OMV and let it run correctly.
It seems that the OMV5 installation on these embedded platorms needs some tweaks... 
After installation the Web UI interface is not shown, so I have to run these commands to enable it:

`sudo omv-salt deploy run nginx`
`sudo omv-salt deploy run phpfpm`

Then, the problem was that OMV5 isn't able to manage the fstab auto-mounted raid array, so in the web UI I wasn't able to create a shares on the raid device.

This was my intial /etc/fstab:

`# /etc/fstab: static file system information.`
`#`
`# Use 'blkid' to print the universally unique identifier for a`
`# device; this may be used with UUID= as a more robust way to name devices`
`# that works even if disks are added and removed. See fstab(5).`
`#`
`# <file system> <mount point>   <type>  <options>       <dump>  <pass>`
`# / was on /dev/md1 during installation`
`UUID=0dd2dddb-3f85-425a-bce2-0a8a84c3c68d /               ext3    errors=remount-ro 0       1`
`# /boot was on /dev/md0 during installation`
`UUID=a57a2a55-dd77-4582-9441-deecca54d9df /boot           ext3    defaults        0       2`
`# /mnt/array was on /dev/md10 during installation`
`UUID=6c5c81e8-7736-4d9d-8d45-d0c1d920308b /mnt/array      ext4    defaults        0       2`
`# swap was on /dev/md2 during installation`
`UUID=ccf1d339-5771-4dc8-9078-358baff2798d none            swap    sw              0       0`

Then I performed these operations:
1. in OMV web UI unmount the raid array (that in my system was mounted on `/mnt/array`)
2. in OMV web UI remount the raid array
3. have a look at the /etc/fstab, now my /etc/fstab is:
`# /etc/fstab: static file system information.`
`#`
`# Use 'blkid' to print the universally unique identifier for a`
`# device; this may be used with UUID= as a more robust way to name devices`
`# that works even if disks are added and removed. See fstab(5).`
`#`
`# <file system> <mount point>   <type>  <options>       <dump>  <pass>`
`# / was on /dev/md1 during installation`
`UUID=0dd2dddb-3f85-425a-bce2-0a8a84c3c68d /               ext3    errors=remount-ro 0       1`
`# /boot was on /dev/md0 during installation`
`UUID=a57a2a55-dd77-4582-9441-deecca54d9df /boot           ext3    defaults        0       2`
`# /mnt/array was on /dev/md10 during installation`
`UUID=6c5c81e8-7736-4d9d-8d45-d0c1d920308b /mnt/array      ext4    defaults        0       2`
`# swap was on /dev/md2 during installation`
`UUID=ccf1d339-5771-4dc8-9078-358baff2798d none            swap    sw              0       0`
`# >>> [openmediavault]`
`/dev/disk/by-id/md-name-LS220D-EME85:10         /srv/dev-disk-by-id-md-name-LS220D-EME85-10     ext4    defaults,nofail,user_xattr,noexec,usrjquota=aquota.user,grpjquota=aquota.group,jqfmt=vfsv0,acl  0 2`
`# <<< [openmediavault]`
4. I commented out the initial mount array line, so the /etc/fstab is now:
`# /etc/fstab: static file system information.`
`#`
`# Use 'blkid' to print the universally unique identifier for a`
`# device; this may be used with UUID= as a more robust way to name devices`
`# that works even if disks are added and removed. See fstab(5).`
`#`
`# <file system> <mount point>   <type>  <options>       <dump>  <pass>`
`# / was on /dev/md1 during installation`
`UUID=0dd2dddb-3f85-425a-bce2-0a8a84c3c68d /               ext3    errors=remount-ro 0       1`
`# /boot was on /dev/md0 during installation`
`UUID=a57a2a55-dd77-4582-9441-deecca54d9df /boot           ext3    defaults        0       2`
`# /mnt/array was on /dev/md10 during installation`
`#UUID=6c5c81e8-7736-4d9d-8d45-d0c1d920308b /mnt/array      ext4    defaults        0       2`
`# swap was on /dev/md2 during installation`
`UUID=ccf1d339-5771-4dc8-9078-358baff2798d none            swap    sw              0       0`
`# >>> [openmediavault]`
`/dev/disk/by-id/md-name-LS220D-EME85:10         /srv/dev-disk-by-id-md-name-LS220D-EME85-10     ext4    defaults,nofail,user_xattr,noexec,usrjquota=aquota.user,grpjquota=aquota.group,jqfmt=vfsv0,acl  0 2`
`# <<< [openmediavault]`
5. Now I have run `sudo update-initramfs -u` and rebooted the system

Now I can select the device from the devices list in OMV and use it for sharing my files!

The next step for me will be to setup an USB-WiFi dongle based on the Ralink RT5350...