1000001101000/Debian_on_Buffalo #34: sockettimeoutexception on acp_commander

xSirRudolph, 2020-01-30:
Hello 1000001101000 - I messaged you on buffalo forums awhile back, now i'm trying to get debian running again on my LS441DE while keeping my raid arrays intact. I believe last time I was able to use your acp_commander to get the the images onto the drive. But now every time I get this sockettimeoutexception

from admin command prompt:
`C:\Windows\System32\acp-commander-master>java -jar acp_commander.jar -f
LS441DE93       192.168.1.4     LS441D(HORIKAWA)        ID=00643        mac: xxx  FW=  1.860
Found 1 device(s).

C:\Windows\System32\acp-commander-master>java -jar acp_commander.jar -t 192.168.1.4 -pw mypassword -o
A SocketTimeoutException usually indicates bad firewall settings.
Check especially for *UDP* port 22936 and make sure that the connection to your LS is working.
ERROR: Exception: SocketTimeoutException (Receive timed out) [ACP Send/Receive (Packet:80A0 = SPECIAL_CMD_EN_ONECMD)]

C:\Windows\System32\acp-commander-master>`

xSirRudolph, 2020-01-30:
Nevermind, turning off windows firewall let me continue, thanks

closed 2020-01-30 by xSirRudolph

1000001101000, 2020-01-30:
Nice!