1000001101000/Debian_on_Buffalo #35: Cannot allocate skb on queue 0

marien341, 2020-02-11:
Various issues in dmesg:
ksoftirqd/0: page allocation failure: order:0, mode:0x480020(GFP_ATOMIC)

LS421D hangs after a few hours.
For now solved by increasing in sysctl.conf vm.min_free_kbytes = 8192
I will monitor & report but the 2000 & something seems too low.

Thanks for this great work 1000001101000!

1000001101000, 2020-02-12:
Good to hear from you!

I think someone ran into something similar in the past, I don’t recall the details but will look into it in a few days when i’m back. I may end up adding that setting to the installer.

What applications are you running/installing on your device?

marien341, 2020-02-12:
Nothing big - pure-ftpd, smbd and minidlna. I am running at less than 100 MB RAM usage.

1000001101000, 2020-02-16:
I took a look back at the older issue that was similar. In that case the user was running Debian Buster to take advantage of some new Samba feature back before RC1 and long before it was "stable". I believe in that case they changed vm.min_free_kbytes which partially helped but ultimately ended up going back to Debian Stretch which was the current stable distribution at the time. 

If you're finding changing that setting has solved the problem for you I'll do a little research and add it to the installer if I don't find there's a downside. 

In general my goal is to have the Debian install be as "vanilla" as possible but am including whatever tweaks are needed for Debian to run properly for the hardware, this seems to fit.

Let me know if you run into any problems in the meantime.

1000001101000, 2020-02-17:
could you try with 4096 and see if that is enough to solve the problem for your use case?

I'm curious if 4MB is sufficient since for some devices 8MB is a fairly large amount to reserve.

marien341, 2020-02-18:
Will do - not home for a few days so back to you soon. Thanks!

1000001101000, 2020-05-31:
sorry for the delay. Did 4096 end up working in the long term?

closed 2021-06-29 by 1000001101000