1000001101000/Debian_on_Buffalo #36: LS-X1.0TL

bcoff12, 2020-02-25:
I know this is a longshot, but I was curious if there was a way to install Debian to the subject device?  It has no USB / power switch but it would be a lot more useful running Debian than the stock firmware.

Thanks!

1000001101000, 2020-02-25:
Good to hear from you!

It should be possible but I believe we'll need to create a new device-tree for it and will probably need to go through some trial and error to get it working. 

The first step is to confirm some details about the hardware it's running. 

There's an old page out there with some hardware details but I'd like to confirm them first:
https://web.archive.org/web/20180319150521/http://buffalo.nas-central.org/wiki/Category:LS-XL

You should be able to get most of what we need by connecting using ACP_commander and posting the output of "dmesg" 
https://github.com/1000001101000/acp-commander

Once we have that hardware information I'll build a device tree and we can try booting into an installer image with it. This may take a few tries which may require connecting the drive to a PC (or TFTP I suppose) to recover. Before we start you should confirm you're able to access the drive without too much trouble.

bcoff12, 2020-02-25:
I have it in pieces and can easily test.

I restored the original image and booted it.

Output from dmesg and /var/log/messages here:

https://pastebin.com/F9ngexDb

1000001101000, 2020-02-25:
excellent, that confirms most of my assumptions. I'll take a shot at a device tree tonight and we can give it a shot. 

For the first one I won't include any LEDs or Buttons to avoid pin conflicts. Assuming we get that working you can then run through the Debian install like normal. Once we get you running Debian I'd like to do some tests to verify various hardware components and then I can walk you through how I map LEDs and Buttons.

bcoff12, 2020-02-25:
Thank you so much.

I have SSH access working if you need anything else.  I will provide anything needed to get this working.

1000001101000, 2020-02-25:
could you post the output of "find /proc/buffalo"? I think that will give us somethings we'll need later like a list of the leds

1000001101000, 2020-02-25:
also "ls /sys/block"

1000001101000, 2020-02-25:
just to be thorough:
"cat /proc/cpu"
"ls -lh /boot"
and if this works:
"hexdump -C /dev/mtdblock0 | grep -i uimage" 
"hexdump -C /dev/mtdblock1 | grep -i uimage"

bcoff12, 2020-02-25:
Please see:

https://pastebin.com/SusKknFV

1000001101000, 2020-02-25:
perfect. I created a super-basic device tree and added it to the project along with a flash-kernel entry that should allow an install to succeed. New installer images have been generated which incorporate these files:

https://github.com/1000001101000/Debian_on_Buffalo/tree/master/Buster/installer_images/armel_devices

Give it a try and see what happens. With a little luck it should boot straight into the installer. If not we can troubleshoot as needed.

bcoff12, 2020-02-25:
Ok, copied the files to /boot and rebooted.  It hasn't pulled an IP address after several minutes.  I am remote from the device so I will check it in the morning.  Thanks again!

1000001101000, 2020-02-25:
Sounds good. 

Hopefully it’s just an issue with the network (I only enabled 1 of 2 possible adapters). If it’s boot looping it may be something more subtle.

bcoff12, 2020-02-26:
It was in a boot loop.

Is anything logged that I can look at?  I have the drive pulled and mounted on another system.

1000001101000, 2020-02-26:
until we get past the boot loop stage there won't be any debug info we can look at without attaching a serial console. If you are good at soldering (I'm not) that page I sent you showed how to connect via a 3.3v usb adapter. 

I'll look over the device tree again and see if I can find something that could be causing the issue. I based it off a different device that uses the same SoC but a couple of things didn't match what I'm used to for other SoCs. I'll take a look at a few others and see if there's something that needs fixing. 

The main thing you can check on your end is that files on the disk show as valid. If you connect to a linux system and use the "file" command they should show as uboot images along with some details, if not they may not have downloaded correctly..... now that I mention that, I need to verify I built the uImage properly too. I added it to the build process I'm using for the other Kirkwood devices assuming the entry point would be the same but should verify that too.

bcoff12, 2020-02-26:
Files confirmed valid.

Those solder points are tiny!  I can give that a shot if it comes to that.

1000001101000, 2020-02-26:
Hopefully it won't come to that. 

I started over based off the "Marvell RD88F6192 reference design" which is probably what I should have started with. There was one difference in how the console was set up which I wouldn't think would make a difference but if it was causing a problem it would explain the boot loop. 

give that one a try and see what happens. 

I've got a handful of things we can try until something works but are largely stuck with trial and error for now.

bcoff12, 2020-02-26:
Reboot loop.

1000001101000, 2020-02-26:
I Just posted another one. this one is basically the lswxl one with the memory value adjusted. 

If that doesn't help I'm going to try again based on the reference design but with the Stretch kernel to narrow some some possibilities. I'll take a look at the stock kernel source at some point to see if I can find any hints.

bcoff12, 2020-02-26:
Thanks again for your efforts.

This one also went into a reboot loop.

1000001101000, 2020-02-26:
could you take a picture of the board when you get a chance?

I'd particularly like to see the SoC, it'll be the biggest chip and should say 88f6190 or something similar.

I've got a few more things to try yet, but the serial console is looking more and more tempting.

bcoff12, 2020-02-26:
I can take a photo if you need.
88F6-LG02
N4W8590.2
1132 A1P
TW
88F6192-A1
C080

1000001101000, 2020-02-26:
That should give me what I need to confirm the SoC stuff. A picture would help later on so I can match chips to the device tree but most of that comes later. 

I took a peek at the stock firmware. it looks like they use the same kernel for all the kirkwood linkstation devices. That would make me assume this shouldn't be so difficult but there may be a reason this one didn't get supported along with the other devices. I guess we'll find out.

bcoff12, 2020-02-26:
Would the low RAM have anything to do with that?

1000001101000, 2020-02-26:
If read everything correctly this device should have 64MB ram. The older Orion-based Linkstations have the same and work just fine. I wouldn't think it's an issue.

1000001101000, 2020-02-26:
I posted another one (the build process should run in a few minutes to update the installer). This one strips it down to the absolute minimum to boot (including removing ethernet and sata). That should narrow things down further.

bcoff12, 2020-02-26:
Sorry to report that this one also resulted in a reboot loop.

1000001101000, 2020-02-26:
not surprising but it confirms it at least.

1000001101000, 2020-02-26:
I found something interesting. Looks like someone created a kernel patch to boot this device a while back:
https://forum.doozan.com/read.php?2,15699
https://github.com/ingmar-k/Buffalo-Linkstation-Kirkwood-Kernel/commit/3bc4ecb73f668e7fdb191427492ac079b390118f

This is the "old" way of doing things from before device trees and wouldn't work as-is on a modern kernel but should be helpful in figuring things out.

1000001101000, 2020-02-26:
here's something else you can try in the meantime:

try the lschlv2 image from my repo. if that doesn't work either try this one:
http://ftp.debian.org/debian/dists/stretch/main/installer-armel/current/images/kirkwood/network-console/buffalo/ls-chlv2/

bcoff12, 2020-02-26:
lschlv2 - reboot loop.

Second one - boot successfully, allows connection on SSH, and I can login with installer/install.

1000001101000, 2020-02-26:
that's more like it. 

I'm guessing this means it doesn't like the kernel I'm using for the buster installer. Could you try this one next:

http://ftp.nl.debian.org/debian/dists/buster/main/installer-armel/current/images/kirkwood/network-console/buffalo/ls-chlv2/

bcoff12, 2020-02-26:
This one also boots successfully and allows me to login with installer/install.

1000001101000, 2020-02-26:
Excellent. 

The problem seems to be with the kernel I use for the Buster installer. I'm using a custom kernel because the Terastation II and Terastation III require custom patches to work. This hasn't been a problem for the other kirkwood devices but apparently is for the ls-xl for some reason. 

Eventually I would like to get it working with my kernel but I think we should start by getting a good device-tree for the ls-xl on the current Buster kenel provided by debian and expand from there. 

Try running through the install with that last image and see how far you get. It'll be easier to work on the device tree from a functioning install if you can get that far. If that works it'll be fairly simple to adjust the device tree to handle the ls-xl hardware properly.

bcoff12, 2020-02-26:
Will do.  I can't thank you enough.

1000001101000, 2020-02-26:
looks like I am using a different entry point in my uImages 0x00008000 vs 0x01A00000 compared to the ones provided by Debian. 

If that mattered for other devices I should have seen failures when testing the ls-wxl at the very least.  Makes me think it isn't the issue, but I think I'll test with some other devices at some point. 

We'll find out for sure when we start testing new trees later on.

bcoff12, 2020-02-27:
I had issues with my session being disconnected during the install - on my last attempt, I got as far as "determining kernel version" before it died.  The disconnections didn't seem to occur at the same step in the process, so I don't think it was a specific action that caused it. 

I'm going to restore a bootable disk and try one more time and I'll let you know.

1000001101000, 2020-02-27:
Interesting.

Since we’re using a device tree for another device some instability isn’t too surprising.

I’ll try packaging up the installer/kernel that we know works with a device-tree based on our earlier attempts. Hopefully that will work better

1000001101000, 2020-02-27:
I setup a new workflow to create a "kirkwood" installer that uses the default Debian kernel rather than my custom one (basically the same process I use for ARMHF). I'm thinking of making this change permanent to avoid weird issues like this for devices that don't actually need the custom kernel. 

since it uses the same kernel that you've already had success with I expect we'll have better luck. Hopefully switching to a tree meant for the 88F6192 and without entries for non-existant hardware the instability will go away. I've set up the workflow to trigger whenever I push a new dtb, hopefully that will help us iterate faster too.
 
the new images can be found under:
Buster/installer_images/armel_devices/kwtest/

bcoff12, 2020-02-27:
Thank you, I will try it.

I had some issues restoring my Clonezilla image this morning but that's all taken care of.

I just tried with the ftp.nl.debian.org again and it disconnected when I was setting up the user account.

1000001101000, 2020-02-27:
the lschlv2 device tree defines a bunch of hardware (buttons/fans/leds/etc) that the ls-xl doesn't have. If the ls-xl uses those same pins for something important it could be causing an issue. I imagine there  could be some more subtle issues with using a device tree meant for a different SoC too, though looking at how there aren't many relevant differences (which explains why it works at all). 

It could also be something we're missing like setting up a watchdog timer or something of that nature. 

fortunately, some of these issues can be seen from a second window you could watch during the install though many would still fail to quickly to make it out over ssh.

bcoff12, 2020-02-27:
KWTEST - reboot loop.

I will try with two windows open and ftp.nl.debian.org and let you know.

1000001101000, 2020-02-27:
usually if you connect in the seperate window and then choose exit in the menu it will fall to a screen showing the output of syslog in realtime. Depending on what's happening this may give useful info. 

I think for the next device tree I'll start with the working-ish ls-chlv2 tree and pull out some things that shouldn't be there for the ls-xl. starting with something that boots is probably a better starting point.

bcoff12, 2020-02-27:
Ok I am probably spending too much time making this perfect, but I think adding a swap partition and doing a swapon is helping me with my issue....  I will quit trying to get my partition table just right and just get an install completed and let you know!

1000001101000, 2020-02-27:
yes.... 100% that. I hadn't thought of that but a 64MB device will have problems without swap. 

I'm hoping we'll get my custom installer working eventually because there is an additional tweak that I make specifically for systems with <128M to prevent some systemd stuff from failing. It shouldn't be a big deal for your initial install though.

bcoff12, 2020-02-28:
Install is proceeding - slowly.

1000001101000, 2020-02-28:
The speed will be limited by a number of factors, particularly since it pulls all the packages from a Debian mirror. If it's taking more than an hour or hanging in one place I would be somewhat concerned. 

I know I had to make some tweaks to the installer to make certain things work on the ls-gl which also only has 64mb ram. I believe the only one during the install was trying to compress the initramfs using xz, the other issue was systemd services failing at start. 

The code I use is (run from inside the target environment):
echo "tmpfs /run tmpfs nosuid,noexec,size=26M,nr_inodes=4096 0  0" >> /etc/fstab
mount -o remount tmpfs

If it's getting stuck right around installing the kernel it could be a similar issue and this could help.

1000001101000, 2020-02-28:
Let me know how the install turns out. I think I have an ls-chlv2, if so I'm going to pull it out and do some parellel testing. Hopefully that way I can confirm that images I post work before sending them to you.... or at least confirm whether some of these issues are ls-xl specific.

bcoff12, 2020-02-28:
Will do.  The install is proceeding, but it is VERY slow.  I can only guess that it is doing a ton of swapping - "free" shows swap in use but I can't tell how much swapping it is actually doing.  I just don't have any other explanation for why it is taking so long.

1000001101000, 2020-02-28:
depending on how slow that drive is swapping and all the writes involved in downloading and unpacking all the packages could be very slow. I've never seen one go anything like this long but I've only done a few on devices with 64MB ram and I think those were mostly with some cheap SSDs I use for testing.

Probably best to leave it alone and see if if it finishes though it's probably worth opening another session and take a look at the output of dmesg to see if something is spewing out warning/errors.

I'll have to see how it goes on the ls-chlv2 when I get that far. It has the same amount of ram and a slower clock speed which should make an interesting comparison.

bcoff12, 2020-02-28:
dmesg - clean.  Nothing in it since it mounted the target / partition.

1000001101000, 2020-02-28:
what step is it on right now, is it still installing packages?

bcoff12, 2020-02-28:
Correct...  16%.

1000001101000, 2020-02-28:
may as well let it run since you've come this far.

how fast is your internet connection?

bcoff12, 2020-02-28:
Not this slow!  LS210 took under an hour as comparison.  This install looks slightly different but I'm pretty sure it is the combination of the hardware and the old rotational disk I'm testing with.

1000001101000, 2020-02-28:
I wouldn't expect it to take more than 2-3x as long based of CPU/memory alone. I'm sure we can  eventually figure out whatever the issue is. If the install succeeds we'll have a lot more tools to work with. 

If the last few steps fail (the screen will turn red), don't turn it off/reboot. we can probably correct most issues manually and allow the install to finish.

1000001101000, 2020-02-28:
how much swap did you give it?

bcoff12, 2020-02-28:
Currently "Preparing the system" (window title "Making the system bootable"

1GB swap.

1000001101000, 2020-02-28:
fingers crossed

bcoff12, 2020-02-28:
Wish I had used an SSD!!

1000001101000, 2020-02-28:
if this does turn out to be because of the drive I would suspect it's failing. 

I'd love to blame it on 88F6192 vs 88F6281 device tree issues but looking closer, the only differences appear to be related to the SoC on-board sdcard port which isn't even enabled in the device-tree. I suppose it could be some sort of conflict caused by have the fan/buttons defined even though they don't exist but I don't really see how. 

Hopefully your install will finish soon and you can take a look at the SMART data.

bcoff12, 2020-02-28:
It may be.  It is a 1.5TB drive and who knows what its history is...

It is still running.  I am working at 6AM EST tomorrow so I can let it go overnight and check it in the AM.

[57 more replies]

closed 2020-08-31 by 1000001101000