1000001101000/Debian_on_Buffalo #37: Can not install debian on buffalo LS410D

mailminhthang, 2020-03-14:
Hi Mr.1000001101000
I can not install debian on LS410D, my steps are below:
1) clear all partitions in HDD to get blank HDD
2) use TFTP to setup IP for LS410D (use DHCP)
3) install firmware 1.86 with re-partitions and format disk, NAS runs OK as usual
4) use acp_commander to setup SSH (this firmware does not has telnet service built-in) and I can ssh to LS410D ok
5) get 2 files initrd.buffalo, uImage.buffalo.ls410d from Debian_on_Buffalo/Buster/installer_images/armhf_devices/ then copy to /boot with name  initrd.buffalo, uImage.buffalo (I already backup these old files in /boot) and with mode 775
6) run "sync" and reboot
Then Nas can not boot, red light blinks, and Nas can get IP via DHCP. So I try run TFTP again, it comes back to normal Nas without user "installer" and if I restart again, red light blinks again without IP. I thinks your 2 files  initrd.buffalo, uImage.buffalo.ls410d seems to be not compatible with my Nas.  I try with 2 files in Debian_on_Buffalo/Stretch/installer_images/armhf_devices/ but the issue is the same.
Please help me to install
Thanks so much

1000001101000, 2020-03-14:
Good to hear from you!

I booted into the buster ls410 installer and confirmed it's working (I wanted to rule out corruption in the repo files before moving forward).

The procedure you describe sounds like it should work,  though there are some places it could have gone wrong. For your next attempt try keeping your terminal output and post it so we can look closer.

How many times does the red light blink? I'm guessing 6 which indicates it can read the partition but not the boot files.

My best guess is that something went wrong with the download which could explain why the boot failed. The next most likely would be an incorrect filename or something like that. 
 
You can try booting into the installer directly via TFTP, that should remove some of the possible problems. 

I didn't really understand your note about telnet vs SSH. Telnet is available on every model, though older versions of acp_commander couldn't always activate it. If you use my fork of acp_commander it should work without issue: 
https://github.com/1000001101000/acp-commander

mailminhthang, 2020-03-15:
Hi Mr.1000001101000
Nice to get your quick response
Should I explain my steps:
Load 2 files (initrd.buffalo, uImage.buffalo) via TFTP -> format disk, load firmware 1.86 (http://buffalo.jp/support_ap/support/products/ls400.html) via LSUpdater, it works OK (without user "installer") -> active SSH via acp_commander -> replace with your 2 files, sync and restart -> it backs to 6 times red light blinks, no IP and LUpdater, Nas Navigator can't recognize -> load 2 files (initrd.buffalo, uImage.buffalo) via TFTP then Nas becomes OK as previously (without user "installer") -> restart again -> it backs to 6 times red light blinks -> load 2 files (initrd.buffalo, uImage.buffalo) via TFTP then Nas becomes OK as previously (without user "installer") -> restore to 2 original files via SSH then restart again -> Nas becomes OK as previously even I restart many times after.

> How many times does the red light blink? I'm guessing 6 which indicates it can read the partition but not the boot files.

Yes, it blinks 6 times (I've experience installing buffalo Nas like TFTP, load/update firmware, I do many time)

>  didn't really understand your note about telnet vs SSH. Telnet is available on every model, though older versions of acp_commander couldn't always activate it

It's no problem SSH or Telnet, I just say I can active SSH via acp_commander instead of active Telnet

> You can try booting into the installer directly via TFTP, that should remove some of the possible problems.

Could you please guide me to do

Thanks so much

1000001101000, 2020-03-15:
I’d like you to try again but copy&paste your terminal session afterwards. That way if there is an error/typo I might be able to see what’s wrong.

To TFTP into the installer is the same as with restoring the stock firmware. You just use the debian *.buffalo files instead of the normal ones. If it works it should start the ssh installer just like the normal method.

mailminhthang, 2020-03-15:
> To TFTP into the installer is the same as with restoring the stock firmware. You just use the debian *.buffalo files instead of the normal ones. If it works it should start the ssh installer just like the normal method.

I use TFTP here https://anp.synology.me/ls420d/
I replace your 2 files and run TFTP as in picture here https://ibb.co/jwXYqpg
But after that white light blinks and LSUpdater, Nagivator can not find the Nas

1000001101000, 2020-03-15:
You should be able to connect to the installer via ssh after a few minutes.

mailminhthang, 2020-03-15:
> You should be able to connect to the installer via ssh after a few minutes.

I wait for long time but the white light blinks and LSUpdater, Nagivator can not find the Nas, I can't find out whether it has IP or not even I try to change my PC IP to 169.254.11.1 and scan again

mailminhthang, 2020-03-15:
I think white light blinks that means it cannot get the IP
more over let see back picture above  https://ibb.co/jwXYqpg, I don't know why do upload of 2 files repeat twice (4 lines)

1000001101000, 2020-03-15:
Remember, the debian install does not use lsupdater/nasnavigator. Blinking white light is normal for the debian installer.

You never said whether you tried connecting over ssh or checked for port 22 or not.

mailminhthang, 2020-03-15:
When the light is blinking, I login to web admin of my Internet Modem to find Nas IP and I don't see any new IP that it assign for NAS via DHCP.
So how can I find the IP of NAS after TFTP your 2 files to NAS ?

1000001101000, 2020-03-15:
If it didn't pick up an address via DHCP then it most likely failed to boot. You could always scan your network for devices with port 22 open using nmap or similar but looking at your DHCP reservations should be sufficient. 

It really sounds like there is something wrong with the installer files you are using. Did you clone the repository or download them by some other method?

mailminhthang, 2020-03-15:
OK, should I describe my steps by pictures:
1 and 2) download 2 files (let see https://ibb.co/vv8H2PX and https://ibb.co/VwT0KWC)
3) backup old files and copy 2 files to folder TFTP (let see https://ibb.co/nsMm16m)
4) power on LS410d, red light blinks because of empty hard disk (let see https://ibb.co/F5MwxG5)
5) check web admin of Internet modem to see DHCP IP assigned (let see https://ibb.co/n7Ry2Kv)
6) press button in NAS and upload TFTP, I see some thing is wrong here because files are uploaded twice (lee see https://ibb.co/5kdQm8p)
7) After that white light blinks, I wait for long time and check web admin of Internet modem, there is no new IP for NAS and I think NAS still has no IP

Tks so much

1000001101000, 2020-03-15:
The first picture shows the problem. 

Clicking save-as on that link doesn't save the file, it saves the HTML of the github page. If you view the resulting file you'll see it contains HTML similar to:
`<!DOCTYPE html><html lang="en"><head><meta charset="utf-8">` 

If you click the link for the file you'll see a "download" button which you can use to download the actual file.

mailminhthang, 2020-03-16:
Ok, I understand now, I will try and inform you result.
Thanks so much

mailminhthang, 2020-03-16:
Ok, I've installed to Debian.
Thanks so much again

1000001101000, 2020-03-16:
Great!

closed 2020-03-16 by 1000001101000