1000001101000/Debian_on_Buffalo #39: TeraStation 5810 Compatibility?

Borchardt, 2020-03-24:
Hey!

This is more a question than a feature request, but I'm curious what's blocking compatibility for the 5x10 series. In particular, I'm interested in whether it might be possible to get Debian installed on my 5810DN, since I believe it uses the same architecture (armv7-a) as the 3400, which is supported.

Alternately, I'm curious whether you might be able to offer any advice about the viability of using ACP Commander to SSH in and stand up a MinIO server on this NAS without changing the OS.

Many thanks!

1000001101000, 2020-03-24:
Good to hear from you!

I have a TS3210 that I've done some experimenting with, I've made some notes about it here including how to get root access and make modifications to the underlying Ubuntu OS:
https://buffalonas.miraheze.org/wiki/Terastation_TS3210DN

Poking around at it made it pretty clear that the TS3X10 and TS5x10 are based on the same board more or less (they appear to use the same device tree anyway).

There are a few challenges that have prevented me from adding support for these devices so far:
1. The device boots from NAND rather than the hard drives by default (though the OS is on the drives)
2. I wasn't able to get a serial console working so far (I have notes about my attempts on the wiki page).

When I did my initial experiments I tried adjusting the boot parameters for it to boot from disk and attempted starting a mainline kernel but it didn't work. Unfortunately, without a console it's difficult to know why it failed or troubleshoot further and messing with the boot parameters risks leaving the device unable to boot (though it seems to have a failsafe for that).

It also seems like the mainline support is pretty rudimentary, I'm not sure if that would pose any problems or not. I am following someone on twitter who is trying to improve support for a QNAP device which I believe uses the same SoC which may help in the future as well.

I haven't looked at this in a while and hope to come back to it at some point in the future. My hope would be that some one with better electrical/soldering skill than me could figure out how to enable a serial console which would speed things up significantly.

Edortiz101, 2020-05-24:
Hi, 
I'am also interested in installing Debian on the 5010 series, precisely the TS5410DN.
I hope you will be able to figure it out in the future. 
Thanks in advance.
Cheers

Borchardt, 2020-05-24:
I'm long overdue for a response here, too. Thanks for your response, 1000001101000!

I was able to use ACP Commander to get ssh access. With ssh, I was able to update and add some packages. However, running dist-upgrade resulted in a device reset that seemed to undo all of the changes that I made.

1000001101000, 2020-05-24:
I haven't done much with the stock firmware beyond testing some changes to acp_commander to allow it to provide root/telnet access automatically and general poking around. i would guess that trying to do something like dist-upgrade caused it to install some updated packages like init-scripts that likely overwrite something vital. that must trigger some sort firmware repair/reinstall from flash.

for major changes it may be better to set up a chroot environment to isolate them from the underlying firmware. I've been meaning to try/document that process on the LS500 series which have a super limited firmware that boots entirely from NAND.

1000001101000, 2020-11-26:
@Borchardt @Edortiz101 
I'm going through some issues I haven't responded to in a while, I think I forgot about this one when I was working with my TS3210 (briefly) a few months ago.

I setup a Debian chroot and played around a bit to get some benchmarks/etc and learned some things in the process. It turns out the kernel used in the stock firmware on the device has some custom patches to support large memory pages (possibly to get around the 16TB volume limit for 32-bit devices). Many packages not built specifically for this kernel configuration will crash when they run into the unexpected memory layout.

I captured some of this info here:
https://buffalonas.miraheze.org/wiki/Terastation_TS3210DN#Custom_Firmware.2FOperating_System_installation

Eventually I still hope to figure out how to run a custom OS on this device but there are some challenges to overcome. With the boot files being in the flash it makes testing harder than the older devices, I can't afford to risk flashing something untested until I figure out a way to re-program the flash without booting the devices. On other devices I've also been able to test custom rootfs with the stock kernel before working on moving to a custom kernel, for this device it seems I'll need to do both at the same time which will further complicate things.

someday I hope to make progress on this, though it may need to wait till I get the skills/equipment to replace flash chips on this type of board. Maybe in the meantime I'll figure out how to compile compatible packages for it and write up a guide for doing that.

1000001101000, 2023-05-14:
hmmm, taking another look at this has been in the top 10 of my todo list for....checks notes.... 3 years. 

I find myself with less and less time to work on my todo list and other things keep preempting it. Currently I'm working through some kernel bugs/changes keeping some devices booting on Bookworm.

I do still hope to get to that work in one format or another, but couldn't really offer an estimate of when.