1000001101000/Debian_on_Buffalo #40: Terastation Pro II Debian Install Fails

scobes74, 2020-04-23:
Running the debian installer fails at the Select and Install Software stage.
I have tried with both the stretch and buster images and get the same issue.
The terastation is left without boot image (E06 TFTP Mode) and then needs to be recovered with a stock firmware to be usable.
[image]

1000001101000, 2020-04-23:
Good to hear from you!

I have not run into this before. I'll give it a try on mine and see what happens (I already have it online for some other testing). My first thought is this could have been caused by the github outages earlier today. I wouldn't think that would have affected stretch though (This device uses Debian's kernel on Stretch but needs a custom one for Buster which is download from github). If so I'll have to see if I can make that piece more fault tolerant.

If it doesn't fail for mine it might be worth trying again to see if the problem was with github and has resolved itself, otherwise I'll have additional questions about how your setup differs from mine. 

I should have the results of my attempt later tonight.

scobes74, 2020-04-23:
Hi,

Thanks for the extremely quick response!

I downloaded the files last Thursday (7 days ago) and that was when I did
the first install with buster. It failed downloading something to do with
apt.
It took me until yesterday to manage to recover the unit and then I tried
again today with the stretch image.
So I suspect it's not related to GitHub issues.
I was on fw 1.35 prior to the install if that makes any difference.
Let me know if you need any more info.

Many thanks

1000001101000, 2020-04-23:
The current files are new as of a few days ago and include a minor fix for for the startup service and is based on the latest Debian installer image. Which means what I'm testing won't be identical to what you used. So there's some chance the problem has already solved itself (we can always hope).

The firmware version shouldn't make any difference, once it gets into the installer it's past the point that the firmware/bootloader s doing anything.

I started a Buster install on mine, it should finish within an hour or so. If mine fails I'll work through the issue on my device, if it doesn't I'll walk you through how to watch the logs during your next attempt so we can see where things go wrong (if they go wrong).

There have been some github failures a few times this week (though mostly today). I suppose you could have just had really bad luck but I don't find that particularly likely.

1000001101000, 2020-04-23:
My buster install reached the end without issue. My first suggestion would be to download the latest Buster images for this device and try again.

You can watch the log during the install by opening a second ssh session, selecting "start a shell", then type "exit". That will bring you to a screen that scrolls the log as it goes. 

You can also just wait and see if it fails, if so don't do anything in the main window but start a second ssh session and select "start a shell" and then type "tail /var/log/syslog" and send me the output. 

[image]

Hopefully it will just work this time, otherwise we'll need to figure out what the difference is.

scobes74, 2020-04-23:
Ok, that's good news. I assume I can just tftp the buster installer boot
files instead of re flashing to the stock firmware first.

Also when I get to partitioning I am formatting ext3 for /boot, xfs for
root / and keeping the main xfs which I am mounting as/mnt/array.

I am not doing anything with partitions detected on the actual disks just
the raid. Not sure if this is correct or not.

I will have a go at this tomorrow and let you know how I get on.

Thanks

1000001101000, 2020-04-23:
Yes, if you are familiar with the TFTP process it should work with the Debian installer just like the stock firmware.

Your partitioning sounds fine, using the raid devices created by the stock firmware is typically what I recommend. I usually use EXT4 for my rootfs but i wouldn't expect that to make any difference (I can't think of a mechanism for that to cause a failure, but I have never tried it).

Good luck!

scobes74, 2020-04-24:
Hi,
I retried and it failed again.
The log was useful though as I can see that in fact my partition is out of space and so the installer cannot write any more files to it. At least that's what I think is going on.

I've attached screenshots of the error with log showing, and the output of df -h.
Also my partition sizes - I used the exact same partitions that were originally on the device as I wanted to keep my data that exists in the array.

I suspect that I will need to create a bigger root partition which probably means I will lose my data array - could you advise on how to do that?  Would be good if you could post what partitions you have on yours so I can see what is required.

Thanks
[image]
[image]
[image]

1000001101000, 2020-04-24:
That makes sense, too bad the installer doesn't make that more clear on the main screen. I didn't realize how small the default partitions are for this model, that is way to small. You'll probably want to delete everything and start from scratch (except maybe /boot). Some of the customization I made to the install include making sure it creates arrays/filesystems that the bootloader can handle on these older devices, it should be safe to make a larger /boot as long as you set it as RAID1 and EXT3 (and no bigger than 1GB). 

Here are some screenshots I took a while back to put together a partitioning guide (somehow I never get around to finishing that):
[image]
[image]

scobes74, 2020-04-24:
Hi, 
I'm just backing up some more of my files before I recreate the filesystem.  I'm hopeful that it will all work once I've done that.  It takes ages to transfer a large amount of data from these - as I'm sure you know.
I'll keep you posted...

1000001101000, 2020-04-24:
Yeah, upgrading arrays always turns into a week long progress for me.

scobes74, 2020-04-25:
Hi,
I think my install has hung.
kicked it off last night and when I got up this morning it is stuck on the installing pciutils (part of base system).  Last activity was at 06:42, current time 09:16.

it does mention base installer could not determine kernel flavour at 03:58 and then says Not installing any kernel - which sounds bad.

This is using your latest buster image for ts pro 2.

Here's some screenshot of the logs.

[image]

[image]

Any ideas ?
thanks

1000001101000, 2020-04-25:
I don’t think the kernel messages are an issue. In the background there are a few of them which are all related to things I needed to do to allow this installer to work on multiple types of device.

I don’t understand quite what happened but I have seen a failed package download stall the process (rarely).

I’m pretty confident this installer will work having run through a full install with it yesterday. I believe starting over the install and trying again should  be the next step.

I’m guessing you can manually kill off whatever apt or dpkg process is hung via the terminal which should get it to the error menu. If you booted via TFTP you could probably just do that again and start over.

scobes74, 2020-04-25:
Hi,

it is actually continuing the install it seems but really slowly - its now
on 96% installing e2fsprogs so it is doing something. I dont understand why
it is so slow though.
Do you think I should leave it to complete, or kill it and try again ?

1000001101000, 2020-04-25:
if it looks like it's moving I would let it go. Around that point it is normal for it to seem to pause for quite a while. 

What happens is that it goes through the entire process of installing the debian kernel and doing final setup, then it switches over to my code which identifies this as a model which needs the custom kernel and goes through the process of installing that kernel and uninstalling the default one. None of my steps update the progress bars so it seems to hang a while.

End to end I believe my install finished in under an hour, though I was using an SSD which probably sped things up. I don't have a good explanation for why it would stall for 3+ hours as you described, it sounds like it happened during the normal package install process. The last time someone reported something like that it was because their hard drive was starting to fail (this was a single-drive unit). I'd recommend checking out the SMART data of the drives once the install completes.

scobes74, 2020-04-26:
Hi,
well it finally worked - on my 3rd attempt.
I don't think I did anything different so not sure why.
Its up and running now anyway which is great.

Thanks very much for all of your help and your work on this project - really appreciated.

scobes74, 2020-04-26:
Hi,
I have noticed after installation that the CPU usage is really high all the time for md3_raid5.
Is this normal ?

[image]

I have installed webmin to manage my server and the website is really unresponsive and keeps losing connection which I believe is because of the high cpu usage.

1000001101000, 2020-04-26:
I don't have much experience with raid5 but I have a guess. 

Most of these SoCs have hardware XOR devices which allows the operations to be offloaded from the CPU. I have some experience setting them up in device trees for the newer devices. Taking a look at the dmesg data for mine I don't see that it's being initialized.

I took a look at the initialization code in the linux kernel of the device and don't see any reference to it:
arch/arm/mach-orion5x/terastation_pro2-setup.c

I do see where it's being initialized in:
arch/arm/mach-orion5x/kurobox_pro-setup.c

I'm going to try adding that initialization code and see if that works (occasionally things are that easy, but not often). 

stay tuned.

scobes74, 2020-04-26:
OK - thanks.
Does that mean I will need to reinstall completely?
Just need to know whether or not to continue configuring the current install

1000001101000, 2020-04-26:
If I manage to make this work it would just be a kernel update, no re-install required. Though on that topic... make sure to install smartmon tools and check your drives for re-located/pending sectors, I'm concerned you may have a disk that needs replacing. 

I tried adding the line from the other device but that didn't work:
root@ts2pro:~# dmesg | grep xor
[   14.954959] mv_xor mv_xor.0: Marvell shared XOR driver
[   14.970247] mv_xor mv_xor.0: Self-test copy timed out, disabling
[   14.972524] mv_xor mv_xor.0: freeing 1 in use descriptors!

I'll poke around at the code and see what else I can try.

1000001101000, 2020-04-26:
I ran across this note digging through the code:
/*
 * The 5082/5181l/5182/6082/6082l/6183 have crypto
 * while 5180n/5181/5281 don't have crypto.
*/

This made me wonder whether this SoC has needed hardware. I took a look at the datasheet for this SoC along with the SoC in the Kurobox. It turns out this SoC does not have the XOR engine.

http://www.theairportwiki.com/images/d/dc/Marvel_88f5281_data_sheet.pdf
https://www.insidegadgets.com/wp-content/uploads/2014/06/88F5182.pdf

There may be other optimizations you can look into but it looks like writing to your RAID5 array will require heavy CPU usage on this device.

scobes74, 2020-04-26:
Ok, thanks for looking. I'll probably reconfigure then and go for raid 10.
 I assume that will work faster as there is no parity.
What do you use on yours?

1000001101000, 2020-04-26:
I would assume that would be faster. I'll be interested to hear your observations.

Personally, I typically use RAID0 to maximize storage and performance and rely on backups for redundancy. I don't necessary recommend that but it's how I typically operate.

1000001101000, 2020-04-26:
amusingly, I ported the orion code to initialize the xor device to the MV78100 (which does have this device) and was able to enable it on my Terastation III. I'm going to attempt the same with the hardware crypto device and include both in future kernels. 

I did notice that even with it the raid5 sync process didn't seem to use it (or at least was using high cpu%). I searched around and found others have seen the same thing but I haven't run across a solution yet.

scobes74, 2020-04-27:
Well, I tried raid 10 and the installer crashed on me several times - I think it was an out of memory error.
in the end I reduced the number of raid arrays and stuck with raid 0 for root and array drives and it installed ok.
the cpu usage is now low and everything seems much more responsive
thanks

1000001101000, 2020-04-29:
I’ve never had a 128MB device run out of memory in the installer, I am unsure how that would happen.

I’d recommend always using RAID1 for boot/rootfs/swap. RAID0/5/10 should all be fine for data arrays though RAID5 will be cpu intensive for reasons already discussed.

I’d be concerned the instability would be due to drive problems or over heating. I’d recommend checking that SMART data and making sure the fan looks okay.

1000001101000, 2020-05-21:
is there anything else we need to look at for this device or should I close the issue?

scobes74, 2020-05-21:
Hi,

thanks for all of your input on this - yes you can close this as the
solution for me was creating a big enough partition and using raid 1 and
raid 0 which don't require the parity.

So far it is performing really well and I am now able to connect using SMB
v3 which was one of my main reasons for moving from stock.

Thanks again !

1000001101000, 2020-05-21:
Glad to hear it's going well!

closed 2020-05-21 by 1000001101000