1000001101000/Debian_on_Buffalo #41: ls-qvl doesn't boot after update & upgrade

riptide981, 2020-04-24:
ls-qvl doesn't boot after update & upgrade

i don't know how to restore it without reinstall

1000001101000, 2020-04-24:
Good to hear from you!

Can you give me a little more detail? are you saying you ran apt-get update; apt-get upgrade and it failed to boot?

Are you running Stretch, Buster or something else?

Have made any changes to your fstab or mdadm arrays since the last reboot?

riptide981, 2020-04-24:
are you saying you ran apt-get update; apt-get upgrade and it failed to boot? Yes

Are you running Stretch, Buster or something else? Buster (10.3)

Have made any changes to your fstab or mdadm arrays since the last reboot? Nope

Kernel: 5.4.0-0.bpo.4-marvell

fstab:
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/md1 during installation
UUID=9d95f2ed-32b4-4854-b4eb-48512ba237bf /               ext3    errors=remount-ro 0       1
# /boot was on /dev/md0 during installation
UUID=ab9ead36-53df-4d60-a74a-cc85798c6ec1 /boot           ext3    defaults        0       2
# swap was on /dev/md10 during installation
UUID=4f7c41d8-ce50-4e9e-92cc-7b2b3a18b81d none            swap    sw              0       0
# /home was on /dev/md2 during installation
UUID=db63da03-1c95-4e12-ac6f-28c388da4e50 /home           xfs     defaults,nofail,noexec,usrquota,grpquota        0 2

/dev/md[0,1,10]: raid1
/dev/md2: raid5

Hope this helps, thank you

1000001101000, 2020-04-24:
Nice!
Did you notice if the update involved installing a new kernel? One possibility is that something with 5.4 is incompatible with the custom device tree causing the boot to fail. honestly, that's just about my only theory at the moment. 

One way to test would be to copy uImage.buffalo.bak and initrd.buffalo.bak to uImage.buffalo and initrd.buffalo and see if it boots like that. 

I believe oyu should be able to connect one drive to a PC and force start the raid1 arrays and make that change, then boot off that disk. If it works you could then insert the other disks and re-sync the arrays. You could also TFTP boot into the installer (or boot from a 5th disk and swap back to the other 4 once it loads) and do the work from the installer shell.

I can elaborate on all of the above if needed.

I can also unpack my LS-QVL this weekend and try out kernel 5.4 and see if I run into issues too.

riptide981, 2020-04-24:
I'm pretty sure that the problem has to do with updating the kernel.

I will now try uImage.buffalo.bak and initrd.buffalo.bak and return with the results.

1000001101000, 2020-04-24:
Excellent. I'll look into what needs changing for the DTB (still assuming that's the issue). 

It might be time to start a Bullseye section of the site. Reworking the kirkwood DTBs in general is on my todo list.

riptide981, 2020-04-24:
> Nice!
> Did you notice if the update involved installing a new kernel? One possibility is that something with 5.4 is incompatible with the custom device tree causing the boot to fail. honestly, that's [...]

It works like a charm!!! The LS-QVL is up and running.

Thank you so much!

1000001101000, 2020-04-24:
Great!
I’ll take a look at that kernel/device tree when I get some time and will let you know what I discover

1000001101000, 2020-04-30:
out the mainline 5.5 kernel + mainline DTB on mine and found it worked just fine (other than still having the fan speed bug that mine fixes). Eventually I'll get to the bottom of why mine doesn't like the the latest kernels and come up with some sort of fix.