1000001101000/Debian_on_Buffalo #42: Will it work on Buffalo LinkStation 520 LS520?

aalbul, 2020-05-01:
Hi, I am curious if it will work on LS520.
Beside that, is LS220 powerful enough to run Docker?

Regards, Alex

1000001101000, 2020-05-01:
Good to hear from you!

I have not made much progress on LS520 support so far. I'm not sure the degree this hardware is supported by Debian or not (being broadcom rather than marvell like the others) but having looked at the provided kernel source a while back I suspect it would be workable. the main issue is the device boots from the internal NAND which makes replacing the software much more difficult than the other devices. The firmware is also more restrictive than other devices, at least in terms of starting up telnet/etc. Though I have successfully executed commands using the "-s" option of acp_commander:
https://github.com/1000001101000/acp-commander

To make further process we'd need someone to figure out how to connect/enable the serial console on this device so that we could access the bootloader. I could then work on modifying it to boot from disk and add it to the existing process or come up with a Debian image to flash to the onboard NAND.

Digging into this further has long been on my to-do list but it's farther outside my skill set than other things in the backlog currently

I do have some pictures of the board and other notes here:
https://buffalonas.miraheze.org/wiki/Linkstation_LS520D

I've never tried docker on any of these devices (I've only started working with it fairly recently). My understanding is that docker doesn't add all that much overhead so I would imagine it work reasonably well on these devices (at least for things otherwise within their capabilities).

1000001101000, 2020-08-07:
I’ve been spending some time with this device and have started making progress. I’ve put together a procedure for getting a root shell over telnet and a process for installing Debian on a usb disk and running it alongside the stock firmware. Those procedures can be found here:

https://buffalonas.miraheze.org/wiki/Linkstation_LS520D

I’ve been researching ways to actually replace the firmware and have started making progress. There’s still some open questions that I need to resolve before I have something working though.

gabrielmateitoma, 2020-12-20:
I've been following the steps from https://github.com/1000001101000/Debian_on_Buffalo/issues/42
It seems like the UDP port for ACP Commander is closed.
Which firmware of the LS520 are you using?

Thank you

1000001101000, 2020-12-20:
I believe 450 was the last version I worked with.

I have noticed the acp service quitting under some circumstances. It might be worth restarting the device to see if it shows up. 

Also make sure you use this version of acp_commander:
https://github.com/1000001101000/acp-commander

gabrielmateitoma, 2020-12-20:
Thanks for reply. Indeed, 450
I've managed to use ACP Commander , it sees the LS520D, I can issue simple commands, but i am stuck on 
" java -jar acp_commander.jar -t 192.168.0.160 -xfer busybox -xferto /tmp"

I've used NMAP to detect open ports:
Scanning 192.168.0.160 [1000 ports]
Discovered open port 80/tcp on 192.168.0.160
Discovered open port 443/tcp on 192.168.0.160
Discovered open port 445/tcp on 192.168.0.160
Discovered open port 139/tcp on 192.168.0.160
Discovered open port 111/tcp on 192.168.0.160
Discovered open port 22939/tcp on 192.168.0.160
Discovered open port 9000/tcp on 192.168.0.160
but i am getting :
A SocketTimeoutException usually indicates bad firewall settings.
Check especially for *UDP* port 22939 and make sure that the connection to your LS is working.
WARNING: The Timeout could also be caused as you specified (parameter -p) to use port 22939 which differs from standard port 22936.
ERROR: Exception: SocketTimeoutException (Receive timed out) [ACP Send/Receive (Packet:80A0 = SPECIAL_CMD_EN_ONECMD)]

I've tried both 22939 and 22936...

[image]

Can you please advise?

1000001101000, 2020-12-20:
What does `java -jar acp_commander.jar -f` show?

gabrielmateitoma, 2020-12-21:
It shows "Found 1 devices(s)"

gabrielmateitoma, 2020-12-21:
Ok, so, the pattern is as follows :
acp_commander -f : finds the device
i run the transfer command, it requests the admin password - which means for me that it has connected - some time passes...and then it give the timeout message
i run acp_commander -f immediatedly and nothing is found

indeed connection is lost...

Tried 2 times to be sure - always the same

What can i check? 

One further question: which is the most verbose log debug setting?

1000001101000, 2020-12-21:
nice.

That tells us that the devices is responding to discovery packets at least. That should mean there aren't any firewall type issues to worry about, at least for sending commands.

It sounds like it's not managing to authenticate. If you haven't already make sure you try rebooting the device at some point. I've seen them stop responding to commands when doing some of this stuff.

Then start with something simple like:
`java -jar acp_commander.jar -t 192.168.1.160 -c "ls /"`

If you're successfully sending commands it should print a directory listing. Once that's working you can try more advanced stuff like the file transfer.