1000001101000/Debian_on_Buffalo #43: Buffalo LS-500 GL issues on Stretch

markmal, 2020-05-08:
Hi "binary"! Thanks a lot for your work!
 
I had LS-500 GL on Wheezy and then on Jessie for years. Jessie got to EoL. 
Couple of days ago I've upgraded it to Stretch. It upgraded well, it works.
But. After couple of hours it starts beeping one separate beep. 
After couple of min it turns on fan. Then after couple of min it beeps 1bps. 
Then after couple of min it beeps solid, gets unavailable on network and I have to turn it off pressing button.
Logs show nothing. SMART and hddtemp show HDD temp 43C until hang.
lmsensors cant find any mobo/SIC thermal sensors. I look into SIC specs, could not find it. But something turns on the fan (is it just electronic?).  
I've turned off smbd, nmbd services I don't use (I use NFS). No help.
 
I've opened box, the SIC is quite hot for my finger even when there is no activity. Just `top'.

Do you know any way to fix it? 
Does it worth to upgrade to Buster? Will it work? 
Will it be better, I mean not loading SIC this much when idle?

1000001101000, 2020-05-08:
Good to hear from you!

I have a LS-GL but haven't done much with it yet. I believe I had it running Stretch to test my python library for interfacing with the onboard micro-controller (which worked) and tweak the device tree to work on the LS-QL (which didn't). I don't recall seeing any real issues but don't think I did much with it.

I'd recommend looking at the SMART data for relocated/pending sectors to determine if the drive is starting to fail. I think that can also cause drives to run hotter which could be adding to your troubles.

The fan and temperature sensors are all controlled by the micro-controller on this model (which micro-evtd is probably managing). In fact, if the micro-evtd service is failing to start it could explain some of this other behavior (it also manages a watch-dog timer which could explain shutdowns).

Try running the following command to disable the watchdog timer:
`micro-evtd -s 0003`

The service is probably failing because /run is too small, I've added a fix for that to my installer which you can try adding to your device by:
download /Tools/runsize.sh to /etc/initramfs-tools/scripts/init-bottom/
chmod +x /etc/initramfs-tools/scripts/init-bottom/runsize.sh 
update-initramfs -u

You can also try running my fan_daemon script (Tools/micon_scripts/micon_fan_daemon.py) in debug mode which will print out temperature and fan speed every few seconds.

markmal, 2020-05-08:
Cool! I'll try what you advise.
BTW, with open case SIC T is 45C while idle. Is it high or normal?
Disk is also 43C. It is spinning after upgrade. I'll check how can I set smart again to stop it spinning.

1000001101000, 2020-05-08:
fstab works, if you've already done that there's no need for the initramfs thing. It's probably even the better way to do it. 

I've been working on a new testing image for stretch/buster and one of the recent changes was to have it check /run in the initramfs for every device rather than hardcode it at the end of the installer. This way I can move a disk from a device that needs it to one that doesn't without changing anything. 

It depends on the device, my LS-QVL is reporting 46 from the SoC sensor right now which is reasonable for the CPU at idle. I think that's on the higher side for a drive, the one I just checked in 33 but it is idle and in a cool basement.

markmal, 2020-05-08:
I run micon_fan_daemon.py
it is silent
I do not have /sys/devices/virtual/thermal dir
micro-evtd is Active

1000001101000, 2020-05-08:
Run apt-get install python3-serial

markmal, 2020-05-09:
yes thanks, just figured it out. `pip3 install serial` and `pip3 install pyserial` did not help.
daemon turns fan off and on and plays music.
Does the daemon show temperature?
How to turn the music off?

1000001101000, 2020-05-09:
`./micon_fan_daemon.py debug`

It may not work with this device, when i have mine online next i’ll investigate that.

You can dump the raw data from the microcontroller to see if the temperature data is in there using micon_dump.py from:
https://github.com/1000001101000/Python_buffalo_libmicon

Info on what the values mean can be found here:

https://buffalonas.miraheze.org/wiki/Terastation_Microcontroller_Interface

markmal, 2020-05-09:
cool, thanks!

1000001101000, 2020-05-09:
I think it’s a board sensor, possibly even internal to the microcontroller.

1000001101000, 2020-05-09:
I pulled out mine and tried it out with my new test image, it’s now online running Stretch. It’s reading at 27c in a pretty cool room with it’s side open and an old ssd.

I’m going to poke around at a few things and then try switching to Buster

markmal, 2020-05-09:
Thanks!

smartctl long test shows "Extended offline    Completed without error".

Opened box stayed whole night without beeps and hang. So, it is definitely thermal issue.

I am not sure what is watchdog and why it needs to be disabled.

Is it possible dynamically decrease frequency of SIC when it is inactive, and increase when it is active?

1000001101000, 2020-05-09:
some notes:
Mine stayed stable all night on Buster now that I have that up (mine is in pretty ideal thermal conditions though).

The watchdog timer is part of the onboard microcontroller. At boot the boot loader sends it a signal to tell it that the device is booting, once the OS loads a "boot_end" signal must be sent to the microcontroller to inform it that the boot succeeded. If this doesn't happen it flashes leds/plays sounds and/or puts messages on the LCD and then either shuts down or reboots depending on the model. micro-evtd should be handling that for you (I use a python script in my installs) if that weren't working it would be one possible reason this would have trouble. If there was a problem with micro-evtd that you fixed via that fstab entry or something else it could explain why it became stable, depending on what order various things happened.

I was thinking about this more and remembered that the first device I got on Stretch was the LS-QVL which I did by installing via the old deboostrap method then dist-upgrade to Stretch (I think). For some reason it was never stable, but I later started form scratch with the debian installer it was suddenly stable. I doubt your situation is quite the same but given the history of this device I would probably try a fresh install next. I think that would give oyu a clean starting point for future experiments.

If it is just a thermal issue you can continue to work it from that angle. I don't think this cpu supported any kind of frequency scaling but haven't looked into it. If you have any other drives laying around you might want to see if any are newer or require less power. A 500G drive from 2013 vs one from 2007 have very different properties when it comes to heat.

I haven't run any real tests with mine other than getting it booted on Stretch/Buster. If you'd like I can put an old hard drive in it and run some workloads/benchmarks to compare how mine reacts compared to yours if that would help.

markmal, 2020-05-09:
Thanks! You do not have to test with old HDD.
The original HDD had been replaced somewhere in 2015 to 4TB HGST HUS724040ALA640. Just because I needed more size. 

But some load tests with Buster would be appreciated.

Now I am fine tuning the system with using hdparm and disabling unneeded services.
will see. Currently the problem is that OS writes to /var/log/syslog and daemon.log quite frequently. This keeps HDD spinning. I am thinking even to move /var/log to USB flash drive. I know it would not be that reliable. But it is my home NAS. I do not care much about if system logs lost. What do you think?

My SIC is 88f5182-a2. I looked into datasheet it has some frequency pins, but I am not sure if they are helpful.

1000001101000, 2020-05-09:
I've never tried optimizing one of these to spin-down the drives like that though I've heard of such things. I think there are some settings you can set to make it generate less logs, retain less logs and even log to a ramdisk rather than hard disk. mounting /var/log on a usb drive sounds like an interesting compromise as well. I'm sure you can find info on how to do that online somewhere. If you do come up with a good method let me know, it owuld be a good thing to add to the wiki.

In my youth I performed several case mods to add fans to old PCs as part of overclocking type projects. It would be amusing to try to mount a ridiculously oversized fan to the side of the device for a project.... makes me want to buy a hole-saw.

markmal, 2020-05-10:
Logging became quite rear after I turned off my cron based hdtemp logger. So moving may be /var/log to flash may be overkill.

I read about upgrade to Buster. Docs say Buster will have new NIC names, instead of eth0 it will be something MAC like enx0080c9000284. It concerns me because network is only way we can access Buffalo. I would not want to solder console/debugger or to switch HDD back and forth to edit config offline.
When you installed Buster what NIC name is there? If you did upgrade, did you prepare the system for new names before the upgrade?

1000001101000, 2020-05-10:
The predictable name or whatever it’s called doesn’t apply to these devices, I don’t know if it ever will. The interface name is eth0 on mine.

If it were to change the installer’s autodetect would handle it but i’m not sure how dist-upgrade would work. Hopefully it wont come up.

markmal, 2020-05-10:
> The interface name is eth0 on mine. 
good

markmal, 2020-05-12:
I've upgraded to Buster.
It does not boot. It plays siren and christmas lights. :(

1000001101000, 2020-05-12:
was this via dist upgrade or a fresh install?

1000001101000, 2020-05-12:
sounds like it might not be getting very far in the boot process. One of the many things to watch out for on the older devices is the initrd getting to big for that version of uboot to load. Could you post the output of `ls- lh` in the /boot for the device?

markmal, 2020-05-12:
root@ustudio:/tmp/sdd/sdd1# ls -lh
total 39M
-rw-r--r-- 1 root root 166K Apr 29 05:38 config-4.19.0-9-marvell
-rw-r--r-- 1 root root 155K Jan 20 13:38 config-4.9.0-12-marvell
lrwxrwxrwx 1 root root   52 May 12 00:44 dtb -> dtbs/4.19.0-9-marvell/./orion5x-linkstation-lsgl.dtb
lrwxrwxrwx 1 root root   52 May 12 00:44 dtb-4.19.0-9-marvell -> dtbs/4.19.0-9-marvell/./orion5x-linkstation-lsgl.dtb
lrwxrwxrwx 1 root root   50 May 11 21:07 dtb-4.9.0-12-marvell -> dtbs/4.9.0-12-marvell/orion5x-linkstation-lsgl.dtb
drwxr-xr-x 4 root root 1.0K May 12 00:32 dtbs
-rw-r--r-- 1 root root 8.6M May 12 00:45 initrd.buffalo
-rw-r--r-- 1 root root 5.0M May 11 21:07 initrd.buffalo.bak
lrwxrwxrwx 1 root root   27 May 12 00:30 initrd.img -> initrd.img-4.19.0-9-marvell
-rw-r--r-- 1 root root 8.6M May 12 00:44 initrd.img-4.19.0-9-marvell
-rw-r--r-- 1 root root 5.0M May 11 21:07 initrd.img-4.9.0-12-marvell
lrwxrwxrwx 1 root root   27 May 12 00:30 initrd.img.old -> initrd.img-4.9.0-12-marvell
drwx------ 2 root root  12K Jul 19  2014 lost+found
-rw-r--r-- 1 root root 1.6M Apr 29 05:38 System.map-4.19.0-9-marvell
-rw-r--r-- 1 root root 1.7M Jan 20 13:38 System.map-4.9.0-12-marvell
-rw-r--r-- 1 root root 2.0M May 12 00:45 uImage.buffalo
-rw-r--r-- 1 root root 2.0M May 11 21:07 uImage.buffalo.bak
lrwxrwxrwx 1 root root   24 May 12 00:30 vmlinuz -> vmlinuz-4.19.0-9-marvell
-rw-r--r-- 1 root root 2.0M Apr 29 05:38 vmlinuz-4.19.0-9-marvell
-rw-r--r-- 1 root root 2.0M Jan 20 13:38 vmlinuz-4.9.0-12-marvell
lrwxrwxrwx 1 root root   24 May 12 00:30 vmlinuz.old -> vmlinuz-4.9.0-12-marvell

1000001101000, 2020-05-12:
see how initrd.buffalo.bak is 5M and initrd.buffalo is 8.5M? I'm guessing that is the problem. I don't know what the exact cutoff is on this device but for other's I've worked with it's right around 7MB.

See if you can boot from the *.bak files to get running again on the 4.9 kernel. If that works you can try generating a smallier initrd with:
`echo "MODULES=dep" > /etc/initramfs-tools/conf.d/modules`
`update-initramfs -u`

markmal, 2020-05-12:
How to boot from .bak? 
should I rename initrd.buffalo.bak and uImage.buffalo.bak?
Does boot fail over to vmlinuz.old automatically or I have to re-link?
It is not grub/lilo. What is boot process, is it described somewhere I could refer to?

1000001101000, 2020-05-12:
uboot is hardcoded to look in the first partition of the disk for an ext2/3 filesystem and then load and boot from uImage.buffalo and initrd.buffalo. To boot form the *.bak files you need only rename them.

markmal, 2020-05-12:
Thanks! I'll try.

markmal, 2020-05-12:
I've found what is bigger in Buster initrd. It is lib/arm-linux-gnueabi
Apart from libs are just a little bigger, there are also encryption libs.

`root@ustudio:/tmp/sdd# du -ks initrd-4.9*/lib/arm-linux-gnueabi/* | sort -n | tail -10
136	initrd-4.9.0-12/lib/arm-linux-gnueabi/ld-2.24.so
144	initrd-4.9.0-12/lib/arm-linux-gnueabi/libselinux.so.1
196	initrd-4.9.0-12/lib/arm-linux-gnueabi/libreadline.so.5.2
248	initrd-4.9.0-12/lib/arm-linux-gnueabi/libblkid.so.1.1.0
268	initrd-4.9.0-12/lib/arm-linux-gnueabi/libmount.so.1.1.0
296	initrd-4.9.0-12/lib/arm-linux-gnueabi/libext2fs.so.2.4
336	initrd-4.9.0-12/lib/arm-linux-gnueabi/libdevmapper.so.1.02.1
420	initrd-4.9.0-12/lib/arm-linux-gnueabi/libpcre.so.3.13.3
680	initrd-4.9.0-12/lib/arm-linux-gnueabi/libm-2.24.so
1252	initrd-4.9.0-12/lib/arm-linux-gnueabi/libc-2.24.so
root@ustudio:/tmp/sdd# 
root@ustudio:/tmp/sdd# du -ks initrd-4.19*/lib/arm-linux-gnueabi/* | sort -n | tail -10
324	initrd-4.19.0-9/lib/arm-linux-gnueabi/libmount.so.1.1.0
420	initrd-4.19.0-9/lib/arm-linux-gnueabi/libpcre.so.3.13.3
444	initrd-4.19.0-9/lib/arm-linux-gnueabi/libdevmapper.so.1.02.1
452	initrd-4.19.0-9/lib/arm-linux-gnueabi/**libssl.so.1.1**
568	initrd-4.19.0-9/lib/arm-linux-gnueabi/libsystemd.so.0.25.0
660	initrd-4.19.0-9/lib/arm-linux-gnueabi/libm-2.28.so
740	initrd-4.19.0-9/lib/arm-linux-gnueabi/**libgcrypt.so.20.2.4**
1232	initrd-4.19.0-9/lib/arm-linux-gnueabi**/libstdc++.so.6.0.25**�.1
1296	initrd-4.19.0-9/lib/arm-linux-gnueabi/libc-2.28.so
2052	initrd-4.19.0-9/lib/arm-linux-gnueabi/**libcrypto.so.1.1**
`

I am not sure why they have been added, probably to support encrypted partitions. It is in general not a bad idea, to add some security to NAS (somehow keys need to be passed there though).
But in my case I do not use encryption.
So questions:
- is it possible to exclude encryption support?  
- is it possible to increase bootloader's size of something (is it RAM disk?) to accommodate bigger initrd? 
 or is it possible to have initrd unwrapped on HDD? I am not sure why it has to be unwrapped into RAM?

1000001101000, 2020-05-13:
The only change I made from the default config with mine is the modules=dep setting I shared earlier, it should be that simple. it’s possible this is something that was default back in the wheezy days that is being carried over. It could be a reason to do a clean install (who knows how many such issues could be causing you trouble) or at least purging and re-installing initramfs-tools.

This limit is part of the bootloader. I’m not sure if it’s something compiled into it or part of the environment config. In either case messing with it runs the risk of bricking the device.

markmal, 2020-05-13:
there was modules=dep in /etc/initramfs-tools/conf.d/driver-policy
I've removed that. And purged reinstalled initramfs-tools*
It still creates initrd 8M. May be because I have LVM?

1000001101000, 2020-05-13:
That might explain some of why yours is larger than mine:
`-rw-r--r-- 1 root root 5.5M May  9 12:00 initrd.img-4.19.0-9-marvell`
`-rw-r--r-- 1 root root 3.9M May  9 09:59 initrd.img-4.9.0-12-marvell`

another thing you can try is switching to XZ compression which will make a smaller image via xz compression:
`COMPRESS=xz`

This may not even work for this model since both compressing and decompressing takes a lot of ram.

I'd recommend taking a close look at the config settings and see if there are any other settings that have been set that might need to be removed (like the encryption stuff you found). You could even wipe out /etc/initramfs-tools and re-install initramfs-tools to start from a clean config.

if all else fails you can identify things in the archive that are safe to remove and then create a hook script to automate removing them whenever an initramfs is generated.

markmal, 2020-05-13:
I've found that uBoot sets following cmdline for kernel
root@debian:~# cat /proc/cmdline 
console=ttyS0,115200 root=/dev/sda2 rw initrd=0x00800040,15M panic=5 BOOTVER=1.10

I suspect 15M limits initrd. new unwrapped initrd is 20M

1000001101000, 2020-05-13:
I believe that refers to the packed size rather than the unpacked size.

This is the cmdline on one of my armel devices:
console=ttyS0,115200 root=/dev/sda2 rw initrd=0x00900040,15M panic=5 BOOTVER=2.11

My installer works with that device just fine despite it being 24M when unpacked.

markmal, 2020-05-13:
So, an issue may be with something else. Like naming of disks. Who knows. I think I should start from scratch, and just install the Buster. 
I've installed Debian years ago on empty HDD.
This time I need to be sure it will install into current LVM volume vg1-root, and will not reformat whole HDD, especially vg1-data that has all the backups. Is it possible?

1000001101000, 2020-05-13:
Sounds like a good idea as long as LVM modules doesn't turn out to be the source of your initrd problem.

I don't have much experience with LVM and haven't used it in the installer. That sounds like something that would be possible but I don't know if you'll run into issues or not.

markmal, 2020-05-13:
it looks like in Stretch there was no dependency for libcrypto. I'll check which drivers need libcrypto ...
if it is not hard for you could you try with LVM?

1000001101000, 2020-05-13:
could you send me something that shows you lvm layout and the output of lsblk? I'll take a stab at it.

markmal, 2020-05-13:
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda            8:0    0   3.7T  0 disk 
├─sda1         8:1    0    94M  0 part /boot
├─sda2         8:2    0 465.7G  0 part 
│ ├─vg1-swap 254:0    0   244M  0 lvm  [SWAP]
│ ├─vg1-root 254:1    0   9.3G  0 lvm  /
│ └─vg1-data 254:2    0   3.2T  0 lvm  /data
├─sda3         8:3    0   1.4T  0 part 
│ └─vg1-data 254:2    0   3.2T  0 lvm  /data
└─sda4         8:4    0   1.8T  0 part 
  └─vg1-data 254:2    0   3.2T  0 lvm  /data
mtdblock0     31:0    0   192K  1 disk 
mtdblock1     31:1    0    60K  1 disk 
mtdblock2     31:2    0     4K  0 disk

markmal, 2020-05-13:
I've checked dependencies on libcrypto in binaries of initrd done for Buster (v4.19)

`root@debian:~/tmp/initrd4.19# for F in $(find . -type f); do S=$(objdump -x "$F" 2>/dev/null| grep 'libcrypto'); echo $F $S; done | grep libcrypto`

./usr/bin/kmod NEEDED libcrypto.so.1.1 required from libcrypto.so.1.1:
./usr/lib/arm-linux-gnueabi/libcrypto.so.1.1 ./usr/lib/arm-linux-gnueabi/libcrypto.so.1.1: file format elf32-littlearm ./usr/lib/arm-linux-gnueabi/libcrypto.so.1.1 SONAME libcrypto.so.1.1 1 0x01 0x0f4746b1 libcrypto.so.1.1
./usr/lib/arm-linux-gnueabi/libssl.so.1.1 NEEDED libcrypto.so.1.1 required from libcrypto.so.1.1:
./usr/lib/arm-linux-gnueabi/libkmod.so.2.3.4 NEEDED libcrypto.so.1.1 required from libcrypto.so.1.1:

Basically **kmod** needs this. 
There is no such dependency in binaries of initrd for Stretch (v4.9)

1000001101000, 2020-05-14:
Sounds like progress. Is there a hook script related to kmod in your /etc/initramfs-tools?

markmal, 2020-05-14:
no hook scripts.
but kmod and libkmod used in many places - bin/udevadm, lib/systemd/systemd-udevdm usr/lib/udev/rules.d/80-drivers.rules ...

is there a way to add netconsole to UBoot?

1000001101000, 2020-05-14:
nevermind, it's in /usr/share/initramfs-tools. I had a hard time imagining why this would be needed for kmod but then realized it's probably related to all the new(ish) kernel signing/keychain. I did run across a bug report that made it look like they were going to reduce the number of libraries needed for kmod crpyto.

I have the same libraries in mine but it isn't enough to cause a problem, maybe that plus lvm is making it too big. I'm talking a look.

markmal, 2020-05-14:
I am reading how to read/change UBoot environment vars. Still not clear how to connect to UBoot command shell. Is it possible after boot is done, from normal bash shell?

1000001101000, 2020-05-14:
loading the initrd elsewhere in memory might fix the issue but I'm not certain at this point. This can be done without a serial console using fw_setenv but you  need the serial console to fix things if something goes wrong. 

One good thing is that it is easier than most models to attach a serial console to the device, this would allow you to undue any changes you make if something goes wrong:
https://web.archive.org/web/20180427031703/http://buffalo.nas-central.org/wiki/Add_a_Serial_port_to_the_ARM9_Linkstation

before we go down that route I think we can get the packed size of the initrd down enough to stop the issue by tweaking the compression settings used to create the initrd.

try creating `/etc/initramfs-tools/conf.d/compress` with:
`COMPRESS=xz`
`XZ_OPT=-2e`
`export XZ_OPT`

then run `update-initramfs -u' and compare the size of inird.buffalo and initrd.buffalo.bak. On mine this reduced the size from 5.5M to 4.1M while only adding ~90sec to the creation process. If it gets under 7M I think it's be worth trying a reboot.

markmal, 2020-05-14:
I am thinking to modify kernel params using fw_setenv .
There is netconsole available in UBoot. That will be helpful if it it can be enabled. But probably this will require to rebuild uboot loader image to include netconsole binaries and reflash it. Quite a risky.

I do not think compression can decrease size in RAM. It can on disk. But in RAM initramfs is decompressed, is in't it? Documents say that initramfs is already most effective sort of initrd. System can actually move its pages into swap, if memory is required. 
So, I do not think size of initramfs is a culprit. But rather some UBoot settings for kernel or names udev assigns.

1000001101000, 2020-05-14:
I'd say you try my strategy to confirm whether compressing it a little smaller helps and I will try artificially enlarging my uncompressed size to establish if the unpacked size is an issue.

although uboot supports netconsole that doesn't necessarily mean this ~ 13 year old version supports it or has in available. Likewise any attempt to activate it would involve messing with boot parameters which may result in an unbootable device (or needing the serial console to recover).

If it comes to that I can help get you started.

markmal, 2020-05-15:
with XZ initd is below 7MB. It boots to some point. I hear HDD works. 
But never responds to ping. After couple of minutes it fails with 2bps beeps and red LED flashing.

1000001101000, 2020-05-15:
Disappointing.

I’ll have to try an LVM setup to see if I end up in the same place.

I have a few ideas but they get increasingly complicated.

1000001101000, 2020-05-15:
I think I have an easy-ish solution. The issue seems to be that they added this encryption support to kmod which then requires a bunch of crypto libraries to be packaged with it. It looks like the current marvell kernel doesn't  actually require any of this so I downgraded to the stretch version of kmod:

`wget http://ftp.us.debian.org/debian/pool/main/k/kmod/libkmod2_23-2_armel.deb`
`wget http://ftp.us.debian.org/debian/pool/main/k/kmod/kmod_23-2_armel.deb`
`dpkg -i libkmod2_23-2_armel.deb`
`dpkg -i kmod_23-2_armel.deb`
`update-initramfs -u`

Between that and the xz compression my initrd is down to 3.0MB

A more ideal solution might be to compile the current kmod without the encryption options or adjust the initrd to use the the busybox version of modprobe but I think this should get you back in business.

markmal, 2020-05-15:
Thanks,

but it is still unclear, where is a cause of this size limitation. Is it that parameter in uboot kernel parameters ?
> console=ttyS0,115200 root=/dev/sda2 rw initrd=0x00900040,**15M** panic=5 BOOTVER=2.11

I wander, can it be adjusted? We cannot keep with the older version of kmod forever. 
Initrd is already uses busybox. But for some reason kmod is standalone, not busyboxed, if I understand the things correctly. udev also has dependencies on libcrypto.
You also mentioned that for install you have even bigger initrd. And the size itself is not an issue in this case.

Another option is to compile kernel without initrd, or with a slim empty initrd.

And another option is to put into uImage.buffalo not a real kernel but rather a some 2-nd stage boot loader program that has open(file based) config for kernel(s), and has a failover capability (from new kernel to old one) like lilo or grub have.
What do you think?

1000001101000, 2020-05-15:
I would like to start by saying that I like the way you think. When I first ran into hard initrd size limits on the Terastation III I asked all those same questions and did some fairly interesting things with kernel+initrd, though eventually moved to just reducing the initrd size to <7M.

I think you're mostly right about it being the size of the unpacked archive, i just don't think the 15M size is the actual culprit.  For example the ls-wxl has this in the cmdline `initrd=0x00800040,12M' but it works just fine with my installer which is 28M unpacked. At one point in earlier testing I also changed that value and confirmed it didn't help for the problem I had at the time (though it does need to be bigger than the compressed size of your initrd).

initrd-less kernel is an option, that's actually what I used while trying to figure out the initrd on the TS-III. You'd need to compile in all the sata/lvm modules though which will introduce other potential problems.

The other problem is uboot has really tight restrictions on kernel size, the really old versions found on many orion devices (most likely including ls-gl) will crash if you load a kernel larger than 2M. Even later devices it's more like ~5M.

you might find that changing the memory address uboot loads the initrd to later im memory will solve your problem (i've read elsewhere that helps in some cases). Some of these buffalo devices ignore the boot args in the flash-memory (the ts-III does) but for most it should be possible to do from fw_setenv. It just runs the risk of bricking the device (at least enough to require serial console). 

I think a lightweight custom initrd is a good idea, though it would take a bit of work (and be really hard to troubleshoot without a serial console). Here is an example of someone doing that:
https://github.com/rogers0/OpenLinkstation/blob/404d568117e5a4ec921cda4a2c1cc49859e25a8d/lib/shell_lib#L178

all of these seem more difficult than just ditching LVM, staying with downgraded kmod, or compiling a custom kmod package with the dependency dropped but they also sound interesting to pursue. 

I'd have never got support for the Terastation III working if I hadn't tried basically all of the above at one point though fortunately it led to a process that was less complicated and as such easier for me to publish/share.

markmal, 2020-05-16:
I trust you. I just don't understand how the compressed size of initrd can impact boot loader. Only one idea - may be it tries to load it fully into some segment of RAM for further decompression...?
Were you able to read original u-boot scripts?

What do you think about an idea with 2-nd stage boot loader? this can be just a tiny program that reads params from a config file, and then loads a particular kernel and initrd. It can use any location/size in memory to load kernel and initrd into. Kind of an addition to uboot.
I understand that itself it will be loaded into those 2MB, preallocated by uboot... I need to think more...
I also understand it will require serial console for debugging.

1000001101000, 2020-05-16:
I don’t fully understand it but the Terastation II/III models behave the same. It seems like kernel reserves certain locations in memory that end up corrupting the initrd if they are too large. TS-III doesn’t actually use the env bootcmd and I didn’t get a full serial console so I never tried relocating that way. But ended up doing it in a roundabout way by loading a large (70mb) initrd.buffalo that began with ~40mb of zeros before the real initrd. That required a custom kernel with a hardcoded cmdline, and may have only worked because it had 512MB Ram to play with.

I think chainloading uboot instances is a thing, but I don’t think it existed back when this version was compiled.

Unfortunately, buffalo didnt publish uboot source for many devices which limits our information about it.
https://opensource.buffalo.jp/ls-gl-112a.html

Building something dynamic into an initrd could be interesting but would have limits. I think they may have stopped including kexec support in recent kernels but it’s something to look at.

You can try changing the load location as discussed. You’d need to change it in the cmdline, load command and boot command.

You can see the env information with the following.
`echo /dev/mtdblock2 0x000 0x1000 >  /etc/fw_env.config`
`fw_printenv `

I don’t know what address to try but I think it was around ~40mb for the Terastation III

markmal, 2020-05-16:
I am not saying that chain loading existed. What if we make it?
We put chain loader into uImage.buffalo. So uboot will think it loads kernel, but actually it will load the chain loader. which is small. And then the chain loader will load kernel and initrd into places we can define in a separate config file. 
This chain loader needs to be able to read ext3 /dev/sda1.

1000001101000, 2020-05-16:
I accomplish something somewhat like that for my multi-distro testing but it all happens inside Debian. Basically I have a disk image with different rootfs on different partitions and some scripts to generate appropriate initrd/uImage to switch between then at the next boot. 

You could try something like that or build some sort of menu system in an initrd accessed over ssh/netconsole/serial. You’d still need a kernel for that but you could custom build it with kexec/etc so that you could pivot to other kernels if that’s what you’re after.

This might apply:
http://www.denx.de/wiki/view/DULG/CanUBootBeConfiguredSuchThatItCanBeStartedInRAM

markmal, 2020-05-16:
Thanks I'll take a look.

From what I see in ubutenv it boots kernel img to 0x00100000, then initrd to 0x00800000
so it gives 7340032 bytes for max kernel uncompressed size. If it is bigger, then depending what is loaded next, either kernel or initrd will override the other one.

1000001101000, 2020-05-16:
Technically true, and watching for overlaps like that is important. However, on these older uboot versions the bootloader actually segfaults if you try to load a larger kernel. You can see it happen if you have a console connected.

markmal, 2020-05-16:
I see. too bad. 
May be something is wrong with decompression? Did you try to load uncompressed kernel?

1000001101000, 2020-05-16:
The kernel decompresses itself, it actually happens after the bootloader executes the kernel. The crash i’m talking about is much earlier.

I’ve never seen issues related to kernel decompression but I have for initrd decompression. Ironically for low memory devices (< ~80MB ) I need to use a larger gzip installer image because they actually run out of memory trying to unpack the xz version though I may switch to xz using lower settings thanks to the testing we did the other day.

markmal, 2020-05-16:
interesting. 
but uboot docs tell thet it is uboot bootm command uncompressing kernel. https://www.denx.de/wiki/publish/DULG/to-delete/UBootCmdGroupExec.html

> The bootm is used to start operating system images. From the image header it gets information about the type of the operating system, the file compression method used (if any), the load and entry point addresses, etc. The command will then load the image to the required memory address, uncompressing it on the fly if necessary. 

I am going to try it. When no decompression is required, no memory will be used for its variables, arrays etc... the image should be loaded directly, as it is. I do not know why do we need compression these days. Probably in past it was required so kernel can be stored on one floppy disk.

1000001101000, 2020-05-16:
I think you’ll find it difficult to decompress since it’s a self extracting zimage rather than a zip archive. Unless i’m wrong.

I don’t know what drives the 2mb limit but it’s why debian keeps the marvell kernel so small.

I think what your reading is the modern documentation, you may want to see if the wayback machine has docs for the ~12yr old version we’re using.

Never hurts to experiment in any case.

1000001101000, 2020-05-16:
I don’t actually know if the 2mb limit applies on this device tbh, been a long time since I actually tried on a device. I think most if not all orion5x have the issue but haven’t tested with the lsgl.

[30 more replies]

closed 2020-11-26 by 1000001101000