1000001101000/Debian_on_Buffalo #44: Buffalo LS-Q6.0TL/R5 Compatibility

sleepydance, 2020-05-14:
Hi,

I've just had a go at installing Debian on a Buffalo LS-Q6.0TL/R5, but when I reboot after copying across the installer files, the device displays the four green drive lights and a blue flashing power light for around a minute, then shuts down.

I've used the uImage.buffalo.lsgl Buster installer since it apparently shares the same architecture as the LS-GL.

Is it possible to install Debian on the LS-QL and if so, is shutting down on boot a known issue?

Many thanks, Andy

1000001101000, 2020-05-14:
Good to hear from you!

The LS-QL is strange, I have not been able to get it to boot one of my kernels/installers yet despite the same approach working on the other devices of that era.

I ended up trading someone for one which I have used to do some testing but haven't been able to get far without a serial console. I attempted to add one but found that it was beyond my current soldering equipment and skill set.

There are details on how to accomplish that here:
https://web.archive.org/web/20181210235914/http://buffalo.nas-central.org/wiki/LSQuad_Serial_port

If I had console output from my attempts I imagine I could sidestep whatever the issue is and get it working fairly quickly, without it's fairly blind trail and error (though I managed it on several armada/kirkwood devices that way).

It's possible I'll come up with a way to do so without console access but it's hard to say when/if i'll be able to pull that off (though I will probably make another attempt in the nearish future).

sleepydance, 2020-05-15:
Hi,

Thanks for your quick reply, much appreciated, at least now I can put it to one side. I'll keep an eye out here in case you succeed in the future...

Thanks again, Andy

1000001101000, 2020-07-08:
@sleepydance 

I finally got around to connecting a serial console to this device and have it running Debian Stretch. I'm in the process of creating a device tree for it and should have something ready for you to try in the next few days.

I damaged the board on mine somewhat setting up the serial console and now my power led is broken, I'm also having trouble with the front usb port which might be related. Would you be willing to try out an install on yours and then help me figure out which pin the power led is on (I can provide a script to help)?

1000001101000, 2020-07-09:
The installer images for the LS-QL for Stretch and Buster are now posted.

1000001101000, 2020-08-29:
Let me know if you got a chance to try this out. I'll probably close the issue pretty soon now that the LS-QL is fully supported.

1000001101000, 2020-11-26:
Closing out old issues. This device now fully supported, let me know if you need anything else.

closed 2020-11-26 by 1000001101000