1000001101000/Debian_on_Buffalo #45: Debian Buster LS-QVL Ethernet Speed

VR-Punk, 2020-05-15:
Hi, 
with the original buffalo firmware the network speed was up to 100mb/s, the debian speed is 40mb/s mostly. 

Is there a limit on the debian network drivers ?

1000001101000, 2020-05-15:
Good to hear from you!

short answer no. 

The long answer is that it's difficult to track down performance bottlenecks on these devices and I am yet to find a simple solution to measuring and improving performance. This is because there are a ton of "layers" involved with something like a samba file transfer. It's fairly simple to measure raw sequential R/W to disk and and network throughput which can help identify issues but there are a lot of CPU/memory/filesystem/etc that are harder to dig into.

In the case of the LS-QVL I see more or less full Gigabit performance between it and a PC on the same switch and full speed sequential reads from the disk I have installed: 

root@lsqvl-Buster:~# iperf3 -c testhost -f M
Connecting to host testhost, port 5201
[  5] local 192.168.1.168 port 34972 connected to 192.168.1.178 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  74.1 MBytes  73.8 MBytes/sec   47    315 KBytes       
[  5]   1.00-2.01   sec   102 MBytes   101 MBytes/sec    9    406 KBytes       
[  5]   2.01-3.01   sec  75.0 MBytes  75.2 MBytes/sec    2    450 KBytes       
[  5]   3.01-4.00   sec   102 MBytes   103 MBytes/sec    2    458 KBytes       
[  5]   4.00-5.00   sec   113 MBytes   112 MBytes/sec    0    458 KBytes       
[  5]   5.00-6.00   sec   112 MBytes   112 MBytes/sec    1    458 KBytes       
[  5]   6.00-7.00   sec  73.8 MBytes  73.7 MBytes/sec    0    496 KBytes       
[  5]   7.00-8.00   sec  95.8 MBytes  95.7 MBytes/sec    1    505 KBytes       
[  5]   8.00-9.01   sec   113 MBytes   112 MBytes/sec    0    505 KBytes       
[  5]   9.01-10.01  sec   112 MBytes   112 MBytes/sec    0    505 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   972 MBytes  97.1 MBytes/sec   62             sender
[  5]   0.00-10.01  sec   971 MBytes  97.0 MBytes/sec                  receiver

iperf Done.

root@lsqvl-Buster:~# dd if=/dev/sda of=/dev/null bs=1M count=10240
10240+0 records in
10240+0 records out
10737418240 bytes (11 GB, 10 GiB) copied, 68.3111 s, 157 MB/s

There are lots of performance tuning options available both in Debian and specific to samba/nfs/etc that you can explore. If you find one that is helpful, please let me know so I can link to it in the wiki pages.

1000001101000, 2020-05-31:
Is there anything else you want me to look at?

VR-Punk, 2020-05-31:
Sorry, i will close this issue. Thank you for your help ! But i am going to start a new issue, because of my broken fan management  in a few days or weeks.

closed 2020-05-31 by VR-Punk

1000001101000, 2020-05-31:
Good deal.

Near the top of my list is re-working the device-tree for the QVL to deal with a fan issue and to ensure it works with newer kernels and is maintainable for the future. I look forward to hearing about your observations. 

If you're problem is that the fan GPIO pins are reversed in the mainline device-tree for Buster, you can try my custom one which corrects the issue. I've heard elsewhere that it doesn't work on newer kernels than ~4.19 which is what I'll be looking into next but if you're running 4.19 it might help the immediate issue.