1000001101000/Debian_on_Buffalo #46: Debian Buster on LS 220D

clusterjester, 2020-05-18:
Hello,
some days ago I installed Debian Buster on my LS 220 D Linkstation. Installation worked out pretty well after using the right acp_commander.jar. 

Then, I installed Openmediavault 5. 
This started the problem as after reboot of the linkstation, I wasn't able to ssh to the machine anymore. Could be, I messed up the network interface through the OMV interface. 
Even worse, I can't find the linkstation, neither through my router nor through nmap. Ping says (buffalo used to be the hostname):

ping: buffalo: Name or service not known

I'd go back one step and reinstall Debian 10 again from scratch and get rid of OMV, but I'm out of ideas, how to access the Linkstation at all. 

Does anyone around have an idea?

Any ideas are very welcome!

1000001101000, 2020-05-18:
Good to hear from you!

Recovering from this type of situation can be challenging depending on what tools/skills you have available and what exactly went wrong.

If it's something that can be corrected from the filesystem like a bad entry in /etc/fstab or /etc/network/interfaces you can try connecting the drive(s) to any system running linux and correct it by mounting the rootfs and editing the files. 

If you need to do something more complex like use apt/dpkg to uninstall OMV etc you can connect them to another armhf linux system and access the system via chroot. I think it's possible to do this on a X86 system using qemu but I've never looked into it.

Another option (if you don't have any of the above handy) is to use the installer as a recovery environment. You could either load the boot files to the drives with a  PC or boot into them over TFTP. This article describes how to get the ls220 into TFTP mode and what address/ports to use:

https://web.archive.org/web/20150325081911/https://www.aaronhastings.me/completely-recovering-from-a-bricked-buffalo-linkstation-ls200-series-nas-and-opening-the-firmware-too/

Once you get into the installer you need to start the install process until you get to the partition ing menu (that's when disk are detected and sata/etc drivers are loaded). then select "go back" and select "execute a shell" from the menu. To access your system you then mount it's partitions and then chroot into it similar to this:
`mount /dev/sda3 /mnt`
`mount /dev/sda1 /mnt/boot`
`mount -t proc none /mnt/proc`
`mount -t sysfs none /mnt/sys`
`mount -o bind /dev/ /mnt/dev`
`chroot /mnt /bin/bash`

This will give you a shell to access your system and fix whatever you need. You could probably even start the OMV web interface by starting the necessary services/etc

clusterjester, 2020-05-18:
Hello 1000001101000,
thanks for your comprehensive answer. 
This gives me something to look into. 

I'll let you know how it works out, although it might take a few days. 

Thanks and best regards

clusterjester, 2020-05-22:
Hello 1000001101000,
it seems that the openmediavault network config cleaned /etc/network/interfaces. 
after adding 'iface eth0 inet auto' and rebooting, network came up just fine. 
I used a Salcar USB SATA HDD Dock to access the disks. 
Thank you for your help and the work you put into the project!

closed 2020-05-22 by clusterjester

1000001101000, 2020-05-22:
Nice!