1000001101000/Debian_on_Buffalo #48: Support for TS-HTGL6FE

vKnmnn, 2020-05-24:
Hi @1000001101000 

any chance this will run on this model?
the CPU is a ARM926EJ-Sid(wb) rev 0 (v5l) 
Hardware: MV-88fxx81
MemTotal:       126280 kB

I would just try it, but i don't know how to unbrick the device if something fails

1000001101000, 2020-05-24:
Good to hear from you!

That should work fine with the ts2pro installer files.

The process doesn't make any changes to the hardware/flash/etc so restoring the stock firmware is always possible if desired.

I've got a guide for that process here:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP

vKnmnn, 2020-05-24:
alright, it booted fine and says Terastation arm installer on the display, put it doesn't seem to have network connectivity. no ICMP ping and no open port 22

1000001101000, 2020-05-24:
Nice!
The display code runs before the network detect/ssh stuff. That part takes a few minutes so it could be a timing thing.

vKnmnn, 2020-05-24:
Yes, that might be the case. I'm running the installer now. 
Are there any hints on the partitioning? i didn't find it in the wiki.

1000001101000, 2020-05-24:
Actually yes.

I beleive for this device you need to make the root partition bigger than the stock firmware did. 2G works but i’d recommend more like 5G.

There are strict requirements for /boot that are mentioned in the instructions as well.

vKnmnn, 2020-05-25:
sadly the installer died on me while deleting partitions and creating the first md for /boot and now hangs on 52% forever. i am left unable to create partitions so i think i'll have to partition one of the drives with another machine so i can get a booting system again.

vKnmnn, 2020-05-25:
This is a tough device to work on. A few more restarts and i was able to create a swap partition. Problems went away then.
Installation is running now.

I saw you mentioned something about the XOR offloading that you wanted to add for the kernel. Have you made any progress there?

1000001101000, 2020-05-25:
Edit: the E04 message most likely means there was something wrong with your /boot. Could you post the details of your partitions/md devices?

 I may need to add some code to the installer to enable swap early to try to help prevent running out of memory like that. Devices with 64mb need that to even make it to the partitioner but I haven’t seen that on 128mb devices personally.

For this device I determined the SoC does not have dedicated XOR hardware like most others. I was able to patch the kernel to enable it on the Terastation III though. Adding some other missing hardware for that device is among the things I’m working on.

vKnmnn, 2020-05-25:
i partitioned my 4 hdds in the same way:
1G boot raid2    (md4)
1G swap raid10 (md5)
8G /        raid1   (md2)
490G  /mnt/array1 raid5 (md3)

i remember unmounting /boot when the partitioning was finished, to write the ext3 again with the correct inode size, assuming the installer wouldnt do that  automatically. that might have messed things up, right?

is there a way to supply a sort of PIT file, a partition table that the installer will apply automatically? navigating the installer is a pain on a slow device like that :)

i noticed some entries matching "xor" in the kernel log, so i assumed there would be an XOR chip.

1000001101000, 2020-05-25:
I adjusted the installer to automatically set the inode size, no manual steps are required anymore. Doing that will prevent your system from fully booting because you would end up with an fstab entry with the wrong uuid (unless you fixed that after reformatting the volume).

The device should still boot if that's the case, it would just hang before starting ssh/etc. Did you put the boot files back in place after you reformatted?

Also, some devices can't handle 1GB /boot volumes. I use 768MB for my armel devices to avoid that issue.

vKnmnn, 2020-05-25:
The Tera station did boot with a 1G boot partition before, so that's not the culprit here.

I did not recreate any file on /boot, because I did the modifications right after partitioning, when the installer was initializing.
I expected the boot configuration to happen last. So right now my boot partition is in am unknown state. It exists but I dont know what's on it. 

You have any idea how I can mount the boot partition now, to put the kernel there?

1000001101000, 2020-05-25:
I'm not exactly sure what would happen in that scenario. likely depends on whether you re-mounted it before it needed it etc.

How to fix it depends a bit on what you have available. I would probably just load the installer and run through it again.

You could probably load the installer, run it to the partitioning menu so it detects disks, then use it to chroot into the environment, mount /boot in the right place, fix fstab and then run update-initramfs to generate new boot files, then unmount and reboot. In theory that would set things right.

vKnmnn, 2020-05-25:
That sounds like a good way to go. The trouble is, though, that I can only boot the installer if I destroy one of the raid1 partitions and put a file system on it.

1000001101000, 2020-05-25:
you could boot into it via TFTP to avoid that, though that may be a bigger pain.

1000001101000, 2020-05-25:
Also, you’ll want /boot, / and swap to all be raid1

vKnmnn, 2020-05-25:
Alright, i have the installer running again.
RAID1 for all mountpoints except the big partition.

now the finish-installation part goes wrong: 
```
May 25 21:39:00 flash-kernel-installer: error: flash-kernel failed
May 25 21:39:00 main-menu[4262]: (process:22597): dpkg-divert: warning: diverting file '/sbin/st
art-stop-daemon' from an Essential package with rename is dangerous, use --no-rename
May 25 21:39:00 main-menu[4262]: (process:22597): dpkg-divert: warning: diverting file '/sbin/st
art-stop-daemon' from an Essential package with rename is dangerous, use --no-rename
May 25 21:39:00 main-menu[4262]: (process:22597): dpkg-divert: warning: diverting file '/sbin/st
art-stop-daemon' from an Essential package with rename is dangerous, use --no-rename
May 25 21:39:00 main-menu[4262]: WARNING **: Configuring 'flash-kernel-installer' failed with er
ror code 1
May 25 21:39:00 main-menu[4262]: WARNING **: Menu item 'finish-install' failed.

```

1000001101000, 2020-05-25:
could you post the output of `cat /proc/cpuinfo`?

vKnmnn, 2020-05-25:
```
~ # cat /proc/cpuinfo 
processor       : 0
model name      : Feroceon rev 0 (v5l)
BogoMIPS        : 333.33
Features        : swp half thumb fastmult vfp edsp 
CPU implementer : 0x41
CPU architecture: 5TEJ
CPU variant     : 0x0
CPU part        : 0x926
CPU revision    : 0

Hardware        : Buffalo Terastation Pro II/Live
Revision        : 0000
Serial          : 0000000000000000
```

1000001101000, 2020-05-25:
could you try manually running flash-kernel and see what happens?

```
chroot /target /bin/bash
flash-kernel

```

vKnmnn, 2020-05-25:
Yes, i will. currently i'm reinstalling, again :)
should be ready in 15 minutes.

1000001101000, 2020-05-25:
If you still have trouble I can try running an install on mine for comparison. 

I actually ran through it with this model fairly recently since this is typically the device I use for testing since it's the only one of this architecture I have with a serial console. Still, I made some changes recently for the 64mb ram devices and could have broken something without noticing (I guess).

vKnmnn, 2020-05-25:
Right, it failed again.
I chrooted to target and ran flash-kernel. /proc wasn't mounted.
bind-mounted proc and repeated the step, flash-kernel exits without output and exitcode 2.

vKnmnn, 2020-05-25:
I am not sure what flash-kernel is doing, but it compares the machine name from /proc/cpuinfo with Variations of the Hardware names.
But none of these match `Buffalo Terastation Pro II/Live`:
```
Machine: Buffalo Linkstation LS-CHLv2
Machine: Buffalo Linkstation LS-QVL
Machine: Buffalo Linkstation LS-VL
Machine: Buffalo Linkstation LS-WSXL
Machine: Buffalo Linkstation LS-WTGL
Machine: Buffalo Linkstation LS-WVL
Machine: Buffalo Linkstation LS-WVL/VL
Machine: Buffalo Linkstation LS-WXL
Machine: Buffalo Linkstation LS-WXL/WSXL
Machine: Buffalo Linkstation LS-XHL
Machine: Buffalo Linkstation LiveV3 (LS-CHL)
Machine: Buffalo Linkstation Mini
Machine: Buffalo Linkstation Mini (LS-WSGL)
Machine: Buffalo Linkstation Pro/Live
Machine: Buffalo/Revogear Kurobox Pro
```

vKnmnn, 2020-05-25:
which means, the corresponding entry for Terastation 2 Pro is not available in /usr/share/flash-kernel/db/all.db

1000001101000, 2020-05-25:
You should also see my buffalo_devices.db with an entry for this device. Usually “machine not found” is a pretty straightforward error message.

Could you output ls /boot or ls /target/boot

vKnmnn, 2020-05-25:
nothing there
```
root@debian:/# ls -a /usr/share/flash-kernel/db 
.  ..  all.db
root@debian:/# ls -a /boot/
.  ..  lost+found
```

1000001101000, 2020-05-25:
Ah, looks like I probably broke something. Flash-kernel is triggering earlier than it should.

Let me run through it on mine and figure out the issue.

vKnmnn, 2020-05-25:
i manually copied buffalo_devices.db from sources directory but flash-kernel still seems to fail. 
I'll wait for your feedback.

vKnmnn, 2020-05-25:
the entry for Buffalo Terastation Pro II/Live is missing the DTB-Id field.

1000001101000, 2020-05-25:
That’s actually intentional, this SoC never got DTB support. It uses the older “machine id” method along with hardware support built into the kernel a long time ago.

vKnmnn, 2020-05-25:
ah i see. so the script fails on the check ` test -e /boot/vmlinuz* `

1000001101000, 2020-05-25:
Yeah. Normally the debian installer would just install the default kernel then run flash-kernel. Mine is supposed to skip that step so that it can choose which kernel to install later. It looks like I messed that part up somehow.

vKnmnn, 2020-05-25:
yes, i saw in the logs, it tried to find a suitable kernel to install and couldn't find one

1000001101000, 2020-05-26:
Is this buster or stretch?

vKnmnn, 2020-05-26:
Buster

1000001101000, 2020-05-26:
I just ran through a Buster install on my TS-HTGL using the installer and kernel downloaded straight from github. I didn't run into any errors.

I could understand if it failed randomly due to a momentary network issue but I believe you said you tried twice and ran into the same thing. 

At this point I would recommend downloading a fresh copy of the installer files (just in case) and starting the install again. If you hit the issue again download the /var/log/syslog and post it.

1000001101000, 2020-05-26:
[image]

vKnmnn, 2020-05-26:
Starting a new install now.

vKnmnn, 2020-05-26:
I ran a new installation of Buster with the latest initrd and uImage from github: 
```
$ sha256sum initrd.buffalo uImage.buffalo.ts2pro 
85c7ec7bab04154d45e95cf7f5d56fe093f886af5f4772c9aa993ed75f657aa5  initrd.buffalo
b7eb685b5d7989eee501dce6669341ebdb58dead0347cc29ff52ec0ab8c0bc21  uImage.buffalo.ts2pro
```
Here's a few things i noticed:
- kernel 4.19 gets installed and configured, then is deleted later. there could be time saved here.
- shortly before flash-kernel gets run, there's a command that changes the display on the station to alternate between `Terastation arm Debian Installer` and `System Now Shutting Down` which is a bit early, because there are still more steps to be done and the device isn't actually shutting down.
- the installer downloads from US-Mirrors, now matter where we are, which sometimes results in very slow speeds.

I think part of what went wrong before is that somewhere, the automatic installer had an error occur and dropped to the advanced installer, which doesn't seem to be handling that case well in regards to the modifications you made (extra packages and such)

nevertheless, on the first boot after installing, the installer boots again.
so i take out the drive where the installer is on.
then i get `Loading Kernel` for about 5 minutes, before showing `E04` again.
now i'm trying to repair the boot partition.

vKnmnn, 2020-05-26:
Okay, i have successfully rebuilt the raid and i have a successful boot.

where does this leave us?

1000001101000, 2020-05-26:
That mostly makes sense. I have seen the installer fail because of network/timeout issues once or twice in testing. It seems like most of the apt-get type commands don't have any retry logic. 

I really hate having to hard-code the mirrors but the ssh based installer requires a mirror is set so that it can download components required to start ssh/etc. You could always build your own images using my scripts and change the hardcoded mirror (I don't know if anyone has tried that yet)

unfortunately taking a while to boot is normal for these devices. If you watch the serial console you can see that this is mostly uboot loading and searching each drive for boot files then finally loading and booting them from a drive. The actual boot process for debian goes pretty fast. Actually, now that you fixed your swap issue it should go faster, I think by default it will wait 90 seconds if fails to detect the swap device before moving on with out it

I don't understand what happened to your swap, I hope that hasn't been happening to others without them noticing. 

The on-board microcontroller decides whether the device powers back on after being shutdown, this means you have to send it a command to tell it you want to restart rather than shutdown at the end of the install process. That command also adds the shutting down message. I'm currently run it in the last custom script of the install process which is why it displays for so long. I might be able to move it to the actual shutdown scripts to prevent that in the future.

I never did find a method to prevent it from installing the default kernel, instead I added logic to uninstall it and replace it with the custom kernel for devices that need one. All The Terastation-III devices need one because they aren't supported in the default kernel. The Terastation-II needs one for Buster because Orion5x PCI support is broken in kernels after 4.12 (they completely overhauled PCI in 4.13) . They actually broke the ethernet driver at about the same time then fixed it in ~5.4, my custom kernel includes a patch that fixes it.

The path issue is normal for Debian Buster. Presumably for security reasons regular users don't have sbin on their path.

Sounds like you're up and running. Let me know if you run into anything else I can help with.

vKnmnn, 2020-05-26:
Thanks for giving these details, that's very helpful.

I had started the installation of OpenMediaVault and left it running. I just came back home to hear the device beeping it's E04 message again...

so i attached one of the disks to another computer to see the contents of /boot
```
$ sha256sum .mnt/*
657d3a8b9911a339f027b57c3f1048e6301bd40b31a969757f40ad89e70f7e61  .mnt/config-4.12.13
2eb6b71e84697f7119ad962b3d63fd6b9d134cc0d3f5d52816ac3278059ee59c  .mnt/initrd.buffalo
20d46b5a59e26d794304073f494e3893cb9af29f33149f4f504ea6ac3948f705  .mnt/initrd.img-4.12.13
cf0e80d624fea392c3e4a8273975a4398caed29b071181d8e9dc9fcc6c02ff9d  .mnt/System.map-4.12.13
32ce360253430af6e2bcc862ceb0921384378132c572bedda21c6dab44b1252a  .mnt/uImage.buffalo
8bfaa601ef4c9e5e38f868687040636d1eaa5e25ab065bebb8d472f62ea8fd1b  .mnt/vmlinuz-4.12.13

657d3a8b9911a339f027b57c3f1048e6301bd40b31a969757f40ad89e70f7e61  .mnt/config-4.12.13
2eb6b71e84697f7119ad962b3d63fd6b9d134cc0d3f5d52816ac3278059ee59c  .mnt/initrd.buffalo
20d46b5a59e26d794304073f494e3893cb9af29f33149f4f504ea6ac3948f705  .mnt/initrd.img-4.12.13
cf0e80d624fea392c3e4a8273975a4398caed29b071181d8e9dc9fcc6c02ff9d  .mnt/System.map-4.12.13
32ce360253430af6e2bcc862ceb0921384378132c572bedda21c6dab44b1252a  .mnt/uImage.buffalo
8bfaa601ef4c9e5e38f868687040636d1eaa5e25ab065bebb8d472f62ea8fd1b  .mnt/vmlinuz-4.12.13
```

that looks alright to me. why did it die again?

vKnmnn, 2020-05-26:
Also, i don't get any output on the serial port. Am i using the wrong settings? I'm on 115200, 8N1, NOR, VT102

1000001101000, 2020-05-26:
That looks alright to me too. 

Randomly shutting off and beeping like that typically means the watchdog timer was not disabled properly at startup. I've put a fair amount of work to try to make that process bullet-proof but it sounds like it failed. Next time you boot run `systemctl status micon_boot.service` to see the status of that service. I'd be interested to know if it just failed randomly (it's supposed to retry a lot of times before giving up) or if OMV/etc messed up the python dependencies etc.

I've been thinking of adding logic to have it send the signal in the initrd to make it even more reliable. I've actually got a script that does that and also enables the serial console that I haven't added to the installer if you're interested. I haven't decided what to do with it yet since some folks might want to use that serial port for something else I haven't made it a default.

Here are my minicom settings:
```
+-----------------------------------------------------------------------+                 
| A -    Serial Device      : /dev/ttyUSB0                              |                 
| B - Lockfile Location     : /var/lock                                 |                 
| C -   Callin Program      :                                           |                 
| D -  Callout Program      :                                           |                 
| E -    Bps/Par/Bits       : 115200 8N1                                |                 
| F - Hardware Flow Control : Yes                                       |                 
| G - Software Flow Control : No                                        |                 
|                                                                       |                 
|    Change which setting?                                              |                 
+-----------------------------------------------------------------------+ 
```

1000001101000, 2020-05-26:
The watchdog timer on this device is only a few minutes, which is frustrating but at least gives an error message reasonably quickly. On the TS3400 series the timer is set to 4 hours and triggers a silent reboot. It took me months to notice and understand what was happening. 

I didn't know anything about the older devices at the time. I ended up writing the python library to deal with it before realizing others had already dealt with this via micro-evtd and custom kernel code. Although I'm fond of the python library I should probably re-work the startup process to use micro-evtd which seems more reliable overall.

vKnmnn, 2020-05-26:
Do you know of a way to get the tftp-boot process to work from linux? it's a real pain having to take out the hdd everytime something goes wrong and resyncing the array.

i saved the omv-install log. it doesn't seem like an unplanned shutdown. the omv-setup did it, i think.
[omv-install.log](https://github.com/1000001101000/Debian_on_Buffalo/files/4683660/omv-install.log)

Now i'm working on getting it back to booting, hoping i don't have to reinstall yet again.

vKnmnn, 2020-05-26:
aha! now that i have the serial console i can see that the kernel does boot indeed, so the boot partition is still alright.
but the machine is running out of memory on boot! So maybe the swap partition still doesn't get mounted.
```
Out of memory: Kill process 244 (systemd-network) score 5 or sacrifice child
[   45.043177] Killed process 244 (systemd-network) total-vm:14588kB, anon-rss:644kB, file-rss:0kB, shmem-rss:0kB

```

1000001101000, 2020-05-26:
might explain why the watchdog script is failing too. 

I've been using an initramfs hook script to activate swap before starting the system to avoid that type of issue. I usually only run into it when test installer images becuase I end up changing the UUID of the swap partition. I don't know you keep running into this but I think that script could help you too.

I'll post that and the serial console script along with setup instructions in a little bit.

1000001101000, 2020-05-26:
I added the files to Tools/
copy install-micro-evtd /etc/initramfs-tools/hooks/
copy enable_serial.sh /etc/initramfs-tools/scripts/init-top/
copy early_swap.sh to /etc/initramfs-tools/scripts/init-bottom/

then run `initramfs-update -u`

vKnmnn, 2020-05-26:
i had to change `MODULES=dep` to `MODULES=most` in `/etc/initramfs-tools/conf.d/modules` in order to run `update-initramfs -u`

the early_swap script of course throws a bunch of errors due to missing devices, i suppose that can be ignored. maybe we can test if we're actually in an initramfs-session and only then run the script? I'm not sure if i understood that script's purpose well enough. It's not really supposed do be run by update-initramfs, right?

anyway, upon reboot, the initrd is broken, doesn't get recognized by u-boot. I'm in tftp mode now.

1000001101000, 2020-05-26:
without `MODULES=dep` the resulting initrd will be too big boot on this device. What error made you change that?

all the rest can be ignored, I had actually forgotten about that part. I don't know why it executes that stuff when adding to the initramfs, I can probably add some logic to suppress the errors and/or figure out how to make it stop doing that.

vKnmnn, 2020-05-26:
first it fails, because sys and proc aren't mounted in chroot. then, when they're mounted, this happens:

```
root@debian:/# update-initramfs -u
update-initramfs: Generating /boot/initrd.img-4.12.13
mkinitramfs: failed to determine device for /
mkinitramfs: workaround is MODULES=most, check:
grep -r MODULES /etc/initramfs-tools

Error please report bug on initramfs-tools
Include the output of 'mount' and 'cat /proc/mounts'
update-initramfs: failed for /boot/initrd.img-4.12.13 with 1.
```

1000001101000, 2020-05-26:
that's weird. could you post /proc/mounts?

1000001101000, 2020-05-26:
...this may be corrected by mounting /dev/ as well. I thought it was a different error in that case.

`mount -o bind /dev /target/dev`

vKnmnn, 2020-05-26:
i was able to pull out the syslog of the installed system. I'll put it here, in case you wanna take a look at it.
[terastation_syslog.log](https://github.com/1000001101000/Debian_on_Buffalo/files/4684707/terastation_syslog.log)

I think that the systemd service file for micon is broken and that's why it shutdown. 
```May 26 15:38:45 debian systemd[1]: /etc/systemd/system/micon_boot.service:5: Unknown lvalue 'Restart' in section 'Unit', ignoring
May 26 15:38:45 debian systemd[1]: /etc/systemd/system/micon_boot.service:6: Unknown lvalue 'RestartSec' in section 'Unit', ignoring```

this is /proc/mounts from chroot:
```
root@debian:/# cat /proc/mounts 
/dev/md2 / ext4 rw,relatime,data=ordered 0 0
/dev/md0 /boot ext3 rw,relatime,data=ordered 0 0
none /sys sysfs rw,relatime 0 0
none /proc proc rw,relatime 0 0
```
yes, the bind-mount of /dev helped. funny, that it warns about sys and proc not being mounted, but not dev :)

1000001101000, 2020-05-26:
that is funny. In some circumstances (or versions?) it specifically says that it could not find in /dev/disk-by-id/.... 

The systemd service being broken sounds troubling. Is your display being set with the debian version/etc?

vKnmnn, 2020-05-26:
I'm on the next boot-cycle. the new initrd still doesn't do the trick. I'll try rebuilding it after removing those extra three files.

i'm not sure what you mean by the last sentence. When i had a successful boot of the system, it did say Terastation ARM Debian 10.4.

vKnmnn, 2020-05-26:
I took a look at the service file. It has double Restart and RestartSec entries, but that shouldn't keep the service from running.
```
[Unit]
Description=Tell Terastation it finished booting so it doesnt shutdown
Wants=network-online.target
After=network-online.target
Restart=on-failure
RestartSec=30

[Service]
ExecStart=/usr/local/bin/micon_scripts/micon_startup.py
Restart=on-failure
RestartSec=30

[Install]
WantedBy=multi-user.target

```

1000001101000, 2020-05-26:
That's what I was asking, if the display is getting updated then the service is running successfully. I'll fix the duplicate entry to clear up those errors though.

1000001101000, 2020-05-26:
are you trying to build the initrd from a chroot in the installer? I would expect that to work without much issue (once you switch back to MODULES=dep).

[39 more replies]

closed 2020-08-29 by vKnmnn