1000001101000/Debian_on_Buffalo #49: Terastation System Error E04 can't load kernel. Then EM Mode

andybjackson, 2020-05-26:
Having succeeded with getting Debian Buster onto my Linkstation LS-VL, my efforts with Terastation. I have enough working parts for one between two castoffs (a HS-DHTGL & a TS-HTGL).

The install went well (after learning the hard way, see PS below). The resulting installer logs look good. I can’t find any logs for the failing boot attempts.

Following install, the system ends up in EM Mode panic. Once after loading kernel for some time and then throwing "System Error E04 can’t load kernel".

I have tried multiple times with multiple disk numbers (1, 2 and all 4) and partition formats.

The latest and most apparently successful incantation is as follows.

The folder /boot exists on the first partition which is formatted as `ext3` and marked boot.

[root@sysresccd /mnt/sda1/boot]# ls -al
```total 16848
drwxr-xr-x  2 root root    4096 May 26 07:23 .
drwxr-xr-x 18 root root    4096 May 26 07:18 ..
-rw-r--r--  1 root root  164677 Oct 13  2019 config-4.12.13
-rw-r--r--  1 root root 5674948 May 26 07:23 initrd.buffalo
-rw-r--r--  1 root root 5674884 May 26 07:23 initrd.img-4.12.13
-rw-r--r--  1 root root 1549363 Oct 13  2019 System.map-4.12.13
-rw-r--r--  1 root root 2063608 May 26 07:23 uImage.buffalo
-rwxr-xr-x  1 root root 2063536 Oct 13  2019 vmlinuz-4.12.13
```

/etc/fstab:
```
# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>   	<dump>  <pass>
# / was on /dev/sda1 during installation
UUID=43f3a284-1eec-4aa9-877c-2f3077dc15f7 /           	ext3	errors=remount-ro 0   	1
# swap was on /dev/sda5 during installation
UUID=867f5c7d-aa4e-4302-b6b5-82bec686db04 none        	swap	sw          	0   	0
```

Any ideas on what I might do next?

PS It may help others to know that when the installer doesn’t complete, it leaves the ‘installer’ `initrd.buffalo` & `uimage.buffalo`. The login is therefore user: installer and password: install. I wasted quite some time investigating why `root` or `[user]` logins weren’t working, thinking the SSH login was to a resulting Debian system rather then the installer.

1000001101000, 2020-05-26:
Good to hear from you!
You don't appear to have a separate partition/array for /boot, This is required for the boot files to install.

If needs to be the first partition, formatted ext3 and either a regular partition or RAID1 array.

andybjackson, 2020-05-26:
Good to hear from you too.

I have tried at least four different permutations, including:
1. A dedicated boot partition on it's own, formatted to ext3.
2. Preserving Buffalo's own partitioning (which was formatted ext4 I think).
3. This one, with the Debian "Guided" partitioning, all in one ext3.
4. Both GPT and MBR partition tables.

I am happy to try again. Would be grateful for a clear steer on:
1. What partition table style to use GPT or MBR. It seems that Buffalo might have it's own different style as I wasn't on occasion able to read the disk using a Linux live boot (like System-Rescue-CD or PartEd Magic).
2. What partitions to create and how to format them: Order, Path, Purpose, Size, Format)? My guess:
```
1 /boot boot ext3 5GB
2 / root ext4 1GB
3 - swap 1GB
4 /[everything else] ext4 [Rest of drive
```
3. What to do beyond putting the 2 installer files into /boot?

Also, where to look for information as to why the installed system wouldn't boot (e.g. log location)? I suspect that something else is at fault (e.g. too many or too few disks installed, incorrect location for Linux kernel, incorrect Linux kernel)?

Your repository and you guidance is admired and appreciated.

1000001101000, 2020-05-26:
I think proper partitions will solve all your problems. 

I would stick to to MBR for this device though GPT might work. 

/boot needs to be: 
<= 1GB or the device won't boot. 
the first partition on the drive(s)
formated ext3 with inode size 128
either a regular partition or RAID1 array with metadata version 0.9

If you create it with the installer it will handle the inode size/metadata versions for you.

I'd say 2G is minimum for the rootfs, I recommend more like 5G.

The boot loader and Debian don't really care how many disks you have, it's just the stock firmware installer that fails because of too few disks.

There is no log when it fails to boot. The installer sets up the rear serial port as the serial console which can be used for debugging. This setting persists through reboot but not after the device is powered off/on.

andybjackson, 2020-05-26:
Thank you! Back on the path to recovery. Terastation requires all four disks to rebuild itself (having tried 1, 2 and only then 4). Now back to where I was when I started - albeit wiser. Will update, and hopefully close this issue shortly. And then write up a pull request to improve the documentation.

1000001101000, 2020-05-26:
always glad to get documentation help!

Feel free to post your partition layout for review before you move forward. That's where most issues happen.

1000001101000, 2020-05-26:
Also, if you want to load the stock firmware with less drives for some reason I have some notes about how to do that on this page:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP

specifically, in between booting in EM mode and running the install:
`java -jar acp_commander.jar -t <device ip address> -c "echo MAX_DISK_NUM=2 >> /etc/nas_feature" `

andybjackson, 2020-05-26:
About to pull the trigger on the following:

- Not sure where to locate sda4?
- And wondering whether root (sda2) should have more space?
- Planning to partition & format sdb, sdc & sdd under Debian.

```
SCSI1 (0,0,0) (sda) - 500.1 GB ATA WDC WD5000AAKS-0       ?       ?
>     #1  primary  999.3 MB  B  f  ext3          /boot    ?       ?
>     #2  primary    5.0 GB     f  ext4          /        ?       ?
>     #3  primary    1.0 GB     f  swap          swap     ?       ?
>     #4  primary  493.1 GB     f  ext4          /opt     ?       ?
SCSI2 (0,0,0) (sdb) - 500.1 GB ATA WDC WD5000AAKS-0       ?       ?
>         pri/log  500.1 GB        FREE SPACE             ?       ?
SCSI3 (0,0,0) (sdc) - 500.1 GB ATA WDC WD5000AAKS-0       ?       ?    
>         pri/log  500.1 GB        FREE SPACE             ?       ?
SCSI4 (0,0,0) (sdd) - 500.1 GB ATA WDC WD5000AAKS-0       ?       ?    
>         pri/log  500.1 GB        FREE SPACE             ?       ?

```

1000001101000, 2020-05-26:
From a "will it boot?" standpoint it looks fine to me.

Once you're comfortable with the process you may want to explore using software raid rather than simple partitions. 

5GB is usually plenty for the rootfs but it depends on your use case.  

Buffalo and other systems will usually name the large partition "array1" or "disk1" etc. For the purpose of the install you could also just skip that part and set something up later after the install.

andybjackson, 2020-05-26:
Thank you.

Added more space to my root partition and left the remaining space free (thinking that I'l experiment with software raid, lvm, btrfs/zfs and the like when I'm confident the host os is solid.

Crossing fingers, toes & the rest.

```
SCSI1 (0,0,0) (sda) - 500.1 GB ATA WDC WD5000AAKS-0       ?       ?
>     #1  primary  999.3 MB  B  f  ext3          /boot    ?       ?
>     #2  primary   64.0 GB     f  ext4          /        ?       ?
>     #3  primary    1.0 GB     f  swap          swap     ?       ?
>         pri/log  434.1 GB        FREE SPACE             ?       ?
SCSI2 (0,0,0) (sdb) - 500.1 GB ATA WDC WD5000AAKS-0       ?       ?
>         pri/log  500.1 GB        FREE SPACE             ?       ?
SCSI3 (0,0,0) (sdc) - 500.1 GB ATA WDC WD5000AAKS-0       ?       ?    
>         pri/log  500.1 GB        FREE SPACE             ?       ?
SCSI4 (0,0,0) (sdd) - 500.1 GB ATA WDC WD5000AAKS-0       ?       ?    
>         pri/log  500.1 GB        FREE SPACE             ?       ?
```

andybjackson, 2020-05-26:
All good. Thank you. Nice touch with the LCD message.
Next steps - document PR, fan control, further partitioning & formatting, maybe with webmin (if enough oomph).

closed 2020-05-26 by andybjackson

1000001101000, 2020-05-26:
The LCD was half the fun of implementing that protocol, doubly so for devices with the RGB backlight.

take a look at the included fan_daemon (I believe there is some documentation in the wiki somewhere). That should be running by default.

Let me know how webmin goes. I believe either webmin or OMV chocked on devices with 128M memory when installing modules. I don't remember the details, might be a good thing to troubleshoot/document for the wiki when you get that far.

andybjackson, 2020-08-30:
Webmin caused me more problems than it solved, even on my LS421VL where performance wasn't an issue. Better to solve the administration by hand (Samba).

1000001101000, 2020-08-30:
That matches my experience, I wish there were better options available for that type of thing.