1000001101000/Debian_on_Buffalo #50: Looks like I bricked mine, any suggestion/recommendations

pankajshrestha, 2020-05-28:
Hello,

I have the ls420d, started the process following wiki and the linked youtube video, while partitioning the system, it kept showing error that it could not partition the two 1 TB each drive, looks like apart from the hard drives that we can remove, it has some storage, I removed all the partitions from the two drives, planning to add them later from linux, then moved forward with creating the necessary /boot, / and swap. The installation moved forward normally and seem to have completed successfully. 
After that I am not able to access the drives, it is not responding to pings or ssh. What are my options? I am able to put the stock firmware again, but how since now I cannot get to the thing? 

Checked to see if the IP may have changed by checking all the IPs in my home network, not able to find Buffalo drive. 

Please recommend how I can move forward. 

Best Regards
Pankaj

1000001101000, 2020-05-28:
Good to hear from you!

One thing worth checking is the LEDs on the front, that will give us a clue to what's wrong.
If it turns solid white it means it completed booting (though could have some sort of config issue)
If it continually flashes white it loaded boot files successfully but could not complete the boot process
if it is flashing red 7 times it means it could not locate the boot partition
if it is flashing red 6 times it means the boot partition was found but it could not find the boot files.

Most failures of that type are caused by something being wrong with the /boot partition (missing,too big, invalid raid type,wrong fs etc).

I'd suggest loading the installer files manually and trying again:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Preparing-Device-for-Debian-Install#option-2-load-files-onto-device-manually

When you finish choosing your partitioning options take a screen shot and post it here before you continue. I can take a look and make sure it appears correct.

Here are the requirements for the /boot partition:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Running-the-installer#installation-notes

If you need to re-load the stock firmware I have the procedure here:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP

pankajshrestha, 2020-05-28:
@1000001101000 , really appreciate your prompt response. 

Right now the Red light is flashing 7 times, please recommend how I should move forward. Really appreciate your help.

pankajshrestha, 2020-05-28:
@1000001101000, I am not able to follow the instructions at (https://github.com/1000001101000/Debian_on_Buffalo/wiki/Preparing-Device-for-Debian-Install#option-2-load-files-onto-device-manually) clearly.

Do I need to take the drive out of the NAS to perform these activities? Are there any details step by step guides available?

pankajshrestha, 2020-05-28:
> Good to hear from you!
>
> One thing worth checking is the LEDs on the front, that will give us a clue to what's wrong.
> If it turns solid white it means it completed booting (though could have some [...]

1000001101000, 2020-05-28:
That tells us you don't have a valid boot partition. You'll either need to create one by connecting the drive to a PC and load the installer files to it or boot into them via TFTP.

there are more details on this page:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP#Prepare_the_hard_drive

The process is the same with debian as with the stock firmware. You basically create a boot partition, load the files and then boot.

pankajshrestha, 2020-05-28:
> ocess is the same with debian as with the stock firmware. You basically create a boot partition, load the files and then boot.

Understood, I will give that a try and update, need to get a usb dock to be able to connect to PC, is there a way for me to use the existing NAS to perform this?

1000001101000, 2020-05-28:
You can TFTP boot into the installer directly which avoids the need to remove the drives.

I haven't done it personally, but others have successfully.

That process is something like this:
1. Install a TFTP server on your PC. I'm hoping someone will try this with tftpd32/64 and write a guide for me to share:
2. Configure it to server the uImage.buffalo and initrd.buffalo files.
https://tftpd32.jounin.net/
3. Configure your PC to have ip address 192.168.11.1
4. connect the NAS to the PC via ethernet cable
5. boot the nas, when the red LED starts blinking press the function button
6. the device should connect to your PC, transfer the files, and then boot from them.

You'll want to then quickly reconnect it to the normal network so that it can get a proper IP address and move forward with the install. It may be easier to do with the stock firmware first since that doesn't require actual internet access at that stage.

pankajshrestha, 2020-05-28:
> You can TFTP boot into the installer directly which avoids the need to remove the drives.
>
> I haven't done it personally, but others have successfully.
>
> That process is something like this:
>
> 1. [...]

@1000001101000 , you are awesome, followed your steps, and I am back in install screen.

pankajshrestha, 2020-05-28:
My current state of disk partitions:

[image]

1000001101000, 2020-05-28:
excellent!

Now you'll need to create your partitions for /boot / and swap.

1000001101000, 2020-05-28:
This is an example of what a raid setup looks like when ready to continue:
https://github.com/1000001101000/Debian_on_Buffalo/blob/master/.images/17-all_volumes.png

pankajshrestha, 2020-05-28:
> This is an example of what a raid setup looks like when ready to continue:
> https://github.com/1000001101000/Debian_on_Buffalo/blob/master/.images/17-all_volumes.png

This is what I have, since I have two disks, I have created array1 and array2

[image]

1000001101000, 2020-05-28:
That looks like it should work. 

In the future you may want to re-do it with boot, / and swap as RAID1 devices but you should probably get more comforatable with software raid before doing that.

pankajshrestha, 2020-05-28:
> That looks like it should work.
> 
> In the future you may want to re-do it with boot, / and swap as RAID1 devices but you should probably get more comforatable with software raid before doing that.

If I want to use the RAID functionality, correct? I was using RAID 0 earlier to use full disk space, so I was not using RAID and I am not very familiar as RAID.

1000001101000, 2020-05-28:
there are a few different reasons to use RAID with these devices. 
RAID1 for /boot / and swap is generally so that the OS is spread across all the disks and will survive even if a disk fails.

For the large data partitions RAID1 can be used for redundancy or RAID0 can be used for maximum space and performance. 

Even if you decide not to use RAID for your data partitions using RAID1 for the system partitions is generally a good idea.

pankajshrestha, 2020-05-28:
> there are a few different reasons to use RAID with these devices.
> RAID1 for /boot / and swap is generally so that the OS is spread across all the disks and will survive even if a disk fails.
>
> For [...]

@1000001101000 , really appreciate your prompt responses and feedback and understood about RAID, and I now know how to do that if I want to. 

I am sshed into the box, all thanks to you. 

I can create a short instruction based on what I did with TFTP, I believe you mentioned that would be good resource, I can create that, please confirm.

1000001101000, 2020-05-28:
Yes! that would be a big help to me. A few sentences and screenshots that I could add to the wikis would help a lot of people in your same situation.

pankajshrestha, 2020-05-28:
> Yes! that would be a big help to me. A few sentences and screenshots that I could add to the wikis would help a lot of people in your same situation.

Here you go:

Loading NAS with firmware using TFTP

1.	Download and Install the TFTP installer (https://tftpd32.jounin.net/), I used the 64bit version (https://bitbucket.org/phjounin/tftpd64/downloads/Tftpd64-4.64-setup.exe)
2.	Setup your network with the static IP as follows:
[IMAGE 1] 
3.	Configure it to server to use the uImage.buffalo and initrd.buffalo files. Screenshot below:
a.	Download ulmage.buffalo and initrd.buffalo into a folder
[IMAGE 2] 
4.	Connect the NAS to the PC via ethernet cable
5.	Boot the NAS, when the RED LED starts blinking press the function button
6.	The device should connect to your PC, transfer the files, and then boot from them. You will see the two files being transferred in TFTP
7.	Then quickly reconnect it to the normal network so that NAS can get a proper IP address and move forward with the install

pankajshrestha, 2020-05-28:
> Here you go:
>
> Loading NAS with firmware using TFTP
>
> 1. Download and Install the TFTP installer (https://tftpd32.jounin.net/), I used the 64bit version [...]
IMAGE 1:
[image]

IMAGE 2:
[image]

1000001101000, 2020-05-28:
Awesome! I'll working on getting this added to the wiki(s). 

At some point I'll have to try do doing the same on a linux system too.

pankajshrestha, 2020-05-29:
Closing this issue, thanks again @1000001101000

closed 2020-05-29 by pankajshrestha

1000001101000, 2020-05-29:
Thanks again for trying out that TFTP utility.