1000001101000/Debian_on_Buffalo #51: LS441D wont boot into Installer

commi, 2020-06-13:
Hi,

i have an issue with a LS441D. It reports in the Buffalo tools as "LS441D-EM445"

i successfully replaced got a root-shell and replaced the .buffalo files in /boot, but the system always reboots into normal operation (webserver and all) an is still running its stock 3.3 kernel (uname -a).

i tried this several times and double checked that the old initrd.buffalo and uimage.buffalo were replaced (hashing everything)

i fact, it does not matter at all what i do i its /boot-partition, i can delete _everything_ there, and it still comes up normal.

i seems to load its kernel directly from its flash, and i do not know how to get i to boot another kernel.

any ideas?

1000001101000, 2020-06-13:
Good to hear from you!

The “EM” part of the hostname indicates that the device is in Emergency Mode. This means that the device is running in a recovery ramdisk. It could have loaded that from the disk or from the NAND depending on your exact setup. Either way this means that the /boot you are seeing is part of the ramdisk and doesn’t persist through ra reboot.

You should be able to get the files onto the physical disk by manually mounting it first. Start by listing existing block devices:
`ls /sys/block` I believe in raid configs it’s usually md0. you can the mount that device some where: `mount /dev/md0 /boot`

You should be able to proceed normally from that point.

commi, 2020-06-13:
ah i did not now it changes its name depending on Emergency Mode, sorry for the confusion. 
Right now it's reporting as "LS441D455". I did my above changes in normal mode (in EM mode telnet was not working anyway for me).

/boot _was_ mount from md0, which was sda1 at the time. my changes also persist between reboots, but seem to have no effect.

i downgraded the firmware to 1.81 now with a factory reset, and will try again. before it was 1.86

commi, 2020-06-13:
Oh, after factory reset it works! so with firmware 1.86 something in way it boots must have changed or i repeatedly overlooked something major:)

so now with 1.81 it works. Installing Debian now.

1000001101000, 2020-06-13:
I’m guessing something happened with your first attempt that caused it to subsequently get stuck in EM mode for your future attempts.

I don’t think the boot process has changed significantly across firmware versions, in fact it’s basically the same as it was on earlier generations of devices for many years.

Sounds like you’re on the right track. Most issues with the install are related to not specifying the /boot volume properly at the partitioning menu. Hopefully it will work on the first try.

commi, 2020-06-13:
Hm could be it, i was just confused because otherwise the system behaved as if it was in normal mode (web ui, mounts, ftp access, buffalo tools icons). maybe the "emergency" kernel from flash did then boot into the normal root.

thanks for helping

closed 2020-06-13 by commi

1000001101000, 2020-06-13:
That actually makes a lot of sense, though I'm not sure I've seen it before. If something was wrong with the *.buffalo files on the disk (or maybe even the partition) it could conceivably start normally from the flash copy of the initrd/kernel since all the processes use the same files. I'll try to remember that in case people have the same issue in the future.

commi, 2020-06-15:
Just some follow-up: 

i successfully installed Debian and did some setting up, but after some changes the LS only booted to EM-mode. I do not know why; i installed additional disks and some partitions on them but left the original boot and root alone.

the only way i found to get it to boot again was with the Buffalo Firmware Updater with `noFormatting=0` in the INI.

After that i had the FW 1.86 booted, replaced the *.buffalo files, but no joy, same behavior as in the initial issue description.

Then i force downgraded the firmware to 1.81 (i keep the old updater), _without_ formatting. That left the /boot partition unchanged for some reason (so the debian-installer was still there) but it now boots into stock 1.81.

I factory reset the LS (with the hardware-buttons). it still boots into stock.

i will completely wipe the disk now, and factory reset the LS again.

1000001101000, 2020-06-15:
I have an idea what might be going on though it depends on a number of assumptions.

My main suspicion is that there are drive(s) that still have stock boot files on them which re superseding your debian boot files when booting and resulting in EM mode.

I have some notes from when I did some analysis on this model a few years ago here:
https://buffalonas.miraheze.org/wiki/Linkstation_LS441D

One of the things I noted is that it seemed to only boot from drive 1 or 2. I do not recall how exactly I confirmed that, but if I was correct that could explain scenarios where it doesn't boot as expected when moving drives around. 

I've never actually determined how it chooses which drive to boot from for this series device. For older models which I've observed the boot process over serial console it goes through an elaborate system of checks to determine if each drive was formatted by the current device or not and then compares the timestamps on each of the boot files before choosing which ones to load and then use.  I'm guessing that whatever logic it uses is choosing one of the other disks to boot from instead of the one you want. 

If I'm right about that the easiest way to avoid the issue would be to restore the stock firmware with all the drives added and then load the installer files to the existing RAID1 /boot and reuse that array for the subsequent installation. You could also make sure any new drives you add are completely blank (or at least don't have a valid /boot) before adding them post-install.

we initially talked about this being the result of booting from NAND. Looking back at my notes I would think it wouldn't actually boot from NAND until you press the function key to initiate a NAND firmware recovery. According to those notes the normal boot order is:
`bootorder=scsi fail usb1 usb2 tftp`

This makes me think it won't automatically roll to NAND, though many of these devices ignore the bootorder in favor of a hardcoded sequence. Some devices (like the TS3000 series) do boot to NAND if it fails to boot from disk first. If that really is what is happening then getting the /boot partitions straightened out on all the disks should make it stop happening. 

Let me know if that makes sense and how your next attempt works out.

commi, 2020-06-15:
wow thanks for the detailed infos:) a can follow that, and it's good to now that the first two disks are important.

but i'm still stumped with this thing; to make sure the other disk do not interfere i did all the steps today with just one disk plugged. so rn i have one disk in there, booted to stock (web ui and all), with an boot partition without stock kernel.

this is after i wiped the disk, let it format itself (boot+pressed button on the nas), and replaced the files, rebooted:

```
till@tbc:/boot$ telnet 192.168.178.64
Trying 192.168.178.64...
Connected to 192.168.178.64.
Escape character is '^]'.

BUFFALO INC. LinkStation series
LS441D455 login: root
Last login: Tue Jun 16 00:40:28 JST 2020 on pts/0
[root@LS441D455 ~]# cd /boot
[root@LS441D455 boot]# ll
total 202648
drwxr-xr-x    3 root     root        4.0K Jun 16  2020 ./
drwxrwxr-x   22 root     root        4.0K Jun 16  2020 ../
-rw-r--r--    1 root     root       43.1K Jun 16 00:42 conf_save.tgz
-rw-r--r--    1 root     root      180.2M Jun 16 00:16 hddrootfs.buffalo.updated.done
-rw-r--r--    1 root     root       13.3M Jun 13 04:18 initrd.buffalo
drwx------    2 root     root       16.0K Jun 16 00:09 lost+found/
-rw-r--r--    1 root     root        4.1M Jun 13 04:18 uImage.buffalo
[root@LS441D455 boot]# lsblk
NAME      MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
mtdblock0  31:0    0     1M  0 disk
mtdblock1  31:1    0    32M  0 disk
mtdblock2  31:2    0   382M  0 disk
mtdblock3  31:3    0    32M  0 disk
mtdblock4  31:4    0    66M  0 disk
sda         8:0    0 119.2G  0 disk
|-sda1      8:1    0 976.6M  0 part
| `-md0     9:0    0 976.4M  0 raid1 /boot
|-sda2      8:2    0   4.8G  0 part
| `-md1     9:1    0   4.8G  0 raid1 /
|-sda3      8:3    0 384.5K  0 part
|-sda4      8:4    0   512B  0 part
|-sda5      8:5    0 976.6M  0 part
| `-md2     9:2    0   976M  0 raid1 [SWAP]
`-sda6      8:6    0 102.3G  0 part
  `-md101   9:101  0 102.2G  0 raid1 /mnt/disk1
[root@LS441D455 boot]# file initrd.buffalo
initrd.buffalo: u-boot legacy uImage, initrd, Linux/ARM, RAMDisk Image (gzip), 13963120 bytes, Mon Jan  1 10:01:01 2018, Load Address: 0x00000000, Entry Point: 0x00000000, Header CRC: 0x6517AA7D, Data CRC: 0xBF502EB8
[root@LS441D455 boot]# uname -a
Linux LS441D455 3.3.4 #1 Fri Jun 19 18:04:05 JST 2015 armv7l GNU/Linux
```

as you can see it runs stock, there nothing mounted or plugged besides this disk, and the /boot/*buffalo files are replaced.

next i will try the exact same thing with two disks in there (because that was the case my last successful installation). i dont like that because i dont want a useless raid1 for /boot, as this makes replacing the disks harder, but i want to know if thats the difference:)

1000001101000, 2020-06-15:
> i dont like that because i dont want a useless raid1 for /boot, as this makes replacing the disks harder, but i want to know if thats the difference:)
I didn't understand this statement.

commi, 2020-06-15:
> > i dont like that because i dont want a useless raid1 for /boot, as this makes replacing the disks harder, but i want to know if thats the difference:)
> 
> I didn't understand this statement.

i meant, if /boot is a raid 1 between disk1 and disk2 a have to clone the partition to a new disk when i want to replace disk2. i think the stock firmware automatically creates raid members for /boot and / on new disks and syncs them, but i would have to so this manually.but thats not important, i'll use what works:)

update: u have now 

- added a blank disk in slot2, 
- wiped disk1, 
- reformatted with the function key on the box, 
- placed the installer, 
- and now it boots into the installer

so exact same steps as the try before but with two disks. this is also the setup that worked the first time (but i broke subsequently).

after the install i'll make disk images to have a know good state and will try to add more disk then.

commi, 2020-06-16:
update: LS runs fine now an survived many reboots.

i have raid1 for boot on disk 1 and 2 and have no GPT on disk 3 and 4 (full disk fs)

my takeaway is: 

- Disks 1 and 2 need to be present and formatted correctly to actually boot into the custom kernel.
- One disk works with stock but /boot seems to be totally ignored. Where it get's its kernel in those circumstances ¯\_(ツ)_/¯ ?

Maybe something else was wrong with my setup, but over multiple tries that was the only variable that changed anything.

From what you wrote yesterday disk 3 and 4 _should_ have no influence, but i did not experiment with that.

I'm quite happy now with the setup and again your repo was super helpful, i probably wouldnt have attempted this without it:)

1000001101000, 2020-06-16:
Sounds like you've got it under control.

I think that I manually tested to confirm the booting from disk 1/2 thing back when I wrote that. At the time that made sense to me but I've learned a lot in the ~2 years since I wrote that and now know now that most similar devices don't work that way. This makes me wonder if I was correct back then or not despite being fairly sure I tested it at the time.