1000001101000/Debian_on_Buffalo #52: [off topic] Serial on LS220D

squarecode, 2020-06-26:
Hello!
Thanks again for providing this awesome stuff!
This might be off topic, but I have trouble finding the Serial interface points on the PCB of the ls220d. Within the Device tree I found an entry about serial and the std out beeing output on there. I want to experiment with this board a little (I have already modified the flash nand flash contents and want to go further). However I cant seem to find the matching testpoints, resistors or header for the serial output. This would be really helpful to debug and modify the boot process. I have already probed around a lot. [Over at the OpenWRT wiki ](https://openwrt.org/toh/buffalo/ls421de)I found another Linkstation with the same Marvell chipset and there the serial was close to the lower left corner of the chip. Therefore I have probed both sides carefully in this region specifically....

 I was wondering if maybe someone over here already found the connections?

Thanks
Stefan

1000001101000, 2020-06-26:
Good to hear from you!

I've never heard of anyone getting a serial consol working on the ls200 series.

I have some pictures of the board and a link to the SoC datasheet on this page:
https://buffalonas.miraheze.org/wiki/File:220_board2.JPG

I know that the serial port is on gpio0/1. It's not clear to me from the datasheet which actual pins that is on the SoC though. There are two testpoints right next to the painted notch on the SoC which look like they could be connected to pin 0/1. I just don't know if pin0/1 = gpio0/1.

Unfortunately, Buffalo likes to hide the serial connection on these devices, I beleive on some models they put the test point underneath one of the chips for some reason. 

Let me know if you do get it working, I'd like to add that information to the wiki.

squarecode, 2020-06-27:
Thanks for your fast response!
I've just made a small script so toggle gpios with a known frequency so that I can probe around, knowing what to look for. This makes it much easier. However I don't have my scope with me at the moment so research online is the only thing that I can do at the moment. 
On kernel.org I found a link list with informations:
```
Armada 370 Flavors:
        88F6710
        88F6707
        88F6W11
    Product Brief:   http://www.marvell.com/embedded-processors/armada-300/assets/Marvell_ARMADA_370_SoC.pdf
    Hardware Spec:   http://www.marvell.com/embedded-processors/armada-300/assets/ARMADA370-datasheet.pdf
    Functional Spec: http://www.marvell.com/embedded-processors/armada-300/assets/ARMADA370-FunctionalSpec-datasheet.pdf
    Core: Sheeva ARMv7 compatible PJ4B
```

 However they are all down. Within these I should be able to find the matching pinout and get a better idea where to look for it, or if I can map the serial to another pin (leds or even testpads that are already discovered). Do you have any documentation other than the datasheet from the site you linked above?

Have a nice weekend,
Stefan

1000001101000, 2020-06-27:
The datasheet I mentioned is one of the above links via the wayback machine. It seems to have everything but the pinout, though it is possible that I just missed it. I just took a look at the functional spec, if has 1000+ pages of useful info, though I didn't see a pinout there either. It's probably in one of those documents, I just can't find it for some reason.

It would be really useful to have someone label what GPIO pin correspond with which test points on a picture of the board. If you do, I'd be interested to know what hardware/techniques I should get to do that for the other model devices.

1000001101000, 2020-06-27:
you can map the uart/console to different pins by modifying the device-tree, though this is only active after the kernel loads (doesn't give bootlaoder access).

squarecode, 2020-06-28:
> I just took a look at the functional spec, if has 1000+ pages of useful info, though I didn't see a pinout there either

Could you give me access to this? You can find my mail address on my profile page on github. This would be awesome!

> It would be really useful to have someone label what GPIO pin correspond with which test points on a picture of the board. If you do, I'd be interested to know what hardware/techniques I should get to do that for the other model devices.

This is what I'm planning to do. I should be able to identify most of them with the tools I have at home. To do this I'll probably modify my pcb a little. Drag important pins of the flash and nand to required levels to protect them against false writing and then probing the pins with a brute force type attack on the gpio pins. For this I'll probably take my scope and trigger on the test point under 'test'. A little test script running on the box will then cycle through the unknown gpios and by clocking them with a known frequency I can set up a pass fail type auto trigger, which will make it clear which gpio in the sequence it was. Repeat that for all the testpoints and we should get them all. Of cause pre-select the ones that are not obvious in the first hand. If this method works, I'll certainly make a tutorial with maybe even only a small logic analyzer that can be bought for about 10€ . But thats future wishful thinking...

> you can map the uart/console to different pins by modifying the device-tree, though this is only active after the kernel loads (doesn't give bootlaoder access).

Yes I've already looked at this. However the processor should be able to do the same by just writing a few registers during the first few boot layers. So by modifying the uboot process we should be able to get the serial mapped to other pins. At least that is what I'm used to on smaller micros. (I should mention, that I only have electronics and little to none embedded linux background)

1000001101000, 2020-06-28:
Here are the links:
http://web.archive.org/web/20140617183747/http://www.marvell.com/embedded-processors/armada-300/assets/ARMADA370-datasheet.pdf
http://web.archive.org/web/20140617183701/http://www.marvell.com/embedded-processors/armada-300/assets/ARMADA370-FunctionalSpec-datasheet.pdf

Buffalo doesn’t publish their uboot source code which would make modifying the bootloader without a serial console pretty difficult. You can modify the uboot environment variables from userspace but without a console you wouldn’t be able to recover if that caused a problem.

I can probably find a tutorial about logic analyzers somewhere, i’ll look into getting a cheap one of these days.

squarecode, 2020-06-28:
Okay thats something to consider for sure... Backing up the spi flash and flashing it with external hardware would be the worst case solution. I just looked at the links, thank you so much I did not know this page before! I found actually the entire pin list. Have to look through it tomorrow (it's already quite late over here). You can find the information on page 66 of the Hardware Spec pdf. 

As soon as I have access to my analyzers I'll try to figure everything out.

squarecode, 2020-06-28:
I just couldn't resist... Here I have found the information:
[image]

now i can sleep peacefully ;-)

squarecode, 2020-06-30:
After working a little on a printout of the pcb bottom, I've probably found the pins. The Chip orientation took a little effort, because the laser engraving is actually not the correct orientation! The arrow in the lower left corner is. To get to this point I took the USB lanes as obvious pins and matched the layout from the datasheet to them. Everything now makes a lot sense and on the top side of the pcb I've immediatly found the serial lanes with the new knowledge.
Later the day I'll do some soldering to verify, but the pins are tied via a 4,7kOhm resistor to the VDDA rail. After removing these we should have a working UART. 
If this indeed works I'll upload a few high res images on where to connect something, so you can add these to your collection.
I'll catch you later.

squarecode, 2020-06-30:
It does work!
As promised here you have an Image describing the pin location and what resistors to remove. Also the boot log with and without interruption:

Too large to upload here in the issue tracker ;-)
https://drive.google.com/drive/folders/1rYEgoIc2LDBapYcMtrX-bgB10CaHhF_L?usp=sharing

1000001101000, 2020-06-30:
Good work! 

I'll throw these on the wiki.

closed 2020-06-30 by squarecode

1000001101000, 2020-06-30:
are there any special tools you use/recommend for the soldering?

I can see the resistor on my board bit wouldn't be able to solder anything nearly that small with my current iron/skills.

squarecode, 2020-06-30:
to be honest I've done this with my portable soldering iron... This is far from ideal but did the job. Although I would consider myself advanced to professional. Maybe this will help you:
To remove the resistors (as they are only needed if you ever wanted to disable the serial again, which you probably dont want to) I often use a lot of fresh solder to completly heat the entire resistor and pull it by the surface tension from the board. This method makes removing parts way easier, because you only have to care about the part getting hot and not about switching fast between both contacts of the part to get it off. For the wireing afterwards, I use thin wire and shorten the exposed conductor to about 1mm after pre tinning it. This ensures that the insulation will not shrink too much during soldering. I've attached a image from my crappy setup at the moment... I don't even have access to an usb to serial adapter at the moment so a raspberry pi does the work of interfacing. So this is soldered with my worst possible iron, please don't judge.... :laughing: 
[image]

squarecode, 2020-06-30:
By the way: The serial interface from your Debian device tree works flawlessly! Input and output as expected

1000001101000, 2020-06-30:
That's good to know! 

The TS1400 devices use UART1 to communicate with an onboard microcontroller which took me a long time to get working properly because it doesn't use the default pins for that port. Once I realized that possibility I was able to sort it out with a little trial and error. I'm glad that isn't needed in this case.

squarecode, 2020-06-30:
Yeah... the pin mux tools for some of the larger processors make the routing job on the pcb a lot easier however the reverse engineering not :laughing: 

My final goal would be to boot the device from the soldered nand flash. It always bothered me that the hdds where running all the time however I was only accessing the device maybe once a day. Changing the hddparams did change it to the better however everytime the kernel decided to write or read something the hdd spun up again. In a period of 5 minutes about 2 times, so no gain in performance there... Therefore I started by looking at the hardware, how to get that done

1000001101000, 2020-06-30:
If you know a good guide for reducing writing (logs/etc) and spinning down drives I'd like to link that in "post install" options. It's among the many things I've been meaning to look into further. 

NAND boot would be a nice option, too bad so many devices don't have it. It might not be too hard to setup, I assume since it can load the firmware from the NAND that all the drivers/config are already part of the stock firmware build. You can probably just write a suitable image to the nand and modify the env variables in uboot.

squarecode, 2020-06-30:
Yes I'll probably never understand why... The money spend on the nand flash is so small compared to the effort they had to put into recovery tools and unhappy costumers because of corrupted Images on faulty hdds. Its only a factor of time really till the device will break down and leave people in anger...

I've only tried to optimize hddparams for a few hours so no real effort there, as the goal of a non active drive till I need it was just not reachable for me.

Yes thats my plan. The nand boot seems to be the best option for me. Although other devices should be able to boot from a USB device instead (I've found that entry in the uBoot envs). This might sound bad for people that like to have another drive connected via usb, however in my opinion the benefits from a splitted os/data structure outweights. (For this pcb I've already considered a possible mod to add another usb Interface in order to use this for booting.)

As I've already mentioned I'm a hardware guy but I've always wanted to play around with embedded linux at the very low level. The problem there is, without a well known hardware you can just not find any support or informations on about how to get started. I have to learn a lot about backing up the old boot files, recovering with the backup and finally about the boot process with uBoot itself. This device will definetly teach me some of that :laughing:

1000001101000, 2020-06-30:
The only advantage I see is that it makes tweaking/replacing the OS a lot easier having it on the disks. Buffalo has pretty much kept to the same firmware/boot processes since ~2006 which is probably why they are only now switching to booting primarily from NAND on newer devices. It's an annoying support strategy, for devices under warranty they would walk users through booting from TFTP to recover devices but never seem to have published an official recovery procedure. I've tried to fill that gap with the wiki.

I believe you are already familiar with the openwrt post regarding booting an ls400 from the nand. I assume a similar process will work for you.

Otherwise, I know at least one other person has replaced the uboot on these devices:
https://sites.google.com/site/shihsung/88fxxxx-soc/ls441de

1000001101000, 2020-07-01:
I ordered a cheap logic analyzer and am looking at soldering irons/etc. I'm curious to see if I can do the same on this device:
https://static.miraheze.org/buffalonaswiki/8/8e/Ts1400d_board1.JPG

squarecode, 2020-07-01:
Cool! Would be awesome if you can get also this board to speak.
https://drive.google.com/file/d/1PsQvEwAY8kNfkNJMtPNO9ujNn87lA9HD/view?usp=sharing
The two vias that I've marked could be your candidate... If you can, measure from them against a connection to a 3.3V connection, the resistance should be in the 1k-10k range if they have also disabled the uart on this device. I took the 3.3V connection from the spi flash, as it's quite large and the pinout is known. But the unpopulated connections on the side of the board are looking like uart and Jtag interfaces. Probably time well spend to check these first. Or even check whether they conduct from these contacts to the marked vias...

1000001101000, 2020-07-04:
I've captured the main led blinking on an ls210, it's a start. Do I need to remove the resistors before I would see a signal on the UART?
[image]

squarecode, 2020-07-05:
So you have a picture from the pcb? I haven't found one on my quick google search. If it's the same pcb as the ls220 then yes you'll need to remove them

1000001101000, 2020-07-05:
It appears to be basically identical 
[image]

It makes sense it wouldn’t work if that resistor is pulling the line high the whole time. I think so anyway.

squarecode, 2020-07-06:
Yes exactly. The datasheet mentions also, that the pins should be pulled up if they are not hooked up to something to prevent glitches. Did you succeed?

1000001101000, 2020-07-06:
Not yet, I'm waiting on some soldering supplies to arrive. I've also started poking around another (older) device I've been trying to add support for. I'm fairly confident I can remove the SMD resistor though I've not had great luck soldering to commercial pcbs when the pads are so small. I ordered some pretty thin solder and some thin tips which should fit my iron. I'm hoping that will help.

1000001101000, 2020-07-09:
I got some of the supplies and was able to get a working serial port on the LS-QL which has allowed me to (finally) get support for it working. 

I'm planning to spend some time with devices that I have open questions about mapping out gpios and trying to find and document UARTs for. Thanks for all your help!

For finding the port on the ls220.... Am I correct that you were able to identify it mainly by the location compared to the documented pinout? How did you confirm that it was the correct one before removing the resistor, was it just that one side of the resistor measured as ~0 ohms to VDD?

squarecode, 2020-07-10:
Sorry for my late reply.

Yes, it was a combination of the pinout matrix given in the datasheet and the application note that unused UART Pins should be terminated with about 1-10k Ohms to the VDDA Rail. The only two possible pin configurations are also listed within the datasheet. After that a close look at the PCB was all it took.

1000001101000, 2020-07-20:
I'm making progress with some of the older devices:
[image]

Still working on my soldering technique for small pads.

squarecode, 2020-07-20:
Wow, I'm really glad that it actually solved some other problems and that you are getting along with the logic analyzer.  The fine soldering comes with practice ;)