1000001101000/Debian_on_Buffalo #53: Buffalo LS-GL issues on Stretch leds won't blink

tctdavid, 2020-08-01:
first thanks for your work that made my old LS-GL back to live again
the install options said that I can find control of the leds but after install success and online I found the leds won't blink
that's not a big problem but it help to see it's blinking than I know it is Crash or not

the directory /sys/class/leds of my LS_GL is empty can you tell me how to fix this by the way I am not good with linux

thanks for your help

1000001101000, 2020-08-01:
The reason for this is a bit complicated but I think the solution you want is pretty easy.

The reason that you don’t see LEDs under /sys/class is because they are controlled by the on-board microcontroller rather than the linux kernel. I’ve got a note about that here:

https://github.com/1000001101000/Debian_on_Buffalo/wiki/Device-Specific-Notes#all-terastations-except-the-ts1200d-and-the-linkstation-pro-ls-gl

I believe your previous install used micro-evtd to manage the fans/leds. Unfortunately, when I was setting this stuff up I didn’t realize that the protocol micro-evtd uses is the same as the TS5000 I was working on at the time. So instead of updating micro-evtd I ended up implementing a communication library in Python and building the needed tools around it. The reason you’re seeing different behavior is because the install is using my tools rather than micro-evt:

https://buffalonas.miraheze.org/wiki/Terastation_Microcontroller_Interface
https://github.com/1000001101000/Python_buffalo_libmicon

Oddly enough when I realized my tools work with the LS-GL (when I got one) I realized micro-evtd used the same protocol. I ended up creating a fork of micro-evtd that supports sending LCD commands which I use inside all my installer images (it saves a lot of space doing that instead of putting a python in the installer image).

originally, micro-evtd was built specifically for the LS-GL and includes a bunch of features specific to it for handling LEDs and buttons. For my tools I just included examples of how to do these things in python since each model is a little different and would require some custom code for each model (something I wish someone would do but doubt i’ll ever have time for).

Since you have a LS-GL you have the option to disable my tools and install the original micro-evtd. That should restore the behavior that you are used to. It’s still available in Stretch/Buster and seems to work.  

```
systemctl disable micon_fan_daemon.service 
systemctl disable micon_boot.service 
apt-get install micro-evtd
reboot
```

tctdavid, 2020-08-02:
thanks for answer me so fast, I can understand the process you describe how you work  to create the install image of LS_GL 
so I know you use Python to control the leds ,but my linux level is at only a little above of what you tell to type ,

So I did disable the two service and install micro-evtd 
but after reboot the box become even more silence ,the bootup ready music is gone ,and the leds just light on solid green

than I  remove the micro-evtd & enable the two services , install python3 python3-pip but no luck with pip3-install pyserial
it show command not found , anyway I reboot it again the box now is still quiet when bootup ready

I used the image of Stretch instead of Buster because my last success install half year ago follow a japen website ,but the system became unstable after running a while ，sometime it can run smooth for a mouth sometimes didn' last for a day ,after that I keep searching solution and then through youtube I found here 

I didn't realized that debian already had a new release ,so would the problem go away if I reinstall the LS_GL with Buster Image  
and thanks for all the hard work you did

1000001101000, 2020-08-02:
I haven’t seem any stability issues with mine on Stretch or Buster... but I only run it with the case open and with an SSD that doesn’t generate much heat. I also typically don’t run much on it. As such I don’t tend to see thermal issues that could affect others.

Running out of memory is another potential issues. Make sure you are using a swap partition.

tctdavid, 2020-08-09:
I will make it short
the buster is OK for LS_GL bootup but not fullly OK with the apps I want to run on this box for I don't have the ability to fix it
(1.samba_OK 2.mldonkey_need_recompiler  3.plexmediaserver_cannot_stop_appropriatly) and the LEDS still not blinks

So I went all the way back from the start to install the stretch and the system now match I need and running smoothly

P.S.
during the test installation I used apt-get install smartmontools try to monitor the hd temperature 
if with the micro_vetd installed it make continue beep at bootup , remove both app can stop the noise but after that led that indecate network will keep flashing even there no network connection and no matter what I try it will not stop

tctdavid, 2020-08-30:
now the box is working great but I don't have time to test the LEDs problem
if I have any new clues I will comeback to report 
thanks for all the hard work again

closed 2020-08-30 by tctdavid