1000001101000/Debian_on_Buffalo #54: will this work on a Terastation TS-2.0TGL/R5

apepper95, 2020-08-04:
i can get a hold of a Terastation TS-2.0TGL/R5 some people call it a Terastation Pro i just want to know if this will work on it 

thanks

1000001101000, 2020-08-04:
Good to hear from you!

It sounds like the device you are talking about is this one:
https://www.buffalotech.com/products/terastation-pro
https://web.archive.org/web/20180427031712/http://buffalo.nas-central.org/wiki/Category:TerastationProV1

If so, It was one of the early 32-bit PowerPC models and will not work with this project. 

Looking at the links above I see that folks got as far as running a custom kernel on it back in the day, though I don't see whether they got any farther than that. It also looks like Debian (and all the distros really) dropped PowerPC 32-bit support a few years ago, the last PowerPC release looks to have been Debian 8.

You could create a USB drive with a Debian 8 install on it which you could connect to the device and run as a "chroot" environment. You could then ssh into that environment and use it like a mostly normal install. I think you could even kill off the services in the stock firmware at that point and have it basically just running Debian. You could even create a startup script in the stock firmware to do this automatically after boot. I'm in the process of writing up how to do something similar for a newer device which can't currently boot straight into Debian (the ls520d).  

I confirmed it's still possible to create such an environment using qemu-debootstrap:

```mkfs.ext3 /dev/sdx1
mkdir /mnt/ppc/
mount /dev/sdx1 /mnt/ppc/
qemu-debootstrap --arch "powerpc" "jessie" "ppc/" http://archive.debian.org/debian/```

Hypothetically it might even be possible to create a kernel patch to allow this device to boot the Debian 8 kernel natively. 

If I had one I might give all of the above a try and write it up as a separate project, I assume the old PowerPC Linkstation would have the same capabilities. It would be an interesting challenge to see how far I could take support for it.

All of that said I wouldn't recommend spending money on this ~14 year old device.

1000001101000, 2020-08-04:
Unless it's really a TS-H2.0TGL/R5 which would be a TS2PRO and is supported. It seems like all the pictures I search for TS-TGL end up really being TS-HTGL. They appear to look very similar but I assume the label on the back would show which it is.

1000001101000, 2020-08-29:
Someone has offered to send me one of these. In the coming weeks/months I'm going to see how far I can get trying to get Debian 8 working on one.

cdwijs, 2020-12-28:
I've just finished installing Debian on my Buffalo Terrastation Pro TS-HTGL/R5 TS-H4.0TGL/R5 BUF-TS-HTGLR5(A).
Enter these commands to put the correct files on a ext3 partition on the first hard drive. Alternative, serve those files via TFTP:
```
wget https://github.com/1000001101000/Debian_on_Buffalo/raw/master/Buster/installer_images/armel_devices/initrd.buffalo
wget https://github.com/1000001101000/Debian_on_Buffalo/raw/master/Buster/installer_images/armel_devices/uImage.buffalo.ts2pro
mv uImage.buffalo.ts2pro uImage.buffalo
```
See these questions for for information
https://www.linuxquestions.org/questions/debian-26/install-debian-on-a-thumb-drive-on-my-buffalo-terrastation-fails-on-hdd-ok-4175687607/
https://community.unix.com/t/install-debian-on-a-thumb-drive-on-my-buffalo-terrastation-fails-on-hdd-ok/381246/2

1000001101000, 2020-12-28:
Good to hear from you!

I believe the OP was referring to the TS-TGL which is an old PowerPC model which won't work with these installer files which are for ARMv5 systems. I've spent a little time working with the PowerPC systems recently but haven't been able to get the sata controller to work with modern kernels (yet).

cdwijs, 2020-12-28:
Thank you. I assumed my NAS was the same model as OP. Apologies for the noise.

1000001101000, 2020-12-28:
It's always good to hear from people using the installers.

I often wish I had some usage statistics since i have no idea how many installs have actually been performed, just how many issues have been opened!

cdwijs, 2020-12-28:
Installing Debian was actually my last resort. I wanted to run arch on it, but Archlinuxarm did not know the device yet.
however, I am grateful for your work, as i now have a NAS I can completely control (within the limits of my knowledge). It's slow, but it makes for a good slave NAS. (I also have a main NAS that dumps backups to this one)

CharlieTemplar, 2021-04-21:
> Someone has offered to send me one of these. In the coming weeks/months I'm going to see how far I can get trying to get Debian 8 working on one.

Is this something that happened?  Just inherited one of these and interested in using any version of debian instead of the original bufalo FW.
If not I'll likely rip out the mobo and replace with a microatx or raspi maybe

1000001101000, 2021-04-21:
Good to hear from you!

I spent a bunch of time at the end of last year trying to get it to work but couldn't quite get there.

Oddly enough I ended up getting an old HD-HLAN running modern Debian on kernel version 4.19 using some of the tools that folks used to run etch/lenny on them back in the day. With some tweaking I was even able to get my old HD-HTGL working too though I had to resort to manually editing the MDADM superblock.

I was able to use the disk image from that effort to boot the TS-TGL but it required some nasty hacks that had to be repeated for each boot the bootloader couldn't read the drives directly. Once booted it mostly worked except that the SATA drivers would crash almost immediately after detecting drives. Searching around it sounded like I was running into the same issues folks got stuck on ~12 years earlier and I more or less gave up. 

It's possible I'd look at that project again but I doubt it, though it was really interesting.