1000001101000/Debian_on_Buffalo #56: LS-QVL Seagate IronWolf 8TB

VR-Punk, 2020-08-29:
hi,

is this 8TB HDD unofficial compatible with the LS-QVL ?

https://www.seagate.com/internal-hard-drives/hdd/ironwolf/

1000001101000, 2020-08-30:
I've not heard of any drives that haven't worked with these kirkwood devices. The main thing I would worry about is heat, though I'm not aware of anyone who had thermal issues with the LS-QVL. It might be worth comparing the wattage of these drives to what you had before. I have heard of overheating the LS441DE with WD Red drives, but that model seems to have worse airflow.  

The other thing to think about is the 16TB volume limit for this architecture. You should be able to do a 16TB RAID 1+0 with those drives but a 24TB RAID5 wouldn't work, for example.

VR-Punk, 2020-08-30:
thx, i didn't know about the 16TB volume limit

closed 2020-08-30 by VR-Punk

1000001101000, 2020-08-30:
For some reason I haven't seen it very well documented. I first learned about it the hard way trying to use 3x 6TB drives in a RAID0 on the LS-QVL several years ago. After a lot of digging I've come to mostly understand the issue and that it affects any 32-bit architecture which uses 4k memory pages (which seems to be all of them). 

Interestingly, I recently noticed that the TS3210 has a customized kernel that allows larger memory pages which I'm guessing also removes the volume limit. I assume they can do that because of something specific to that Alpine processor. Someday I might investigate whether that could be done for other devices but I'm guessing that's beyond my skillset.