1000001101000/Debian_on_Buffalo #57: Can not find base packages.

dewcansam, 2020-09-01:
`W: Failure while installing base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package archive is at fault)
`
# Model Info #
Buffalo
LinkStation Pro Duo
LS-WTGL
LS-W2.0TGL/R1-US v.3

Also found these messages:
`I: Running command: debootstrap --arch armel --foreign --include=flash-kernel,haveged,openssh-server,busybox,libpam-systemd,dbus,u-boot-tools,mdadm,gdisk,apt-transport-https,gnupg,wget,ca-certificates,python3,python3-serial,i2c-tools,xz-utils buster /mnt/target http://deb.debian.org/debian/
W: Cannot check Release signature; keyring file not available /usr/share/keyrings/debian-archive-keyring.gpg`

from generate_disk_image.sh
`target_hostname="lswtglv3"
machine="Buffalo Linkstation LS-WTGL-v3"
arch="armel"
distro="buster"`

Hope that helps. Ask if you need more or (lol) if i am doing it wrong.

dewcansam, 2020-09-01:
I think I might have found it 
$ /usr/bin/python --version 
Python 2.7.6

Even though Python3 is listed as installed.

1000001101000, 2020-09-01:
It's normal for Buster to have Python2 installed as the default even with Python3 installed. 

Can you give me some information about the system you're running this on? I wonder if there is a difference between versions of debootstrap or something.

dewcansam, 2020-09-01:
Linux dewcan 4.4.0-148-generic #174~14.04.1-Ubuntu SMP Thu May 9 08:18:11 UTC 2019 i686 athlon i686 GNU/Linux

NAME="Ubuntu"
VERSION="14.04.6 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.6 LTS"
VERSION_ID="14.04"
32b (x86)

I actually start with a Kubuntu base then screw it all up by adding all the gnome stuff. :)

dewcansam, 2020-09-01:
found this in the debootstrap.log file
```
dpkg: regarding .../init_1.56+nmu1_armel.deb containing init, pre-dependency problem:
 init pre-depends on systemd-sysv | sysvinit-core | runit-init
  systemd-sysv is not installed.
  sysvinit-core is not installed.
  runit-init is not installed.

dpkg: error processing archive /var/cache/apt/archives/init_1.56+nmu1_armel.deb (--unpack):
 pre-dependency problem - not installing init
```
```
dpkg: regarding .../systemd-sysv_241-7~deb10u4_armel.deb containing systemd-sysv, pre-dependency problem:
 systemd-sysv pre-depends on systemd
  systemd is unpacked, but has never been configured.
```

1000001101000, 2020-09-01:
seems like it could be an issue with that version of Debootstrap. You could try adding gawk to the list of packages but I doubt that would fix it. 

since debootrap has very few dependencies you could probably try grabbing debootstrap from the Stretch repo and see if that fixes your issues.

1000001101000, 2020-09-01:
oh, maybe you just need to install that keyring. that would make more sense (that ubuntu 14 couldn't have the latest debian keyring)

dewcansam, 2020-09-01:
I tried switching to strech but got the same issue.
```
W: Cannot check Release signature; keyring file not available /usr/share/keyrings/debian-archive-keyring.gpg
.....
I: Keyring file not available at /usr/share/keyrings/debian-archive-keyring.gpg; switching to https mirror https://mirrors.kernel.org/debian
```

Well I added the keyrings with this:
$ sudo apt-get install debian-archive-keyring debian-keyring debian-ports-archive-keyring
I tried with some other commands first and none seemed to work.

Now it's failing big time with:
```
~/temp/buffaloDeb/debootstrap/Debian_on_Buffalo/Tools/debootstrap/generate_disk_image.sh: line 158: /mnt/target/etc/flash-kernel/machine: No such file or directory
mount: mount point /mnt/target/proc/ does not exist
mount: mount point /mnt/target/sys/ does not exist
mount: mount point /mnt/target/dev/ does not exist
cp: cannot create regular file ‘/mnt/target/usr/bin/’: No such file or directory
```

There are none of the directories there that used to be there. I also tried to switch back to buster but got same result.

Retried to grab the keyring (I had mistyped the command).
```
rsync -az --progress keyring.debian.org::keyrings/keyrings/ .
```

But still got:
```
W: Cannot check Release signature; keyring file not available /usr/share/keyrings/debian-archive-keyring.gpg
```

Added gawk:
```
### CLS 2020-09-01 18.36 +++ added gawk 
qemu-debootstrap --arch "$arch" --include=flash-kernel,haveged,openssh-server,busybox,libpam-systemd,dbus,u-boot-tools,mdadm,gawk,gdisk,apt-transport-https,gnupg,wget,ca-certificates,python3,python3-serial,i2c-tools,xz-utils "$distro" "$target" http://deb.debian.org/debian/
```
still failed.

Changed mirror:
http://mirrors.edge.kernel.org/debian/
still failed.

I have no idea what's going on. Just about out of options. 

So my next question would be can I use the dd image file as it is ? Or do I HAVE to clear these warnings up ?
One was made regardless of the warnings.

1000001101000, 2020-09-02:
I don’t think the system will function until those are resolved.

I think you need this:
https://packages.debian.org/buster/debian-archive-keyring

You should be able to unpack it and the  specify it as a command line option. You might be able to get away with disabling signature checks entirely.

The relavant options are:

  --keyring=K            check Release files against keyring K
      --no-check-gpg

dewcansam, 2020-09-02:
Closed with copy from other post:
Got the img file made finally !! So, I ran the current version of Mint in a vm. I had a couple of hoops to jump through but it was not too much.
In summary building the img file on a current system fixed both issues. May I recommend a version check at the start of generate_disk_image.sh ? Then the script can terminate before running into issues and a warning to update ;)

closed 2020-09-02 by dewcansam

1000001101000, 2020-09-02:
The only issue with a version check is I need to figure out what versions to support. I'll have to take a look and see what's possible.

Either way I need to spend some time working on some error handling.