1000001101000/Debian_on_Buffalo #58: losetup errors

dewcansam, 2020-09-01:
`losetup: invalid option -- 'D'`

I modified my script to:
`losetup --detach 1>>$log 2>/dev/null`
as i also was trying to setup a log file and dump losetup general messages.

1000001101000, 2020-09-01:
I've only tried this on Debian Stretch and Debian Buster. The error you are seeing sounds like you have a different version of losetup than either that I've worked with. What OS/version is your system running?

dewcansam, 2020-09-01:
Linux dewcan 4.4.0-148-generic #174~14.04.1-Ubuntu SMP Thu May 9 08:18:11 UTC 2019 i686 athlon i686 GNU/Linux

NAME="Ubuntu"
VERSION="14.04.6 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.6 LTS"
VERSION_ID="14.04"
32b (x86)

1000001101000, 2020-09-01:
Is that a LiveCD or is that a full install?

dewcansam, 2020-09-02:
Full install on a old Compaq.

1000001101000, 2020-09-02:
that likely means you're dealing with older versions of various tools which may cause some unexpected behavior like you way with the "-D" option. It also explains the lack of gpg keys for Buster/Stretch. I'm hoping that change you made to the losetup and disabling the gpg check will be enough to get it going.

Otherwise you should consider booting that PC in a Debian live environment and doing it from there. That should make these issues go away.

dewcansam, 2020-09-02:
Got the img file made finally !! So, I ran the current version of Mint in a vm. I had a couple of hoops to jump through but it was not too much. 
In summary building the img file on a current system fixed both issues. May I recommend a version check at the start of generate_disk_image.sh ? Then the script can terminate before running into issues and a warning to update ;)

closed 2020-09-02 by dewcansam