1000001101000/Debian_on_Buffalo #60: TS-XEL02 shuts down after sending installer files and restarting device

mholl1983, 2020-09-19:
I have stock firmware and wanted to try the Debian steps to gain SMB2 for shares. I was able to send the two installer files over ACP commander to my TS-XEL02. Then I did the command to restart and now the device does part of the power cycle but then shuts down after a minute. I'm not able to proceed to the installation steps.

mholl1983, 2020-09-19:
I have roughly 2 TB of files in the box, too. Any suggestions for how to proceed so I can install Debian and not lose my files? (And if needed, I can revert to stock firmware, but I want to do that safely.)

mholl1983, 2020-09-19:
The box itself shows "Initialize SATA0 channel" for a minute before I get a message saying that the system is shutting down.

mholl1983, 2020-09-19:
I saw you posted this in the forums:

"If you don't want to wipe the drives you could remove the drives and start the device (I believe it will still make it to EM mode that way). You could then re-insert the drives and attempt to re-install the firmware without reformating/repartitioning (which should preserve your data provided there's nothing wrong with the data partitions)."

Is that a clue for me? Thanks in advance! Sorry for multiple messages; just a bit panicked and eager to get this issue sorted.

1000001101000, 2020-09-19:
Good to hear from you!

A few thoughts:

I'm mostly guessing about specifics right now. If you could post more details like the commands you're using I can better understand what's going on.

Also, are you trying to install Debian or just enable SMB2 in the stock firmware? It sounds like you want to install Debian but I want to be sure before proceeding. I don't know if it is even possible to enable SMB2 for this particular model's firmware.

One reason you may be having trouble is the filenames required for this device. Specifically, the kernel image for this device needs to be named uImage-88f6281.buffalo. If you uploaded it as uImage.buffalo you could have a mismatch between the kernel and initrd which would explain what you're seeing.

Either way it sounds like you have a device which is partially booting and then failing and shutting down. To get it working again you'll need to get a proper set of boot files back in place on the boot partition (uImage-88f6281.buffalo and initrd.buffalo). There are several ways to accomplish this but depend on whether you want to end up with Debian or the Stock firmware. It also depends on what you have available. I'm going to assume oyu don't have an easy way to hook all four drives up to a PC running Linux, correct me if I'm wrong. Let me know If you have a spare drive to test with that would also help.

You're on the right track with that last post. Try booting with no drives and see what it does (The LCD screen is a great help with that). I think it will either boot into EM mode from the NAND (if it has that feature) or go into TFTP mode. Either mode can be used for recovery.

mholl1983, 2020-09-19:
Thank you so much for the reply! Sorry if unclear on details. I'll answer your questions inline.

> I'm mostly guessing about specifics right now. If you could post more details like the commands you're using I can better understand what's going on.

These are the commands I've run so far:

[image]

> 
> Also, are you trying to install Debian or just enable SMB2 in the stock firmware? It sounds like you want to install Debian but I want to be sure before proceeding. I don't know if it is even possible to enable SMB2 for this particular model's firmware.

Yes, trying to install Debian. My hope is (based on your reply in the Buffalo forums) that I can plug the SMB1 security hole that the stock firmware unfortunately requires for an SMB share and then gain a little Linux knowledge by adding back some other functionality like DLNA, print server, etc.

> One reason you may be having trouble is the filenames required for this device. Specifically, the kernel image for this device needs to be named uImage-88f6281.buffalo. If you uploaded it as [...]

You hit the nail on the head. I renamed the uImage file to match the one on the box.

I tripped on this part: 

[image]

When I was jumping around, I must've retained the "rename" part and the filenames in the first sentence. User error! I think you identified the misstep I took. Thanks for that -- a learning experience!

> Either way it sounds like you have a device which is partially booting and then failing and shutting down. To get it working again you'll need to get a proper set of boot files back in place on the [...]

Thanks for reassuring me that not all hope is lost. I'd definitely like to proceed with Debian. Unfortunately, I don't have a Linux PC to hook up the drives. I *do*, however, have a spare drive that I could test with, so do let me know what I could try there.

> 
> You're on the right track with that last post. Try booting with no drives and see what it does (The LCD screen is a great help with that). I think it will either boot into EM mode from the NAND (if it has that feature) or go into TFTP mode. Either mode can be used for recovery.

Thanks for sharing that suggestion. Let me know if I should try the no drives step first or if I should bring in the test drive. All ears; hope I addressed your requests for more info above, but let me know if I can provide further details. And thanks again for the timely help!

1000001101000, 2020-09-19:
Excellent.

I believe your next goal is to get the device to boot into the Debian installer without making any more changes to your main drives. Although the installer is usually loaded from disk it actually runs completely inside the device's memory. This should allow you to load the boot files on your spare drive (following the instructions on that wiki page), boot the device, then when the LCD indicates the installer has loaded you can insert your drives and move forward with the install process.

From there you can run the the install normally. Your data should be safe as long as you leave the data array alone during the installation. when you get to the partition menu You'll need to: 
  set the first raid device as /boot and format as ext3
  set the second raid device as swap
  set the third raid device as "/" and format as ext4
  make sure no changes are made to the fourth array (where your data is stored).

Before you finalize the changes on the partitioning screen, post a screenshot so I can take a look.

mholl1983, 2020-09-19:
Thanks so much! I'm in emergency mode w/ the single drive, about to transfer the files. Should I use my existing password or does that go out the door where I botched the firmware?

mholl1983, 2020-09-19:
The files seem to transfer but the station still goes into emergency mode. Should I format the test disk in any particular way?

1000001101000, 2020-09-19:
The process gets a little bit different if you are in EM mode, I haven't really documented it anywhere. 

When you're in EM mode there is typically just a default admin password (it's "password"). One of my enhancements to ACP_Commander is for it to just try "password" if you don't specify one and then prompt you if it gets rejected. As such you shouldn't have to worry about the password while the device is in EM mode.

Like the Debian installer, EM mode runs entirely in the devices RAM. The /boot that you find there isn't mounted on any disk by default. You should be able to get to the installer from EM mode by doing something like this:

1. Boot into EM mode with no drives
2. Insert the drives.
3. confirm that the raid arrays are detected with `mdadm --detail --scan`
  3a. If they aren't try to start them with `mdadm --assemble --scan`
  3b. if that doesn't work there are other steps to look into
4. find the boot array, it will be the smallest and is typically md0
5. mount it on /boot `mount /dev/md0 /boot`
6. copy the boot files like you normally would
7. reboot and hopefully the installer will start

1000001101000, 2020-09-19:
I should add:
get the device in emergency mode however you have to. Then install all 4 drives. Try to avoid having only some of the drives from the array in at a time.

mholl1983, 2020-09-19:
Thanks again. I tried with no drives but I hear the "siren" sound and it's not emergency mode. I tried again with one drive, got to EM, then removed the drive. But for step 3, would those commands need to be done on a Linux system?

mholl1983, 2020-09-20:
Just wondering if it'd be easier to go back to stock before I try again w/ Debian.

1000001101000, 2020-09-20:
sorry, you would send those commands using ACP_commander. you can use "-s" to get a semi-interactive shell or try starting telnet with "-o" (I don't know if that works for this device's em mode).

You could also get back to the stock firmware by follwoing the same method and putting the stock *.buffalo files back in place. If you used acp commander to transfer the files there should be backups named with *.bak already there.

You could just re-install the firmware since your in EM mode but I don't really recommend it. I don't think you'd end up wiping your data or making things worse, but re-installing the firmware with data you want to keep makes me nervous. 

...I mean, so does the Debian install... everything makes me nervous if I don't have a backup.

mholl1983, 2020-09-20:
Thanks again for clarifying. Here's what I tried today:

1. Took out all 4 drives.
2. Inserted my "test drive" in bay 1.
3. TS booted and went into EM.
4. I took test drive out and added back the 4 drives.
5. Device initially went into "degrade mode".
6. After I put drive 1 back in, device went into "set from web i32".

I can't seem to keep the TS in emergency mode. :( The only way I've been able to was by keeping the test drive in bay 1. Can I start with the test drive and bring the boot files over, and THEN during the installer, swap that drive out for my original 4?

mholl1983, 2020-09-20:
PROGRESS! I put the test drive back in and served the files again. Rebooted the TS and it made the siren sound/E06 error, THEN a green LCD message appeared: "Terastation arm debian installer". WOO! Finally getting somewhere and it's all thanks to you!

So, I want to be careful with the next steps. Am I safe to remove the test disk and put back in my original 4 drives before I proceed with the installer?

1000001101000, 2020-09-20:
That's exactly what I was going to recommend.

Usually it will get the same IP address it had previously if you are using DHCP on your network. It should have port 22 open so that you can connect via SSH. If for some reason it got a different IP address you should be able to find it by scanning for devices with port 22 open with NMAP or something similar.

mholl1983, 2020-09-20:
Thank you! 

Based on the forum posts, I set these network settings on my local desktop, after which I'll plug my NAS in directly: 

[image]

Does that look acceptable? Also, I've installed nmap, if you could let me know what specific command to run, I'd appreciate. (Indebted for your help and will leave a donation regardless of his this turns out!)

mholl1983, 2020-09-20:
I just tried to run what i think is the correct command: nmap -sS -p 22 192.168.50.1/24

The TS is still connected to my network. 9 SSH-able hosts show up but none of them is the TS device. :(

1000001101000, 2020-09-20:
Those settings are specifically for TFTP recovery, which you aren’t doing.You don’t need to mess with your network settings... that'll actually cause problems.

Your nmap command will be similar to:
`nmap -p22 —open 192.168.1.*`

1000001101000, 2020-09-20:
Was the NAS connected to the network the whole time? If not the network setup could have failed.

I’d suggest restoring your PC’s network settings, make sure the device is connected, the reboot it into the installer again.

Hopefully you’ll be able to find it at that point.

The point where the network comes up may be a few minutes after the display changes.

mholl1983, 2020-09-20:
Thanks! I moved it to the PC and back again so I'll try what you suggest.

Any risk of the installer going away if I reboot the device or is that in the device's memory now? Worried my first drive will overtake the sequence and I'll get the auto-shutdown issue again.

1000001101000, 2020-09-20:
If I understand correctly, you now have a spare drive with the installer files on it and four you plan to use in the NAS that aren’t in a bootable state. 

You’ll need to boot from your spare drive to get back into the installer but it should be as easy as swapping drives.

***lol, this is basically how I did it the first time. This is where extra drive trays would be really handy. Hopefully you won’t need to swap many more times.

mholl1983, 2020-09-20:
Yep, once again you hit the nail on the head! lol. What a pain; but it's a good learning experience. Will try this again after a break and update you soon. Keep your fingers crossed and thanks for all your help as always!

mholl1983, 2020-09-20:
Actually, went for the drive swap just to start. The spare drive is in but I got the E06/beeping error. Before, it eventually went away and now it seems to be staying there longer. Grrrr. Two steps forward, one step back!

1000001101000, 2020-09-20:
I don’t have a great theory for why that would happen. Best guess would the drive not being seated properly.

I’d say unplug the device, pull out the drive, put it in slot 1, then power on.

mholl1983, 2020-09-20:
Back at it and still getting an error state, even with the test drive reseated. Not sure how I got the installer to show up before. Should it be okay with just one drive in the first bay?

mholl1983, 2020-09-20:
I'm going to try these steps with the stock firmware files and see if that gets me out of a jam first.

1000001101000, 2020-09-20:
Yeah, I'm missing some details regarding how you got it to work the first time. 

I'd say start over and focus on setting up the spare drive again. Set it up with one small ext3 partition and then copy the boot file to it. I believe I have the full commands for that in the instructions.

1000001101000, 2020-09-20:
This would make more sense if what you really did was boot via TFTP, though I don't recall you actually saying that. 

Have you done anything with TFTP during this process?

mholl1983, 2020-09-20:
Hey, no nothing with TFTP during this process. I'll try what you suggest with the drive ext3 partition/starting over.

mholl1983, 2020-09-20:
Some details on the drive, just in case there's any clues there. 

[image]

[image]

1000001101000, 2020-09-20:
the filenames look right, the partitioning looks like it is probably okay. I've never done this from windows so I'm not super confident.

A test would be whether than drive as-is will boot into EM mode, if so then it is booting form those files. You should be able to replace it with my files and boot into the installer instead.

mholl1983, 2020-09-20:
Thanks, it worked at last! I'm going through the installer now.

1000001101000, 2020-09-20:
nice!

mholl1983, 2020-09-20:
Following this:

set the first raid device as /boot and format as ext3
set the second raid device as swap
set the third raid device as "/" and format as ext4

But the video at https://www.youtube.com/watch?v=dfMHlLmsSF8 differs slightly.

first device is /boot and ext3
second device is root (/)and ext3
third device is swap (already set up that way in the installer)

here's what i have so far:

[image]

mholl1983, 2020-09-20:
He also does this for his 3 tb drive:

[image]

mholl1983, 2020-09-20:
Right, you said leave the 4th array alone, of course.

Here's the summary of changes I made. Can you let me know if this looks okay before I start the install? Thanks!

[image]

1000001101000, 2020-09-20:
Looks good!

I think the video was non-raid, he also did things a little different. Was pretty neat having someone do that with my project.

1000001101000, 2020-09-20:
oh, yeah... he sets a mount point for the data array. That would work but then you need to make sure you don't format it! I typically just do the install and then set up the array in Debian later. I think I added instructions for that in the post-installation page.

mholl1983, 2020-09-20:
It's installing now. Cannot thank you enough! Sent a small token of gratitude.

Is there a GUI or something similar to the stock web interface that I can install?

1000001101000, 2020-09-20:
Awesome, thanks!

I know folks have used Webmin (I think he did in that video) and OpenMediaVault. I don't really recommend either. 

OpenMediaVault is a pain to install and can mess up your system but seemed to work reasonably well the one time I tried it. It's designed for faster systems and is pretty slow.

Webmin is pretty easy to install and the interface doesn't seem that slow but some of the backend processes end up eating up a ton of resources and installing plugins seemed to run forever. 

I wish someone would make something similar that is more lightweight but I'm not aware of one.

mholl1983, 2020-09-20:
Cool, thanks. So you'd recommend to keep it command/ssh based?

1000001101000, 2020-09-20:
Yeah, though that's easy for me to say since I've been using Linux for over 15 years at this point. 

These days there are a lot of good tutorials to do just about anything you want, and you could even try new things in a VM before doing it on the NAS in case you're worried about messing something up. Samba intimidated me for years but it's pretty easy if you follow the instructions (it requires separately adding users to samba and linux and setting up shares in the config file).

mholl1983, 2020-09-20:
Thanks again. I hope to get more comfortable and you've helped me go off to the races!

Just starting w/ the raid array rebuild. I see these entries after I do mdadm --detail --scan

ARRAY /dev/md/10 metadata=1.2 name=TS-XEL-EM02C:10 UUID=42d48385:163a55cd:7bc70b                                                                       78:e5487c19
INACTIVE-ARRAY /dev/md2 metadata=1.2 name=TS-XEL02C:2 UUID=859c4b86:c7851c56:8d0                                                                       f5d33:879a3582
ARRAY /dev/md0 metadata=0.90 UUID=6940760e:81cdc4aa:529ae12f:b827b7aa
ARRAY /dev/md/1 metadata=1.2 name=TS-XEL-EM02C:1 UUID=64074d9f:3ef3678e:6a037cb6    

the inactive array kind of caught my eye. is that cause for concern?

mholl1983, 2020-09-20:
I added them using mdadm --detail --scan >> /etc/mdadm/mdadm.conf. Cool?

[image]

1000001101000, 2020-09-20:
remove the duplicates and change INACTIVE-ARRAY to ARRAY

1000001101000, 2020-09-20:
see if you can start the array with 
`mdadm --start /dev/md2`

(it's been a little while since I've done this)

1000001101000, 2020-09-20:
you can check on the status of the array with 
`mdadm --detail /dev/md2`

Once you get it working (you may need to use --assemble) you'll want to update your initrd with:
`update-initramfs -u`

after that reboot and then verify that it is active after booting. 

The last step will be to create an fstab entry for the array. I believe I covered that in the instructions better than I could off the top of my head. This is the step i used to mess up all the time because I didn't understand the boot process very well. You need to make sure your fstab entry is working before rebooting, by default the system won't finish booting if it fails to mount the array at boot. On a PC this wouldn't be a big deal but since you don't have a monitor+keyboard it's a real pain to deal with. 

make sure the array mounts with the following command when you get fstab updated:
`mount -a`
you can verify it mounted by checking the free space
`df -h`

If it mounts succesfully, update your initrd again and reboot:
`update-initramfs -u`

1000001101000, 2020-09-20:
also, don't panic if you manage to mess up the device so it doesn't boot all the way. The installer can be used for recovery if needed. 

.... I should probably write a wiki page about that, I really need to set aside some time to work on documentation to be honest.

mholl1983, 2020-09-20:
Thanks for reassurance!

On the inactive array, I did some details and I get this:

[image]

Not a clue how to proceed on that one. May have to call it a day for now but hopefully getting somewhere.

mholl1983, 2020-09-20:
This too:

[image]

1000001101000, 2020-09-20:
do you still have the "spare" drive plugged in?

what does "--detail" look like for the other arrays.

mholl1983, 2020-09-20:
Spare drive is out of the mix now, so the 4 original drives are in place!

The other arrays look fine and are in "activesync" state.

1000001101000, 2020-09-20:
huh, what type of raid is this? if it's not RAID0 you can probably just re-add the partition and have it re-sync. If it is RAID0 i'd have to look up what to do to be honest.

mholl1983, 2020-09-20:
Which array, you mean? It's RAID5.

1000001101000, 2020-09-20:
The degraded array.

I don't know why it would have lost the partition, I also don't know how to force it back. I think you're better off re-adding it likea new partition and have it rebuild. I've never actually done it with Raid5 I think it's just "--add" but you should probably get some sleep and do a little research before trying it.

You should also install smartmontools and check out the health of the drives while we're at it.

mholl1983, 2020-09-20:
"get some sleep and do a little research before trying it" -- good advice as usual. :) WIll do just that. And thanks for mentioning smartmontools; will make sure I get that running next time too. Cheers!

1000001101000, 2020-09-20:
I suppose you could also see if it magically assembles on a reboot now that it's in mdadm.conf.

mholl1983, 2020-09-21:
Sadly, it didn't magically assemble but good thought!

mholl1983, 2020-09-21:
Morning! Looks like this command set might be a clue:

mdadm --stop /dev/mdX
mdadm --assemble --force /dev/mdX /dev/sdX /dev/sdY

(from https://superuser.com/questions/993259/why-is-my-raid-1-disk-inactive)

So, "X" would be 2 given that it's in the partition name. But what would "Y" variable be in this scenario?

[90 more replies]